lvgl/lvgl #1914: Visibility only on the parent bug

FlyLu, 2020-11-19:
The parent  does not override the child 

lvgl version:
commit f32526562a9b338d0b241a381c7729a5c05f7e60 (HEAD -> release/v7, origin/release/v7)

[image]

Look at the red highlights
This looks like a bugs
Here is my test code
```
void lv_obj_style_test(void)
{
    lv_obj_t *parent = lv_obj_create(lv_scr_act(), NULL);
    lv_obj_set_size(parent, 360, 263);
    lv_obj_align(parent, NULL, LV_ALIGN_CENTER, 0, 0);

    static lv_style_t parent_style;
    lv_style_init(&parent_style);
    lv_style_set_radius(&parent_style, LV_STATE_DEFAULT, 40);
    lv_obj_add_style(parent, LV_OBJ_PART_MAIN, &parent_style);

    lv_obj_t *chlid = lv_obj_create(parent, NULL);
    lv_obj_set_size(chlid, 180 + 40, 206);
    lv_obj_align(chlid, parent, LV_ALIGN_IN_BOTTOM_LEFT, -40, 0);

    static lv_style_t chlid_style;
    lv_style_init(&chlid_style);
    lv_style_set_radius(&chlid_style, LV_STATE_DEFAULT, 0);
    lv_style_set_bg_color(&chlid_style, LV_STATE_DEFAULT, LV_COLOR_MAKE(0x35, 0xC0, 0xE7));
    lv_obj_add_style(chlid, LV_OBJ_PART_MAIN, &chlid_style);
}
```

embeddedt, 2020-11-19:
Works as intended. By default no clipping is done on the corners because it costs performance and the symptom is not usually that obvious. In your case you can use `lv_style_set_clip_corner(&parent_style, LV_STATE_DEFAULT, true);` and that should fix the problem.

FlyLu, 2020-11-19:
@embeddedt I verified and it works ok
 Thank you very much! thank you very much! thank you very much!
 Tell the important thing three times

FlyLu, 2020-11-24:
@embeddedt  hi
click on the corner that is not visible, but the object is selected
how should i fix this problem
[image]

embeddedt, 2020-11-24:
It's a known limitation for performance reasons. It's significantly faster to check whether the click is within a rectangle than within a circle.

You would need to use the advanced hit-testing feature and supply the appropriate radius to `_lv_area_is_point_on`. You can see an example of this being done for the color picker here:

`lv_obj_set_adv_hittest(cpicker, true);`
https://github.com/lvgl/lvgl/blob/5848aa59e591a49724c708165c57a46d4cdd557f/src/lv_widgets/lv_cpicker.c#L870-L873
https://github.com/lvgl/lvgl/blob/5848aa59e591a49724c708165c57a46d4cdd557f/src/lv_widgets/lv_cpicker.c#L906-L920

FlyLu, 2020-11-24:
@embeddedt 　thank you for your reply

let me talk about the actual scene

In my project, the parent object is a circle image, I created a rectangle on it and set it to be transparent
because the object is only visible on the parent, the rectangle excess part will not be displayed
I want to rotate the parent object by dragging to the rectangle, but I don’t want to drag in the invisible area

My thinking is like this
Monitor the LV_SIGNAL_COORD_CHG of the dragged object and set the image angle

[image]

I want to solve invisible places that cannot be dragged

embeddedt, 2020-11-24:
I believe the hit test solution would work for dragging.

FlyLu, 2020-11-24:
thank you so much, I will study the method you said now, if I solve it, I will tell you

closed 2020-11-25 by FlyLu

FlyLu, 2020-11-25:
@embeddedt thank you!  thank you! thank you!

I fix the problem with  hit-testing feature

If the input point is not on the drag area, drag is disabled, and drag is enabled when the event is released

```
drag_new_signal_cb:
    if (LV_SIGNAL_HIT_TEST ==  sign) {
        lv_hit_test_info_t *info = param;

        lv_obj_t *drag_area = lv_obj_get_user_data(obj);
        if (!_lv_area_is_point_on(&drag_area->coords, info->point, LV_RADIUS_CIRCLE)) {
            lv_obj_set_drag(obj, false);
        } 
    }

drag_event_cb:
    if (LV_EVENT_RELEASED == event) {
        if (!lv_obj_get_drag(obj)) {
            lv_obj_set_drag(obj, true); 
        }
    }
```

I am in the morning, you are at night, and vice versa