lvgl/lvgl #1915: list can not auto move ?

Layty, 2020-11-20:
lvgl :  master 

 I use  ↑↓←→ to control list , But It can not auto show down

[image]

config is here
```
/* Maximal horizontal and vertical resolution to support by the library.*/
#ifdef __amd64__
#define LV_HOR_RES_MAX          (160)
#define LV_VER_RES_MAX          (160)
#else
#define LV_HOR_RES_MAX          (160)
#define LV_VER_RES_MAX          (160)
#endif

/* Color depth:
 * - 1:  1 byte per pixel
 * - 8:  RGB332
 * - 16: RGB565
 * - 32: ARGB8888
 */
#ifdef __amd64__
#define LV_COLOR_DEPTH 1
#else
#define LV_COLOR_DEPTH 1
#endif
```

my code  is here 
```
#include "../lv_ex_conf.h"
#include "../lv_drivers/indev/keyboard.h"
#include "../lv_drivers/indev/mousewheel.h"

static lv_group_t*  g;

void mylist(void)
{
    g = lv_group_create();
    // lv_group_set_focus_cb(g, focus_cb);

#if LV_EX_KEYBOARD
    lv_indev_drv_t kb_drv;
    lv_indev_drv_init(&kb_drv);
    kb_drv.type = LV_INDEV_TYPE_KEYPAD;
    kb_drv.read_cb = keyboard_read;
    lv_indev_t * kb_indev = lv_indev_drv_register(&kb_drv);
    lv_indev_set_group(kb_indev, g);
#endif

#if LV_EX_MOUSEWHEEL
    lv_indev_drv_t enc_drv;
    lv_indev_drv_init(&enc_drv);
    enc_drv.type = LV_INDEV_TYPE_ENCODER;
    enc_drv.read_cb = mousewheel_read;
    lv_indev_t * enc_indev = lv_indev_drv_register(&enc_drv);
    lv_indev_set_group(enc_indev, g);
#endif

   lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
   lv_group_add_obj(g,list);
    lv_obj_t* bt;

   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_OK, "Apply"));
   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_CLOSE, "Close"));
   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_EYE_OPEN, "Show"));
   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_EYE_CLOSE, "Hide"));
   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_TRASH, "Delete"));
   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_COPY, "Copy"));
   lv_group_add_obj(g,lv_list_add_btn(list, LV_SYMBOL_PASTE, "Paste"));

   lv_group_set_editing(g,false);
   lv_group_focus_obj(list);
}
```

kisvegabor, 2020-11-21:
> when I press enter and not release and then press ↑ , then show like

LVGL doesn't support pressing 2 buttons at the same time. So enter + up is invalid.

Layty, 2020-11-22:
so I think it'a bug, I think we should not black the last button.
  for the picture is the item4(black one) , because in the end I have release the enter key and just use ↑↓, but item 4 always show black

kisvegabor, 2020-11-23:
I tried your code but couldn't reproduce the issue by using the mono theme. Do you also use the mono theme?

I've noticed you add every button to the group but it's not required. `lv_list` manages its buttons automatically.

Layty, 2020-11-25:
here is my theme config 
```c
/*================
 *  THEME USAGE
 *================*/

/*Always enable at least on theme*/

/* No theme, you can apply your styles as you need
 * No flags. Set LV_THEME_DEFAULT_FLAG 0 */
#define LV_USE_THEME_EMPTY       1

/*Simple to the create your theme based on it
 * No flags. Set LV_THEME_DEFAULT_FLAG 0 */
#define LV_USE_THEME_TEMPLATE    1

/* A fast and impressive theme.
 * Flags:
 * LV_THEME_MATERIAL_FLAG_LIGHT: light theme
 * LV_THEME_MATERIAL_FLAG_DARK: dark theme
 * LV_THEME_MATERIAL_FLAG_NO_TRANSITION: disable transitions (state change animations)
 * LV_THEME_MATERIAL_FLAG_NO_FOCUS: disable indication of focused state)
 * */
#define LV_USE_THEME_MATERIAL    1

/* Mono-color theme for monochrome displays.
 * If LV_THEME_DEFAULT_COLOR_PRIMARY is LV_COLOR_BLACK the
 * texts and borders will be black and the background will be
 * white. Else the colors are inverted.
 * No flags. Set LV_THEME_DEFAULT_FLAG 0 */
#define LV_USE_THEME_MONO        1

#define LV_THEME_DEFAULT_INCLUDE            <stdint.h>      /*Include a header for the init. function*/
#define LV_THEME_DEFAULT_INIT               lv_theme_mono_init
#define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_WHITE
#define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_WHITE
#define LV_THEME_DEFAULT_FLAG               LV_THEME_MATERIAL_FLAG_LIGHT
#define LV_THEME_DEFAULT_FONT_SMALL &lv_font_montserrat_12
#define LV_THEME_DEFAULT_FONT_NORMAL &lv_font_montserrat_12
#define LV_THEME_DEFAULT_FONT_SUBTITLE &lv_font_montserrat_12
#define LV_THEME_DEFAULT_FONT_TITLE &lv_font_montserrat_12

```

totol lv_conf.h is 
```c
/**
 * @file lv_conf.h
 * Configuration file for v7.7.2-dev
 */

/*
 * COPY THIS FILE AS `lv_conf.h` NEXT TO the `lvgl` FOLDER
 */

#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H
/* clang-format off */

#include <stdint.h>

/*====================
   Graphical settings
 *====================*/

/* Maximal horizontal and vertical resolution to support by the library.*/
#ifdef __amd64__
#define LV_HOR_RES_MAX          (160)
[... 724 lines cut ...]
/*Tileview (dependencies: lv_page) */
#define LV_USE_TILEVIEW     1
#if LV_USE_TILEVIEW
/*Time of slide animation [ms] (0: no animation)*/
#  define LV_TILEVIEW_DEF_ANIM_TIME   300
#endif

/*Window (dependencies: lv_cont, lv_btn, lv_label, lv_img, lv_page)*/
#define LV_USE_WIN      1

/*==================
 * Non-user section
 *==================*/

#if defined(_MSC_VER) && !defined(_CRT_SECURE_NO_WARNINGS)    /* Disable warnings for Visual Studio*/
#  define _CRT_SECURE_NO_WARNINGS
#endif

/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/

```

Layty, 2020-11-25:
I am working on linux and use fb, I have try at ubuntu and my arm board,  all the same bug

kisvegabor, 2020-11-25:
Strange. Please make a video from the simulator including your keyboard to see what button you press.

Layty, 2020-11-26:
[image]

1. use   ↓ to select  at   1
2 release  ↓ 
3. press enter and not release
4. press ↓ , at this time enter not release 
5. release ↓ and then press ↓ or ↑ to select others 
6. release ↓ 
7. release enter 

8.  use ↑ or ↓  ,then always have one black

kisvegabor, 2020-12-01:
I still can't reproduce no matter how hard I press ENTER... :slightly_smiling_face: 

Are `event_handler` and `listBtnCall` empty?

Layty, 2020-12-07:
Now I hvae just use our  visual_studio_2017_sdl project and use  lv_demo_keypad_encoder();  the same gif show follow

you can see the first list button "apply" is not ok 
[image]

Layty, 2020-12-07:
1. use mouse to select apply
2. press enter and do not release
3. press ↓ and not realease enter 
4. release ↓ then release enter
5.  you can see apply is always show swlect 

in fact , i can select all list button
[image]

Layty, 2020-12-07:
Chinese "左键" in the gif  means  I use mouse left click to select the list button

Layty, 2020-12-07:
visual_studio_2017_sdl  version is today, I just fix here 
[image]

Layty, 2020-12-07:
I think we need to redraw the whole list when event process, and not just redraw last button and current button

now I try to fix  in 
```
static lv_res_t lv_list_signal(lv_obj_t * list, lv_signal_t sign, void * param)

```

```
    else if(sign == LV_SIGNAL_CONTROL) {
        lv_obj_t* btn2 = NULL;
        btn2 = lv_list_get_next_btn(list, btn2);
        while (btn2 != NULL)
        {
            lv_obj_set_state(btn2, LV_STATE_DEFAULT);
            btn2 = lv_list_get_next_btn(list, btn2);
        }
```

kisvegabor, 2020-12-07:
As I've mentioned earlier it happens because LVGL doesn't support having 2 keys presed at the same time. 
So you can't press ENTER and down together. 

I could also reproduce the issue but you need to manage it o the driver level. E.g. don't "use" the new key if an other one is being pressed.

Layty, 2020-12-08:
yes, but we donot knoe  to how our user  to press button, so the display should display correct  and we can just process the last key  event

embeddedt, 2020-12-08:
You manage it with whatever glue code is being used to process keys and send them to LVGL.

closed 2021-01-14 by Layty