lvgl/lvgl #1918: Preprocessor macros should not end with a semi-colon

xennex22, 2020-11-21:
Preprocessor macros generally should not end with a semi-colon, as they are expanded to include an empty statement.  eg

https://github.com/lvgl/lvgl/blob/baa1f848b1bc74cb7533b99abc3bd3fcfabbe383/src/lv_misc/lv_color.h#L111
https://github.com/lvgl/lvgl/blob/baa1f848b1bc74cb7533b99abc3bd3fcfabbe383/src/lv_misc/lv_color.h#L159
https://github.com/lvgl/lvgl/blob/baa1f848b1bc74cb7533b99abc3bd3fcfabbe383/src/lv_misc/lv_color.h#L521
```(c1).ch.red = (uint8_t)((premult_c1[0]) & 0x1);;```

I don't think we need to bracket for safety so can just leave the semi-colon out

``` # define LV_COLOR_SET_R1(c, v) (c).ch.red = (uint8_t)((v) & 0x1) ```

```(c1).ch.red = (uint8_t)((premult_c1[0]) & 0x1);```

kisvegabor, 2020-11-21:
Unfortunately, I can't see warnings for this. If you do can you send a PR to fix these?

xennex22, 2020-11-22:
I made the change on the dev branch; not sure if that gets merged into dev-8 or not when the time comes.

embeddedt, 2020-11-22:
As far as I know `dev-8` was merged into `dev` a few days ago, so `dev-8` is now outdated.

kisvegabor, 2020-11-23:
Thank you for the PR. It really should be based on `master`. Could you open it again? 

@embeddedt 
Yes, v8 is in `dev`. I've just deleted `dev-v8`.

closed 2020-11-25 by xennex22