lvgl/lvgl #1919: Need better inheriance support

puzrin, 2020-11-21:
Continue of #1874.

Currently LVGL has er primitive inheritance support for widgets.

- Only data inheritance supported
- No support for methods inheritance
- No support for constructor/destructor
- No widget types reflection

This cause different kinds of direct and indirect issues:

- Widgets extension is not human-friendly
- Memory use not optimal (many things can be moved to "class descriptor")
- Many things could be simplified (lv_obj_allocate_ext_attr, user_data, signals)
- Future problems with delegated listeners of bubbled events (filter by src type)
- other...

So, better inheritance support would be highly appreciated.

Known solutions for C:

- https://www.cs.rit.edu/~ats/books/ooc.pdf
- http://ooc-coding.sourceforge.net/
- https://developer.gnome.org/gobject/stable/
- https://github.com/CObjectSystem/COS
- https://github.com/ShmuelFine/COOP

Some of above are no for embedded, but could be useful to understand ideas & programming patterns.

kisvegabor, 2020-11-23:
Thanks for opening this issue. 

I wrote down my thoughts about how I imagined the implementation. It's just a topic starter to have something to talk about.   

The idea:
- class data simply extends the base class's data. Base data is accessed directly. So instead of `a->base->x`, use `a->x`. This way name collision can occur but it will turn out at compile time.
- class methods are managed similarly to data but here the base's class methods are available via a pointer to make the base's constructor/destructor/(lv_signal) available.

Sorry, I couldn't summarize it in a shorter form...

A base class could look like this:
```c
#define BASE_CLASS          \
   struct _class_base_t * base;      \
   void(*constructor_cb)();    \
   void(*descructor_cb)();    \
   uint32_t instance_size;    \  //size of the class with all the data
   uint32_t class_size;    \      //size of the class (mainly the methods)

//Just to have a type that can be referenced later
typedef struct _class_base_t
{
   BASE_CLASS
}class_base_t

typedef struct {
  class_base_t * class;
} base_t
```

Functions to manage classes:
```c
// Initialze a class dsc.	
void class_init(class_base_t * class, uint32_t instance_size, uint32_t class_size, lclass_base_t * base_class)
{
   memset(class, 0x00, class_size);

   //By deafult use the same methods as the base
   memcpy(class, base_class, base_class->class_size);

  class->base = base_class; //Save the base to allow accessing its methods later
  class->instance_size = instance_size;
  class->class_size = class_size;
}

// Instantiate a class 
void * class_create(class_base_t * class)
{
  base_t * instance = malloc(class->instance_size);
  instance->class = class; 

  //Call the constructor of base classes and this class
  construct(instance);
}
```

Describe a Person class
```c
#define PERSON_DATA \
  gender_t gender; \
  string name; \
  uint32_t age; \
  person_t * mother; \
  person_t * father; \

#define PERSON_CLASS \
  BASE_CLASS   \  //Add the base class's methods
  void(*born)(persont_t * person, persont_t * mother, persont_t * father);	\
  uint32_t max_age;		//Static data, same for every person		
	
typedef struct
{
  PERSON_CLASS
}person_class_t

//The Person class
typedef struct
{
  person_class_t * class;
  PERSON_DATA
} person_t;
```

Create a Teacher class based on Person. MAcros could be created to make struct creation easier and less error prone
```c
#define TEACHER_DATA  \
  PERSON_DATA           \   //Add base class's data too
  school_t school_id;	\  //Data new to teachers

#define TEACHER_CLASS          \
  PERSON_CLASS		        \ //Add base class's dsc too
  void (*set_school)(teacher_t * teacher, school_t id);	\
	
typedef struct
{
  TEACHER_CLASS
}teacher_class_t

//The Teacher class
typedef struct
{
  teacher_class_t * class;
  TEACHER_DATA
} teacher_t;
```

Init. the classes
```c
//Init the base class
static class_base_t class_base;
class_init(&class_base, 0, sizeof(class_base_t),  NULL);
class_base->constructor_cb = default_constructor;
class_base->destructor_cb = default_destructor;

//Init the Person class
static person_class_t person_class;
class_init(&person_class, sizeof(person_t), sizeof(person_class__t), NULL);
person_class.born = person_born;
person_class.max_age = 100;

//Init the Teacher class
static teacher_class_t teacher_class;
class_init(&teacher_class, sizeof(teacher_t), sizeof(teacher_class_t), &person_class);
teacher_class.born = teacher_born; //Let's say teachers born differently :) 
teacher_class.set_school = set_school;
```

Instantiate classes and use them
```c
persont_t *adam = class_create(&person_class);
persont_t *eve = class_create(&person_class);
persont_t *john = class_create(&person_class);

john->class->born(john, adam, eve);
john->name = "John Smith"

teacher_t *kate = class_create(&teacher_class);
kate->class->born(kate, adam, eve);  //Call `teacher_born`
kate->class->base->class->born(kate, adam, eve);  //Call `person_born`

//Ways to set the school id
kate->class->set_school(kate, SCHOOL_1);
//OR
teacher_class->set_school(kate, SCHOOL_1);
//OR
set_school(kate, SCHOOL_1);
//OR
kate->school_id = SCHOOL_1;

//set class data
kate->age++;
```

Type checking
```c
if(kate->class == &teacher_class) ...
if(kate->base->class == &person_class) ...
```

kisvegabor, 2020-11-23:
I find it quite simple to implement the above mechanism. If we can make a quick decision it can even go to v8.

embeddedt, 2020-11-23:
I understand the reasons for wanting to keep C, but I still feel like it's a waste of time to reimplement this type of advanced inheritance when C++ has better, more familiar syntax and an existing, battle-tested implementation. I think this needs more discussion before making it into a release.

embeddedt, 2020-11-23:
For the record, I don't wish to drop MicroPython support; we should find a way to maintain that regardless of our decision.

amirgon, 2020-11-23:
One of the advantages of LVGL today, is the simplicity its API.  
It's very easy for a new user (and also for automated scripts) to make sense of the API and use it: just create a widget (with lv_widget_create) and call its functions.

I hope you are not compromising on this simplicity when adopting some inheritance scheme.
Many users just want to make it work, as quick as possible, and not spend time learning a set of inheritance-related abstract macros and data structures when all they want is to create a screen with some widgets / styles / interactions.

embeddedt, 2020-11-23:
@amirgon Much better summary of what I'm trying to say! :+1:

kisvegabor, 2020-11-23:
I agree that real C++ would be better for obvious reasons but it also opens the possibility for other issues. Here are some concerns I have in mind:
- more difficult to predict how the compiled code "looks like" (we might see the unexpected regression speed and memory even only on some platforms)
- some C++ features might be not well supported in the embedded world. E.g. [exception handling in mBed](https://os.mbed.com/forum/mbed/topic/523/) or [some features in Zephyr](https://docs.zephyrproject.org/1.9.0/kernel/other/cxx_support.html). In Zephyr even `new` is not supported so I can't imagine how a user can create a button that still lives when the function exits. 

@amirgon 
My vision is to keep the API quite the same.

embeddedt, 2020-11-23:
>  I can't imagine how a user can create a button that still lives when the function exits.

Doesn't C++11 allow classes to have a custom `new` implementation? We could keep our own independent heap.

kisvegabor, 2020-11-23:
Can we sure if normal `new` is not supported, overloading `new` will work everywhere? If `new` is only a `malloc` why isn't is supported by default?

Honestly, I don't have enough experience with embedded C++ to see all the limitations but I consider it as a "bad sign" that after 15 minutes of Google-ing I found a ton of concerns about C++ in the embedded world and zero advocates. 
It makes me a little bit skeptical.

embeddedt, 2020-11-23:
It looks like the reason Zephyr doesn't support `new` is that it doesn't ship with a complete C++ library by default, however, you can enable it [with these instructions](https://github.com/zephyrproject-rtos/zephyr/issues/24571#issuecomment-617850530).

> a ton of concerns about C++ in the embedded world and zero advocates

I think it depends on the project. An existing C++ project developed with PCs in mind will probably use the STL and other things which significantly inflate code size. On the other hand, NuttX's NxWidgets library is written in C++ and didn't cause excessive size bloat when I looked at it years ago.

puzrin, 2020-11-23:
@kisvegabor 

> Honestly, I don't have enough experience with embedded C++ to see all the limitations but I consider it as a "bad sign" that after 15 minutes of Google-ing I found a ton of concerns about C++ in the embedded world and zero advocates.
It makes me a little bit skeptical.

See ETL, a perfect example of good programming. Modern cpp has no problems on emdeddeds. But as every instrument, it should be used properly.

kisvegabor, 2020-11-23:
Maybe I only need some time to get used to the idea. :slightly_smiling_face: 

What about the Micropython binding?

I've just opened a vote: https://forum.lvgl.io/t/vote-could-you-use-lvgl-if-it-were-based-on-c/3784

amirgon, 2020-11-23:
> What about the Micropython binding?

As I mentioned in https://github.com/lvgl/lvgl/issues/1874#issuecomment-729723079, current implementation of the Micropython binding requires that the public API (lvgl.h) would be declared in C. It will not work with C++.
  
It also requires a set of naming conventions as described in https://github.com/lvgl/lvgl/issues/1763

kisvegabor, 2020-11-23:
> As I mentioned in #1874 (comment), current implementation of the Micropython binding requires that the public API (lvgl.h) would be declared in C. It will not work with C++.

So it's not enough if we keep the current API (e.g. `lv_obj_set_x`) we also need to keep the `lv_obj_t` `struct` (we can't replace it with a `class`). Is that correct?

amirgon, 2020-11-23:
> So it's not enough if we keep the current API (e.g. `lv_obj_set_x`) we also need to keep the `lv_obj_t` `struct` (we can't replace it with a `class`). Is that correct?

Correct, because the [parsing library](https://github.com/eliben/pycparser/tree/master#user-content-13which-version-of-c-does-pycparser-support:~:text=pycparser%20aims%20to%20support%20the%20full,patches%20to%20support%20more%20are%20welcome.)  only process C syntax.

In C++ `struct` and `class` are almost synonyms so this is not your road blocker, you can just use "struct" in place of "class".  
However, there is plenty of other C++ syntax you would probably want to use in C++ headers, and none of it is supported.

puzrin, 2020-11-23:
@kisvegabor 

> What about the Micropython binding?

https://wiki.python.org/moin/IntegratingPythonWithOtherLanguages there are multiple buning generators, some are ok with CPP (but i don't know how those are with micopython)

Existing binding will not survive, but in general - technical possibility should exist.

IMO it's not realistic to plan extensions while core is unstable. I'd expect general stability to ~v10. Also, it's unpleasant question, but answer yourself, what is potential LVGL market and segment of mycropython in it. That will give you idea, should existing binding block cpp or not.

@amirgon 

Please, don't consider my words as disregard of you work. My estimates are neutral and not include my cpp preferenes. If lvgl will do inheriance with C, i'll be ok too.

embeddedt, 2020-11-23:
> IMO it's not realistic to plan extensions while core is unstable. I'd expect general stability to ~v10. 

I agree, in fact I would suggest doing this development in parallel and only marking it for a specific release once we have something of substance. Ideally there should be no gap in functionality (MicroPython users should be able to use all official releases of LVGL). We could simply keep making v8 releases and avoid making any major architectural changes.

C47D, 2020-11-23:
I think the poll showed a clear winner, why is the consideration of changing the language of the whole framework?

[image]

puzrin, 2020-11-23:
@C47D in statistics it's very important to ask right question. For our case right quesion is "will you stop use LVGL if it switch to CPP"

https://forum.lvgl.io/t/vote-could-you-use-lvgl-if-it-were-based-on-c/3784

puzrin, 2020-11-24:
@kisvegabor 

Review of https://github.com/lvgl/lvgl/issues/1919#issuecomment-732093225

1. `\`-separated multiline declarations are so-so
2. You description explains only adding new methods.I could not understand, how to override methods (at least, constructor/destructor), and who is responsible for inheritance of implementations from parent

kisvegabor, 2020-11-24:
> \-separated multiline declarations are so-so.

You can write them in 1 line if you wish :slightly_smiling_face: 

> You description explains only adding new methods.I could not understand, how to override methods (at least, constructor/destructor), and who is responsible for inheritance of implementations from parent.

E.g.
```c

static teacher_class_t teacher_class;
class_init(&teacher_class, sizeof(teacher_t), sizeof(teacher_class_t), &person_class);
teacher_class.born = teacher_born; //Let's say teachers born differently :) 
teacher_class.set_school = set_school;
teacher_class.constructor_cb = teacher_constructor_cb; 
```
In `class_create`  the constructor of the followings will be called (in this order):
1. base
2. person
3. teacher

To reach the ancestors' methods `class_init` saves the `class_dsc` of the base in a linked list (`class->base = base_class;`).

> I don't see why your proposal is better than ooc kit (or it's restriced variant)

[2.4 Member functions](http://ooc-coding.sourceforge.net/docs/ooc.html/Objects-and-Classes.html#Objects-and-Classes) says: 
> A class member function is a normal C function, but there is a very important rule: the first function parameter of a member function is always a class instance object, and this first parameter can not be omitted.

That is class member functions are not encapsulated into the class and can't be inherited/extended.

---

Although there are only 10 answers in the poll now, 30% needs C.

embeddedt, 2020-11-24:
> Although there are only 10 answers in the poll now, 30% needs C.

It's been very interesting to watch that poll - the ratio between C to potential C++ users has not changed at all. Evidently a significant portion of the community wants C.

puzrin, 2020-11-24:
> ```c
> static teacher_class_t teacher_class;
> class_init(&teacher_class, sizeof(teacher_t), sizeof(teacher_class_t), &person_class);
> teacher_class.born = teacher_born; //Let's say teachers born [...]

Do you suggust user to hack methods manually? I don't understand how your suggested class declarations will be automaed to generae all this.

> [2.4 Member functions](http://ooc-coding.sourceforge.net/docs/ooc.html/Objects-and-Classes.html#Objects-and-Classes) says:
>
>
> That is class member functions are not encapsulated into the class and [...]

That's not our case. We need "virtual" funcions, those are ok.

kisvegabor, 2020-11-26:
> It's been very interesting to watch that poll - the ratio between C to potential C++ users has not changed at all. Evidently a significant portion of the community wants C.

Still 25-30% for strict C. 

---

> Do you suggust user to hack methods manually? I don't understand how your suggested class declarations will be automaed to generae all this.

Not hack, simply set a function pointer. What shall be automated? The base's constructor will be used if you don't set a new one. If you set a new one, it and the ancestor constructors will be called as well. 

> That's not our case. We need "virtual" funcions, those are ok.

In C having virtual functions and having methods with the same name in base and derived are very similar. You can consider these as virtual functions too as you can overwrite them. One thing is important though: we need to provide access to the base's methods to call `lv_bar_signal` from `lv_slider_signal`.

embeddedt, 2020-11-26:
I would say that the poll results thus far suggest the community would benefit from a C++ binding, but would *not* benefit from having the library itself be written in C++. The original poll about what language users are currently using LVGL in also shows an overwhelming number of C users.

puzrin, 2020-11-26:
> Not hack, simply set a function pointer. What shall be automated?

In ideal world, class should be created from declaration. In your case user will have to create declaration + add custom code. I'm not sure users will be happy to manually assign all methods to new lass.

> In C having virtual functions and having methods with the same name in base and derived are very similar. You can consider these as virtual functions too as you can overwrite them.

You missed my initial point - creating new lib instead of reuse existing codebase needs clear reasons. If reasons not exist, that's a waste of time. Lack of "member" fn inherit does no look as valid reason, because we are ok with all "virtual" methods, in terms of "ooc toolkit".

kisvegabor, 2020-11-27:
@embeddedt 
> I would say that the poll results thus far suggest the community would benefit from a C++ binding, 

IIRC, @amirgon has mentioned that the MP binding generator could be used to generate C++ binding too. Is it really the case?

---
@puzrin 
> In ideal world, class should be created from declaration. In your case user will have to create declaration + add custom code. I'm not sure users will be happy to manually assign all methods to new class.

It's the same in [ooc](http://ooc-coding.sourceforge.net/docs/ooc.html/Implementing-Classes.html#Implementing-Classes) too. 
[Here](https://stackoverflow.com/questions/41087485/how-to-divide-class-in-c-into-hpp-and-cpp-files) also advised splitting the class to declaration and implementation in C++.

> You missed my initial point - creating new lib instead of reuse existing codebase needs clear reasons. If reasons not exist, that's a waste of time. Lack of "member" fn inherit does no look as valid reason, because we are ok with all "virtual" methods, in terms of "ooc toolkit".

See how parent's virtuals can be accessed in ooc: [5.5.2 Class initialization](http://ooc-coding.sourceforge.net/docs/ooc.html/Implementing-Classes.html#Implementing-Classes). That is the parent's functions are not really "inherited inside" the children but you can access them via `->Parent.`. Let's say the parent has `a`, `b` functions and I want to add `c`. With occ I'll end up: `->Parent.a`, `->Parent.b`, `->c`. 
In my suggestion, you will natively get  this:  `->a`,  `->b`,  `->c``->Parent.a`, `->Parent.b`. `a` and `b` are `Parent->a` and `Parent->b` by default but you can change them to anything. 

My point is to not use an existing lib for such an important thing if we can implement our own solution in  2-3 days.  There is no big magic. In C we can do something like OOC or my proposal. The difference is that a custom solution is well understood, and tailored exactly to our needs.

puzrin, 2020-11-27:
> It's the same in [ooc](http://ooc-coding.sourceforge.net/docs/ooc.html/Implementing-Classes.html#Implementing-Classes) too.
> [Here](https://stackoverflow.com/questions/41087485/how-to-divide-class-in-c-into-hpp-and-cpp-files) also advised splitting the class to declaration and implementation in C++.

You mix different things. Splited implementation relies on declaration, but not tend user to manully patch it, as in your example. In ooc tookkit this happens transparent if i understand right.

> With occ I'll end up: `->Parent.a`, `->Parent.b`, `->c`.
> In my suggestion, you will natively get this: `->a`, `->b`, `` ->c``->Parent.a ``, `->Parent.b`. `a` and `b` are `Parent->a` and `Parent->b` by default but you can change them to anything.

That's reasonable. But we need to decide is the cost of this feature acceptable and what alternatives exists.

1. I don't know how that will be initialized (without extra manual coding). May be possible, but needs check.
2. Declaring class fields via multiline defines looks very suspicious (i don't know, may be that's normal practice in C).
   - And that may close forever road to multiple src inheritance
3. Possible alernative - if you call `.vtable` as `.method`, code will be `MyClass->mehod->foo()` instead of desired `MyClass->foo()`. Longer, but not too shitty.

> My point is to not use an existing lib for such an important thing if we can implement our own solution in 2-3 days. There is no big magic. 

Under lib reuse i mean "not reinvent everything" (copy-pase and some renaming are ok). I agree, in scope of your knowlegde/experience you can see problem trivial, but i see it completely different. Cost of mistake can be huge. I'm not fan of experimenting with live project [users] without extreme needs.

amirgon, 2020-11-27:
> IIRC, @amirgon has mentioned that the MP binding generator could be used to generate C++ binding too. Is it really the case?

Well, not as is, but the same approach and the same C parsing library could be used for that in principle. It might require considerable amount of work, though.

puzrin, 2020-11-27:
http://libcello.org/ - just for fun, what can be done with C.

https://www.youtube.com/watch?v=bVxfwsgO00o&feature=youtu.be&t=409

puzrin, 2020-11-27:
@kisvegabor i thinked a bit, and asked for help.

1. Seems details of vmt (virtual methods table) implementation are not critical.
2. If user will manually assign own vmt functions in constructor - that's not ideal, but acceptable convention.
3. There is the way to flatten nested structs in C99, without macro black magic.

```c
typedef struct foo 
{
  int a, b
} foo;

typedef struct bar 
{
  union
  {
    struct foo;
    struct foo base;
  };
  int c, d; 
} bar;

// Both should work
bar.a;
bar.base.a;
```

http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1570.pdf page 115.

But i'm not 100% sure about compilers. Could anyone check?

embeddedt, 2020-11-27:
I used this to test it:

```c
#include <stdio.h>

typedef struct foo 
{
  int a, b
} foo;

typedef struct bar 
{
  union
  {
    struct foo;
    struct foo base;
  };
  int c, d; 
} bar;

int main(void) {
    bar t;
    printf("%d", t.a, t.base.a);
    return 0;
}
```

GCC didn't like it.

```
test.c:6:1: warning: no semicolon at end of struct or union
 } foo;
 ^
test.c:12:15: warning: declaration does not declare anything
     struct foo;
               ^
test.c: In function ‘main’:
test.c:20:19: error: ‘bar {aka struct bar}’ has no member named ‘a’
     printf("%d", t.a, t.base.a);
```

The problem may be the "declaration does not declare anything".

puzrin, 2020-11-28:
Hmm... specially for GCC, try `-std=c11` or `-fms-extensions`

embeddedt, 2020-11-28:
C11 did not help. It does work with `-fms-extensions`, but that sounds like a compiler-specific flag.

puzrin, 2020-11-28:
May be gcc with -std=c17 will work without custom flag?

https://stackoverflow.com/a/14106853/1031804 that's not gcc-specific feature, but status is unclear.

BTW, guys who helped me with C, told here should be no chips without CPP support (useable for lvgl). It would be interesting to know details about those 30% who can use C only.

embeddedt, 2020-11-28:
My x86 compiler is GCC 7 and it doesn't recognize C17.

The ARM compiler is GCC 9 and it complains even with C17.

It seems that this feature may be in one or more standards but is not easily supported even with GCC. I'm nervous to include it in a library which is used by such a diverse set of compilers (GCC, IAR, Clang, etc.).

> It would be interesting to know details about those 30% who can use C only.

I agree; I'm not sure exactly how to go about doing that though.

puzrin, 2020-11-28:
https://www.google.com/search?q=21st+century+c+ben+klemens+pdf, chapter 11.

Inheritance in modern C.

---

Well, i'd propose following plan. Let's use ugly macro-based manual inheritance for next 2 years. When project become stable, it will be very easy to rewrite if needed.

Macro-based inheritance is ugly for bulk declaration only.

- Usage is ~ ok.
- Extending with couple of fields/methods is not too shity.

Painful question is how to create flat instance struct and flat vmt. But that's small % of code. In return, inheritance will significantly improve lvgl internals & memory use. I'd say, benefit is still huge.

kisvegabor, 2020-11-28:
> Painful question is how to create flat instance struct and flat vmt. 

It's flat in my proposal. 
**data** 
base data is simple added:
```c
#define TEACHER_DATA  \
  PERSON_DATA           \   //Add base class's data too
  school_t school_id;	\  //Data new to teachers
```
**methods**
similar to data but a pointer to base's methods is stored:
```c
#define TEACHER_CLASS          \
  PERSON_CLASS		        \ //Add base class's dsc too
  void (*set_school)(teacher_t * teacher, school_t id);	\
```

---

I wonder if we should postpone v8 and add
1. "classes"
2. faster and more conscious styles system

What do you think?

puzrin, 2020-11-28:
> It's flat in my proposal.

I see. That breaks my brain, but still better than nothing for C99 :(.

First, i suggest to clarify terms, because existing can cause confusion

1. __Instance__ data or __class insance__ - usually has only data properies
2. __Class descriptor__, __vmt__ or __v-table__ - usally sore pointers to [virtual] metods, but also can have pointer to parent and opional data props (for example, outer padding for touch, if the same for all instances)

About your suggestion.

Class descriptors are inherited too because parent required to fill it's part of vmt. Initialisation of class descriptor consists of:

- call parent's init overt our vmt.
- fill ref to parent.
- fill own (extendable) part of vmt.

That's a "constructor of class descriptor", don't confuse with constructor of instance [widget]

If you see no collisions here with your syntax - let's move forward.

> I wonder if we should postpone v8 and add
> 
>  1. "classes"
>  2. faster and more conscious styles system
> 
> What do you think?

If you intend to add classes asap - it certainly worth postpone v8. Can't comment about styles (don't remember details).

In general - up to 1 month shift for big architecture rework - ok in advance. If more - recheck for not falling into endless loop.

embeddedt, 2020-11-28:
If we're going to be adding more things to v8 we would need to continue 7.x development for a longer period of time. This may already be necessary because I'm not sure how quickly existing projects are going to move to v8.

I agree that it would make more sense to bundle these types of changes together instead of continually shifting the APIs in each release.

excitedbox, 2020-11-28:
Embeddedt just pointed me to this thread and I must say I agree that C on it's own is not optimal. I don't think we should switch the language for 2 very big reasons. 1 Support. Many small chips don't support the other options and if they do it is difficult to use. 2. There are already 50+ GUI systems for CPP and Python who needs another?
LVGL made a name for itself by being the only Arduino/Maker GUI system that is supported everywhere. If you switch to CPP people might as well use imgui or the other mature guis out there.

I actually reread that OOC book just last week and had a laugh when I saw it was the first suggestion. That is basically the bible for C based GUI systems. I looked into some smart pointer and ooc libraries on github last week and there are some specifically aimed at embedded systems so even though OO brings more overhead with it, it has to be doable.

C is an extensible language so any higher level features we need we can add to it. D is the language C should become in my opinion. It has the C syntax with the features that make other languages so much easier to use. At this point I think legacy support is holding C back instead of helping.

The API changes I have been working on actually assumed that LVGL worked a little more in the object oriented style without looking too deep into the underlying mechanics of it. (Yea I should have looked). It implements some of the methodologies in it's own way but doesn't give the polymorphism and some of the important benefits that real OOP does.

The thing is. It is easy for us to come in after Kiss and Embeddedt have done all this work and point out what can be done better but what they have done is pretty amazing and has taken a huge amount of effort. I know how hard it is to plan ahead for features that you don't even know you are going to need years down the road. We should try to use as much of that as possible not only because they already did the work but also because a total rewrite will take ages.

-We also need to consider what is the best way to go forward while maintaining as much of the code that is already written.
Nobody wants to start rewriting every single function which is also why I settled for function pointers to the lvgl functions already there.

-Legacy support for devices already out there. (There are MILLIONS of devices running on lvgl GUIs). I picked lvgl based not just on features it had but features it will have and the active development. Many others did as well. You don't want to alienate all your users.

- Assuming staying with C picking a library for OO support. 
Cello has lots of features but if they are slower than C++ you might as well use that instead. C++ was based on C so any C++ function can be made just as fast in C. If we pick some good base libraries and use them to build on I think that could lead to very good results

__________________________________________________________________________________________________________________________________

My Opinion is.
- We should go OOP because long term it is the only thing that makes sense
- Keep as much of the draw code as possible and just change how it is called on. (might not be possible)
- CAREFULLY pick a library/s that implements the OO features we need because I doubt we can do it better ourselves.
- Feature freeze lvgl for now and only fix bugs so that when the switch is made the last old style version is as good as possible.
- Tighter integration with the rest of your program code. (My API uses an Event/Action system which could be extended to call any function like a sensor or motor)
- Build a 4 part framework of: Core Functions, Style System, Event/Action System, Runtime System

_____________________________________________________________________________________________________________________________

Framework:
When it comes to UI there is a lot the web does right and many things it does wrong. Coming from a web background I think focus should be not just on how easy it is for us to develop but also for the user to make additions. This means splitting the system into a few pieces. This requires careful planing so you avoid the bloat that has taken JS frameworks a bridge too far and made them inefficient. A lot of that bloat comes from having too many incomplete frameworks. When you need 5 systems there is a lot of repetition. If lvgl does everything we eliminate that which is why we need to have 3 frameworks in 1.

1. Function Framework- Set of functions that are reusable throughout think a tiny version of QT. (Data structures, math functions etc. this could even include a second project for a sensor or motor library)
2. Style Framework- Set of widget primitives that can be used to build bigger widgets which are then used to build even bigger widgets. (think. HTML elements->bootstrap->UIkit->functional components)
3. Action Framework- Event or Triggers -> Actions system. Like an event listener in JS
4. GUI builder - This is a must because of who the ultimate user will be. People with Arduinos etc. with NO or little coding skills. I could write books just on this but I will save it for another day. I will say though the above setup will make a builder easier to implement.

The simplest way to implement this would be to have a loop (main) with 3 functions: 
1. Is the DOM (A DOM is a tree/list of draw functions and their order) 
2. Executes an Event/action queue (updates the DOM) 
[... cut ...]
If we can figure out a way of solving that as a way to finance lvgl that would kick ass. Maybe selling a few LCDs with official support through the website with some sort of product that helps you build better project cases. Certain plastics can be solvent welded together so selling sheets of that with solvent, Exacto knives and brushes will allow people to build better cases. Maybe even laser cut templates.
I know a plastics researcher from Monsanto who worked for the car industry for 40 years.

embeddedt, 2020-11-28:
> It is easy for us to come in after Kiss and Embeddedt have done all this work and point out what can be done better but what they have done is pretty amazing and has taken a huge amount of effort. 

I appreciate the credit, but I'm not sure I deserve to be listed right beside @kisvegabor in that sentence! He has done a phenomenal amount of work, especially for the last few releases where he has in total almost rewritten the entire library. I just give suggestions and help users on the forum.

> This is why I think a widget should just be a container of arrays. 1 array of styles, 1 of events/triggers and actions they execute.

I like this kind of approach. One thing I've noticed in a lot of smaller GUI libraries is that they have the ability to declare everything as a set of constants. This is really good because with careful planning you can avoid the requirement for heap allocation.

puzrin, 2020-11-29:
@kisvegabor 

Returning to https://github.com/lvgl/lvgl/issues/1919#issuecomment-732093225.

1. IMO this method of class init looks more simple:

```c
typedef struct {
  PERSON_CLASS
} person_class_t

// PersonClass initialized before

typedef struct
{
  TEACHER_CLASS
} teacher_class_t

// Make class initializer a part of vmt for simplicity
teacher_class_t TeacherClass = { .__class_init = teacher_class_init };

class_extend(void *to, void *from) {
    // Pin inherited methods 
	BaseClass(from)->__class_init(to);
	BaseClass(to)->parent = from;
    // Pin new methods & overrides 
    BaseClass(to)->__class_init(to);
};

class_extend(&TeacherClass, &PersonClass);
```

2. Instances should be created via `.create` (or `.new`) method:

```c
teacher_t *teacher1 = TeacherClass.create(/* optional params */);
```

3. May be worth replace term CLASS with VT to minimize confusion. Because our "classes" consist of 2 separate parts - data & v-table.

puzrin, 2020-11-29:
@excitedbox 

> 1 Support. Many small chips don't support the other options and if they do it is difficult to use.

Cold you give a couple of concrete examples?

> 2. There are already 50+ GUI systems for CPP and Python who needs another?

The same question. Could you provide couple of opensource CPP GUI systems, useable & ok with small memory embeddeds?

excitedbox, 2020-11-29:
There are heaps of CPP GUIs. NanoGui, ImGui, Nucklear, GuiLite (only 5000 LOCs), Nana, Elements, MyGui, GTK, TK, wxWidgets, Then there are the free ones that are not open source like the STM Cube or whatever, Microchip Harmony lib, QT for MCU, GUIX, The list goes on for days.

Git lists 3x as many C++ Gui projects as C and many of the C projects never lived past their first few days.

Last week I stumbled onto a talk about the problems a guy had getting a chip to run C++ even though it supported it. It seems many companies don't invest in C++ because they don't think that programmers will use it so even the ones that do support it only half ass the C++ libraries while having proper support for their C libraries.

Also have to keep in mind legacy chips that hobbyists might pull from old devices and tinker with This is after all a project for makers and learning and in many countries the new parts are hard to come by. In South America, Africa, Asia you might not be able to get or afford the newest Arduinos or even ESP32. You might be able to get some broken electronics though and use that chip and display.

C++ support on small MCUs is spotty at best. 
NXP has SOME C++ support but even some of their ARM based compilers based on GCC 9 only have C support.
The TI compiler seems to support C++98 and C++03 
Many PICs under 32bits don't have C++ compiler support at all. The PIC32s use a different ([MPLAB](https://www.microchip.com/en-us/development-tools-tools-and-software/mplab-xc-compilers) ) compiler. There are many many proprietary compilers that are not as mainstream that don't offer c++.  So sometimes it is not even the lack of a compiler but the application specific libraries that companies are paying many thousands for prefer C. Do you really want to hire a bunch of C devs and then need a C++ dev just for your GUI?

puzrin, 2020-11-29:
> There are heaps of CPP GUIs. NanoGui, ImGui, Nucklear, GuiLite (only 5000 LOCs), Nana, Elements, MyGui, GTK, TK, wxWidgets

Some of those have no good layout manager abstractions, some are too fat. I prefer cpp, but could not find something useable for my needs.

If not fall into architecture details, the most trivial sample - fonts quality.

So, i don't agree with "using cpp gives you better choice of useable gui libs".

> Also have to keep in mind legacy chips that hobbyists might pull from old devices and tinker with This is after all a project for makers and learning and in many countries the new parts are hard to [...]

That may sound impolite, but thinking about legacy & price too much - blocks moving forward. Prior to think about Africa, it worth start local :).

embeddedt, 2020-11-29:
To me the fairest solution is an object-oriented C API and a C++ wrapper. Then everyone benefits.

puzrin, 2020-11-29:
I think, fair/unfair criteria is subjective. Objective is answer to "How much extra efforts do you agree to spend for legacy support and who will pay for that" :).

Also, missed market segment can be compensated by faster progress & spread. There is no single answer about right choice.

kisvegabor, 2020-11-30:
@excitedbox 
First of thanks for sharing all these. Your approach relies on the idea of the elements are created on start-up but I think it's an important limitation in normal use:
- memory for all widgets is consumed, not only for the visible ones
- user can't create content based on runtime data dynamically

As we already discussed, I'm ok with it as an extension, but I'd like to keep the current dynamic nature of the library.

@embeddedt 
> I just give suggestions and help users on the forum.

And every user is extremely grateful for this :slightly_smiling_face: 

> This is why I think a widget should just be a container of arrays. 1 array of styles, 1 of events/triggers and actions they execute.

Besides styles + events:
- some control flags, e.g. clickable, scrollable, hidden etc
- coordinates
- parent

In v8 I made an optimization on this by moving the parent-related and other rarely used properties to dynamically allocates struct. It saves a lof RAM because most of the nodes are "end nodes" and have no children. See the new [lv_obj_t](https://github.com/lvgl/lvgl/blob/dev/src/lv_core/lv_obj.h#L225) and [lv_obj_spec_attr_t](https://github.com/lvgl/lvgl/blob/dev/src/lv_core/lv_obj.h#L195).

I'm thinking whether we should make it possible to attach multiple event handlers to an object. I mean NOT by type (onClick, onFocus) but events like now but multiple of them. 

---
Going back to the original ooc topic, I'll make the first implementation of my proposal to see it in the practice, and we will see.

puzrin, 2020-11-30:
> I'm thinking whether we should make it possible to attach multiple event handlers to an object. I mean NOT by type (onClick, onFocus) but events like now but multiple of them.

Not critical, but would be nice to have. It's better to recheck every event type. It's possible to store all listeners in single array, and filter event types inside. That fill work good for "rare" events like "click", but slower for frequent evens like "scroll" and "hover".

> In v8 I made an optimization on this by moving the parent-related and other rarely used properties to dynamically allocates struct. It saves a lof RAM because most of the nodes are "end nodes" and [...]

Splitted objects are not convenient. I suggest revisit that after inheritance improve. It's better to sort data by "real" an "cache" and split out cache only.

kisvegabor, 2020-11-30:
> Not critical, but would be nice to have. It's better to recheck every event type. It's possible to store all listeners in single array, and filter event types inside. That fill work good for "rare" events like "click", but slower for frequent evens like "scroll" and "hover".

Alright, however, it's an independent thing let's discuss it separately. (later :slightly_smiling_face: )

> Splitted objects are not convenient. I suggest revisit that after inheritance improve. It's better to sort data by "real" an "cache" and split out cache only.

It's just internal stuff, hidden by the API function. User can't see anything from it. 

---

I've added the first implementation of the classes See in c919a5f.
Here is a usage example: https://gist.github.com/kisvegabor/606ae61e3cc3d88fe5a079bc12f62014

kisvegabor, 2020-11-30:
@puzrin 
> teacher_t *teacher1 = TeacherClass.create(/* optional params */);

It's doable but why is it better than `person_t *p1 = LV_CLASS_NEW(person);`?

> May be worth replace term CLASS with VT to minimize confusion. Because our "classes" consist of 2 separate parts - data & v-table.

That true but what would be the benefit of handling:
- class methods
- class data
- instance data

In C++ you can also mix data and method declaration.

puzrin, 2020-11-30:
> It's doable but why is it better than `person_t *p1 = LV_CLASS_NEW(person);`?

2 reasons:

- Implicit code is more readable than macro
- Static validation of params

Your question is not correct. If you suggest wrapped call, then you have to prove why extra wrapping level is better :)

kisvegabor, 2020-11-30:
> Your question is not correct. If you suggest wrapped call, then you have to prove why extra wrapping level is better :)

Okay, actually I'm with `create` too. :slightly_smiling_face: 
(That is I have no reasons for the macro :smile: )

> VT is not VMT, it's V-Table. It can contain anything. I just point your attention, currently CLASS-named structs are NOT classes, only 1/2 of class. More close to trait.

I see. Yes, terminology still should be polished. But even at the cost of some inaccuracy, I'd like to avoid names like `vmt`, `vtable` or so and keep cleaner names that are more obvious for people who don't know the internals of C++.

kisvegabor, 2020-12-01:
See the updated implementation:
- usage: https://gist.github.com/kisvegabor/7bd380dfa62dc9e256b1582126c7b4d9
- output:
```
Create 3 people
-----------
person create
person construct
person create
person construct
person create static
person construct
Billy is child of John and Mary

Create 3 teachers
-----------
teacher create
person construct
teacher construct
teacher create
person construct
teacher construct
teacher create
person construct
teacher construct
prof. Mike is child of Mary and Billy
prof. Polly is child of Mary and Billy
Amber is child of Mary and Billy
```

puzrin, 2020-12-01:
1. `_person_data`, `_person_class_dsc` => `_person_props`, `_person_class_props`
2. What is difference between `create` and `create_static`? I don't understand intend of having 2 creators.
3. I don't like unscoped pinning methods after LV_CLASS_INIT(...). IMO this is part of consrctor function.
4. IMO macros except props are not needed. Implicit struct declaration will take not much more efforts, but will be more readable.

kisvegabor, 2020-12-01:
> _person_data, _person_class_dsc => _person_props, _person_class_props

I like it.

> What is difference between create and create_static? I don't understand intend of having 2 creators.

`create` dynamically allocates the instance (equivalent to `new`). `create_static` operates on globals, static, local variables (equivalent to `MyClass myObj;`). 

> I don't like unscoped pinning methods after LV_CLASS_INIT(...). IMO this is part of consrctor function.

The constructor is for the instances. The assignments after `LV_CLASS_INIT` set the methods of the class. 

> IMO macros except props are not needed. Implicit struct declaration will take not much more efforts, but will be more readable.

`LV_CLASS_DECLARE_END(teacher, person)` expands to:
```c      
typedef struct _teacher_class_t {                
  person_class_t * base_p;                           
  _teacher_create;                                  
  _teacher_create_static;                             
  _teacher_class_dsc                                  
}teacher_class_t;                                    
                                                         
typedef struct _teacher_t {                       
  teacher_class_t * class_p;                        
  _teacher_data                                       
} teacher_t;
```
It has a lot of specific names, the order of fields, etc. IMO it'd be error-prone to let it to the user.

puzrin, 2020-12-01:
> `create` dynamically allocates the instance (equivalent to `new`). `create_static` operates on globals, static, local variables (equivalent to `MyClass myObj;`).

I'm not familiar with C naming conventions, so `*_static` looks very confusing for me. I'd propose `.emplace(*p)` as here https://www.etlcpp.com/queue_spsc_atomic.html.

> The constructor is for the instances. The assignments after `LV_CLASS_INIT` set the methods of the class.

I mean not instance constructor but class constructor (initializer). This process is now splitted for too much of small steps in different places. I will update my previous example.

> `LV_CLASS_DECLARE_END(teacher, person)` expands to:
>
> ```c
> typedef struct _teacher_class_t {
>   person_class_t * base_p; [...]

I'd say, this macro use requires notable amount of pre-declared things, and work a bit not as may be expected. But prior to suggest alternative, i'd like to understand better questions below.

I can understand why `base_p` is splitted out from props macro - it needs concrete type instead of void. But what is reason to declare _teacher_create & _teacher_create_static the same way? Do we expect to actively use constructor params and signature should be unique for each instance type?

kisvegabor, 2020-12-01:
> I'm not familiar with C naming conventions, so *_static looks very confusing for me. I'd propose .emplace(*p) as here https://www.etlcpp.com/queue_spsc_atomic.html.

What about `inplace`? Frankly, I needed to check what `emplace` means in a dictionary :smile: 

> I mean not instance constructor but class constructor (initializer). This process is now splitted for too much of small steps in different places. I will update my previous example.

We could have a function like `teacher.class_init()` and move all `teacher.xyz = abc` stuff there. But it has to problems:
1. the smaller one: it requires an extra function to write, take care of its name, assign it to the `class`... And the function will be called only once. So it's just extra work to write that function.
2. It won't make it possible to use `const class_dsc`s when the methods are assigned at compile time.

> Do we expect to actively use constructor params and signature should be unique for each instance type?

Imageine: 
```c
person.create(father, mother);
teacher.create(father, mother, school_id);
```

puzrin, 2020-12-01:
> What about `inplace`? Frankly, I needed to check what `emplace` means in a dictionary smile

I'm ok with any of emplace/inplace. But my english is not enough good to say what is better. I'd use the same as ETL for more simple recognize. ETL will be part of many cpp projects with LVGL.

> We could have a function like `teacher.class_init()` and move all `teacher.xyz = abc` stuff there. But it has to problems:
>
>  1. the smaller one: it requires an extra function to write, take care of [...]

Thanks for explanation. Now understand reasons of sch split. I will try to think how to get best of both. At first glance, `.class_init()` may be "default" from BASE_CLASS, overriden only for non-const. That's only preliminary.

> > Do we expect to actively use constructor params and signature should be unique for each instance type?
> 
> Imageine:
> 
> ```c
> person.create(father, mother);
> teacher.create(father, mother, school_id);
> ```

Ok. 3 custom properties is strong reason to use macros.

[140 more replies]

closed 2021-04-24 by kisvegabor