lvgl/lvgl #1920: dropdown list doesn't process arabic letter

ashrafkamel5, 2020-11-22:
## description
 the dropdown lists don't process Arabic letters.

```c
static char dd_months[] = "يناير\n"
"فبراير\n"
"مارس\n"
"أبريل\n"
"مايو\n"
"يونيو\n"
"يوليو\n"
"أغسطس\n"
"سبتمبر\n"
"أكتوبر\n"
"نوفمبر\n"
"ديسمبر";

lv_obj_t* ddlist_start_month = lv_dropdown_create(NULL, NULL);
lv_dropdown_set_options(ddlist_start_month, dd_months);

```
## output
[image]

## solve
```c
static char dd_months[] = "يناير\n"
"فبراير\n"
"مارس\n"
"أبريل\n"
"مايو\n"
"يونيو\n"
"يوليو\n"
"أغسطس\n"
"سبتمبر\n"
"أكتوبر\n"
"نوفمبر\n"
"ديسمبر";

lv_obj_t* ddlist_start_month = lv_dropdown_create(NULL, NULL);
_lv_txt_ap_proc(dd_months, dd_months);
lv_dropdown_set_options(ddlist_start_month, dd_months);

```
## output 
[image]

## same with the container name

[image]

```c   
 lv_obj_set_style_local_value_str(h, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, "الأساسيات");

```
## and trying This doesn't work
```c   
char ctx_text[40];
    _lv_txt_ap_proc("الأساسيات",ctx_text);

    lv_obj_set_style_local_value_str(container, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, ctx_text);

```

## outputs garbage 
[image]

kisvegabor, 2020-11-23:
Hi,

I've fixed both in `master`. Please verify if they work correctly.

closed 2020-11-23 by kisvegabor

ashrafkamel5, 2020-11-23:
[image]

apparently that the container name is working,
but the whole example crashes now, :\ 
[image]

ashrafkamel5, 2020-11-23:
it was working fine, just before replacing the lvgl folder.
that is quite an error for such a small fix.

kisvegabor, 2020-11-23:
Can you see in Debug where it halts?

ashrafkamel5, 2020-11-23:
[image]

the error was presented in codeblocks simulation.
I saved my work at VS and did the changes and it works fine there.
many thanks,
now it is time for me to add an Arabic keyboard.
can you help, where to start or if there is something I should watch out for?

kisvegabor, 2020-11-24:
Great! 
I don't know what are the requirements for an Arabic keyboard. Could you describe it in a new issue?

commit d90cbe92 (2020-11-23): fix(arabic): support Arabic letter connections

fixes #1920
commit bf08ff44 (2020-12-01): Update Kconfig. (#1917)

* Kconfig: Add missing _HEX to LV_COLOR_TRANSP option.

* lv_conf_checker: Define LV_CONF_SKIP if CONFIG_LV_CONF_SKIP is defined.

* fix(indev): disabled object shouldn't absorb clicks but let the parent to be clicked

* Update README.md

* fix(arabic): support processing again alraedy processed texts with _lv_txt_ap_proc

* fix(textarea): support Arabic letter connections

fixes #1888

* Fix #1924: incorrect date on changelog

December 1st is two weeks from 7.7.2 release.

* Kconfig: Fix LV_CONF_MINIMAL related options.

* fix(arabic): support Arabic letter connection [... cut]