lvgl/lvgl #1921: LVGL Arduno IDE  ESP32  Compile error

dingrenjun, 2020-11-22:
I refer to this example, next week LVGL and lv_examples compilation prompt error

Configure LVGL
LVGL has its own configuration file called lv_conf.h. When LVGL is installed the followings needs to be done to configure it:

Go to directory of the installed Arduino libraries

Go to lvgl and copy lv_conf_template.h as lv_conf.h next to the lvgl folder.

Open lv_conf.h and change the first #if 0 to #if 1

Set the resolution of your display in LV_HOR_RES_MAX and LV_VER_RES_MAX

Set the color depth of you display in LV_COLOR_DEPTH

Set LV_TICK_CUSTOM 1

Configure the examples
lv_examples can be configures similarly to LVGL but it's configuration file is called lv_ex_conf.h.

Go to directory of the installed Arduino libraries

Go to lv_examples and copy lv_ex_template.h as lv_ex_conf.h next to the lv_examples folder.

Open lv_ex_conf.h and change the first #if 0 to #if 1

Enable the demos you want to use. (The small examples starting with lv_ex_...() are always enabled.)

dingrenjun, 2020-11-22:
Error prompt:
C:\Users\Vole\Documents\Arduino\libraries\lvgl-7.7.2\examples\LVGL_Arduino\LVGL_Arduino.ino: In function 'void setup()':
LVGL_Arduino:115:17: error: 'lv_ex_btn_1' was not declared in this scope
     lv_ex_btn_1();

embeddedt, 2020-11-22:
Did you include any lv_examples headers at the top of `LVGL_Arduino.ino`?

benjaminSchilling33, 2020-11-22:
@embeddedt when including `<lv_examples.h>` it compiles for me.

Why is this not already part of the code?

benjaminSchilling33, 2020-11-22:
See: https://github.com/lvgl/lvgl/pull/1922

dingrenjun, 2020-11-22:
> <lv_examples.h>

After I add #include <lv_examples.h>, the file will report an error

Error message:
Arduino\libraries\lv_examples-7.7.2\src/lv_examples.h:17:0,
lv_examples-7.7.2\lv_examples.h:19:26: fatal error: ../lvgl/lvgl.h: No such file or directory

kisvegabor, 2020-11-23:
Do you have the same directory structure regarding lvgl, lv_example, lv_conf.h and lv_ex_conf.h? 
[image]

Mesihas, 2020-11-30:
I have the same directory structure and I get:

'screenWidth' was not declared in this scope
    ` if(touchX>screenWidth || touchY > screenHeight)`

Not sure if still missing something.

kisvegabor, 2020-12-01:
I've just updated the example ino file and removed these variables:
https://github.com/lvgl/lvgl/blob/master/examples/LVGL_Arduino.ino

Trackhe, 2020-12-03:
I have the same problem. srry but i need to ask. Are you testing your code? Its maybe actually broken for Arduino IDE. After change .../lv_exemple.h to "lv_examples.h" came the next error > lvgl/lvgl.h but this cant be fixed with your "Do you have the same directory structure regarding lvgl, lv_example, lv_conf.h and lv_ex_conf.h?" because i have it already. so i replace all lvgl/lvgl.h to lvgl.h and jea next error...............argh in every demo came some ```libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.txt_small_compr_cb+0x0): undefined reference to `lv_font_montserrat_12_compr_az'``` errors and if i have a good day i see
```
/Users/trackhe/Desktop/testlvgl/testlvgl.ino: In function 'void setup()':
testlvgl:99:23: error: 'lv_demo_benchmark' was not declared in this scope
     lv_demo_benchmark();
                       ^ 
```
dacing on my screen. please fix it.

benjaminSchilling33, 2020-12-03:
I set it up for Arduino just 2 weeks ago and with the latest released Arduino version it works perfectly fine (on my PC). 

For a working example see https://github.com/benjaminSchilling33/lvgl_wemos_lolin32_joy-it_3_5

kisvegabor, 2020-12-04:
@Trackhe 
I've just pushed a fix to lv_examples. Please try with the latest version.

Also be sure to enable `LV_USE_DEMO_BENCHMARK` in `lv_ex_conf.h`

Trackhe, 2020-12-04:
For now i m srry if i was a rude guy with my comment.
So i deleted all files and downloaded the master from lv_examples and lvgl again. 14:50 04.12.2020
First error: lvgl/lvgl.h: No such file or directory # Fix lvgl/lvgl.h replace with lvgl.h
And it is Cpmpoling <3 But it render only the first frame. nothing more (this is maybe also the problem why touch is not working)

[image]

[testlvgl.zip](https://github.com/lvgl/lvgl/files/5643173/testlvgl.zip)

embeddedt, 2020-12-04:
Are you calling `lv_tick_inc` somewhere, or is it configured to use Arduino's `millis()` function?

Trackhe, 2020-12-04:
in the loop is a    delay(5);
the example is from https://github.com/lvgl/lvgl/blob/master/examples/LVGL_Arduino.ino
i added only 
```
    pinMode(12, OUTPUT); 
    digitalWrite(12, HIGH); 
```
in setup before lv_init(); for background led

embeddedt, 2020-12-04:
What are the contents of your `lv_conf.h` file?

Trackhe, 2020-12-04:
in lv_conf.h i only changed the 0 to 1

embeddedt, 2020-12-04:
Is `LV_TICK_CUSTOM` set to 1 in that file?

Trackhe, 2020-12-04:
no its standard 0

embeddedt, 2020-12-04:
Set it to 1 and try the code again. It needs to be 1 on Arduino as mentioned in the [documentation](https://docs.lvgl.io/latest/en/html/get-started/arduino.html#configure-lvgl).

Trackhe, 2020-12-04:
this is working i overread the last. i guess i make a toturial to use this in ardino i can only say good dam thank you ^^. 
just one. i have 2-5 fps with the esp32 wroom is this normal?
https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf

embeddedt, 2020-12-04:
Glad it works!

I can't comment on ESP32 speed as I'm not familiar with the platform.

Mesihas, 2020-12-04:
I am following all the steps / file structure  and also started from scratch by installing last version but still get errors like:
Documents/Arduino/libraries/lv_examples/src/lv_ex_widgets/lv_ex_roller/lv_ex_roller_1.c:31:25: error: 'LV_ROLLER_MODE_INIFINITE' undeclared (first use in this function)
                         LV_ROLLER_MODE_INIFINITE);
                         ^

embeddedt, 2020-12-05:
Change it to `INFINITE`; that bug has been fixed in LVGL `master` but the fix isn't in a release yet.

Mesihas, 2020-12-05:
reinstalled both libraries from master:

Documents/Arduino/libraries/lv_examples/src/lv_demo_music/assets/img_lv_demo_music_btn_list_pause.c:1:23: fatal error: lvgl/lvgl.h: No such file or directory
compilation terminated.

Trackhe, 2020-12-05:
> For now i m srry if i was a rude guy with my comment.
>
> So i deleted all files and downloaded the master from lv_examples and lvgl again. 14:50 04.12.2020
>
> First error: lvgl/lvgl.h: No such file or [...]

@Mesihas read it to fix your lvgl/lvgl.h error

kisvegabor, 2020-12-07:
I've pushed a fix to lv_examples to remove some unnecessary includes.

closed 2021-01-25 by stale[bot]