lvgl/lvgl #1924: Incorrect date in Release notes

david-res, 2020-11-22:
v7.8.0 is stated to be released on 20.10.2020 - Just thought it would be worth noting, to not confuse users and get us over excited about the upcoming enhancements and bug fixes ;)

embeddedt, 2020-11-22:
This file is behind... that's very strange. 7.7.2 was already released 5 days ago, and November 20th already passed, so that doesn't look like a typo. I'm not sure if the script tries to autogenerate the date but it does look like something messed up. I think December 1st would be the correct date here, so we'll go with that. :slightly_smiling_face:

closed 2020-11-22 by embeddedt

kisvegabor, 2020-11-23:
Thanks for the fix. 
I type the dates by hand. It seems only the order of number was incorrect. "20.10" is almost "12.01" :laughing:  

I've applied a further fix for dd.mm.yyyy format.
https://github.com/lvgl/lvgl/commit/68300a5dece1df202654b3950a14a82314047d28

embeddedt, 2020-11-23:
@kisvegabor I believe with your fix it is now scheduled for January 12, 2020 instead of December 1st, 2020. Could you double check?

kisvegabor, 2020-11-23:
Oh, you are right. I got confused with the order...
I've reverted your version.
https://github.com/lvgl/lvgl/commit/5848aa59e591a49724c708165c57a46d4cdd557f

commit 45599c2c (2020-11-22): Fix #1924: incorrect date on changelog

December 1st is two weeks from 7.7.2 release.
commit bf08ff44 (2020-12-01): Update Kconfig. (#1917)

* Kconfig: Add missing _HEX to LV_COLOR_TRANSP option.

* lv_conf_checker: Define LV_CONF_SKIP if CONFIG_LV_CONF_SKIP is defined.

* fix(indev): disabled object shouldn't absorb clicks but let the parent to be clicked

* Update README.md

* fix(arabic): support processing again alraedy processed texts with _lv_txt_ap_proc

* fix(textarea): support Arabic letter connections

fixes #1888

* Fix #1924: incorrect date on changelog

December 1st is two weeks from 7.7.2 release.

* Kconfig: Fix LV_CONF_MINIMAL related options.

* fix(arabic): support Arabic letter connection [... cut]