lvgl/lvgl #1925: Timers (tasks) clenup

puzrin, 2020-11-22:
Ref: https://github.com/lvgl/lvgl/issues/1839#issuecomment-722990641

IMO current "tasks" move to strange direction:

- Those are not tasks, those are periodic timers. Naming is confusing.
- Priorities are not effective for real needs and create illusion of correctness.

What is suggested:

- Recommend users to use protothreads for advanced things.
  - May be worth create "our lib", because i could not find "ideal" one.
- Rename "tasks" to "timers" everywhere.
- Drop priorities.

Changes are trivial, but breaking. Worth to sync with other big changes (land to v8 or v9).

Priority - not critical.

kisvegabor, 2020-11-23:
I definitely agree to simplify it. 
It's easy to do so it can even add it to v8. @embeddedt, do you agree?

embeddedt, 2020-11-23:
:+1: Sounds good; I don't rely on priorities when I make use of tasks and I doubt most other users do.

puzrin, 2020-11-23:
One more note. Point implicit in docs: timers should NOT run in MCU over-load mode. Such cases should be solved via protothreas (yield-ing) or preemptive RTOS.

kisvegabor, 2020-11-24:
I've done the update in `dev`.

puzrin, 2020-11-24:
IMO shorting timer to tmr is not needed here.

kisvegabor, 2020-11-24:
I was also thinking about it and thought tmr is a very conventional abbreviation so it's fine to use.

puzrin, 2020-11-24:
IMO:

- typing saving is about zero (not the same as background => bg)
- readability (for novices) not improved

kisvegabor, 2020-11-26:
IMO tmr is even more commonly written than timer in the embedded world but I think both are acceptable.

puzrin, 2020-11-26:
Always use full words in api if you can. Shorcuts are better for local variables.

kisvegabor, 2020-11-26:
Okay, so 1-1 :D 

@embeddedt Please, help us to decide. :slightly_smiling_face:

embeddedt, 2020-11-26:
I vote "timer". The savings are negligible and "timer" is more readable for newer users.

kisvegabor, 2020-11-27:
Alright, I renamed to timer.

puzrin, 2020-11-27:
LGTM.

`lv_timer_create` has `user_data`. Is it really need? Does anyone actively create/destroy temporary timers?

kisvegabor, 2020-11-27:
`user_data` is needed by the MicroPython binding. Adding it to the `create` function is not mandatory but task/therad creates usually have something like a user data:
- https://man7.org/linux/man-pages/man3/pthread_create.3.html
- https://www.freertos.org/a00125.html

A practical use case:  count on the label with a timer and pass the label as `user_data`.

puzrin, 2020-11-27:
ок.

commit 6bafb347 (2020-11-24): refactor(task): rename lv_task to lv_tmr and remove priorities

fixes #1925
closed 2020-11-27 by puzrin