lvgl/lvgl #1927: Compile error on `dev`

embeddedt, 2020-11-23:
https://github.com/lvgl/lvgl/blob/c556826d0b628a40aa5876ec104277248200d6fa/src/lv_core/lv_obj_style.c#L1514

@kisvegabor `_LV_OBJ_PART_SPACIAL_START` is not defined (there is no `SPECIAL` part either). I've been digging through the part definitions but I am not entirely sure which one is supposed to be used.

kisvegabor, 2020-11-23:
Thanks, I've fixed it.

Also updated the main.c and lv_componets.

embeddedt, 2020-11-23:
The core library now compiles; which branch of `lv_examples` should I be using? `dev` is giving compilation errors relating to message boxes and calendars.

kisvegabor, 2020-11-24:
`lv_examples` should be on `dev`. It works for me. What error do you see?

embeddedt, 2020-11-24:
Several.

```
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:33:12: error: unknown type name 'lv_calendar_date_t'
   33 |     static lv_calendar_date_t highlighted_days[3];       /*Only its pointer will be saved so should be static*/
      |            ^~~~~~~~~~~~~~~~~~
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:34:24: error: request for member 'year' in something not a structure or union
   34 |     highlighted_days[0].year = 2018;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:35:24: error: request for member 'month' in something not a structure or union
   35 |     highlighted_days[0].month = 10;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:36:24: error: request for member 'day' in something not a structure or union
   36 |     highlighted_days[0].day = 6;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:38:24: error: request for member 'year' in something not a structure or union
   38 |     highlighted_days[1].year = 2018;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:39:24: error: request for member 'month' in something not a structure or union
   39 |     highlighted_days[1].month = 10;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:40:24: error: request for member 'day' in something not a structure or union
   40 |     highlighted_days[1].day = 11;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:42:24: error: request for member 'year' in something not a structure or union
   42 |     highlighted_days[2].year = 2018;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:43:24: error: request for member 'month' in something not a structure or union
   43 |     highlighted_days[2].month = 11;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_calendar/lv_ex_calendar_1.c:44:24: error: request for member 'day' in something not a structure or union
   44 |     highlighted_days[2].day = 22;
      |                        ^
lv_examples/src/lv_ex_components/lv_ex_led/lv_ex_led_1.c: In function 'lv_ex_led_1':
lv_examples/src/lv_ex_components/lv_ex_led/lv_ex_led_1.c:7:24: warning: implicit declaration of function 'lv_led_create'; did you mean 'lv_arc_create'? [-Wimplicit-function-declaration]
    7 |     lv_obj_t * led1  = lv_led_create(lv_scr_act());
      |                        ^~~~~~~~~~~~~
      |                        lv_arc_create
lv_examples/src/lv_ex_components/lv_ex_led/lv_ex_led_1.c:7:24: warning: initialization of 'lv_obj_t *' {aka 'struct _lv_obj_t *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
lv_examples/src/lv_ex_components/lv_ex_led/lv_ex_led_1.c:9:5: warning: implicit declaration of function 'lv_led_off' [-Wimplicit-function-declaration]
    9 |     lv_led_off(led1);
      |     ^~~~~~~~~~
lv_examples/src/lv_ex_components/lv_ex_led/lv_ex_led_1.c:12:24: warning: initialization of 'lv_obj_t *' {aka 'struct _lv_obj_t *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
   12 |     lv_obj_t * led2  = lv_led_create(lv_scr_act());
      |                        ^~~~~~~~~~~~~
lv_examples/src/lv_ex_components/lv_ex_led/lv_ex_led_1.c:14:5: warning: implicit declaration of function 'lv_led_set_bright'; did you mean 'lv_obj_set_height'? [-Wimplicit-function-declaration]
   14 |     lv_led_set_bright(led2, 150);
      |     ^~~~~~~~~~~~~~~~~

kisvegabor, 2020-11-24:
Have you also added the `lv_compoents` repo? It's added as a submodule. `git submodule update` should solve it.

embeddedt, 2020-11-28:
Thanks. Updating that seemed to solve the problem and everything now compiles.

commit 77e7fe7f (2020-11-23): fix: remove old LV_OBJ_PART related defines

fix #1927
closed 2020-11-28 by embeddedt