lvgl/lvgl #1929: Arabic Keyboard

ashrafkamel5, 2020-11-25:
I just attempted a silly try to add an Arabic Keyboard.
but it failed very badly.
because I don't know where the drawing of the keyboard is.
I tried to change the characters of the upper case array to be like this 
```c
static const char * const default_kb_map_uc[] = {"1#", "ض", "ص", "ث", "ق", "ف", "غ", "ع", "ه", "خ", "ح","ج", LV_SYMBOL_BACKSPACE, "\n",
                                                 "ش", "س", "ي", "ب", "ل", "ا", "ت", "ن", "م", "ك", "ط",LV_SYMBOL_NEW_LINE, "\n",
                                                 "ذ", "ء", "ؤ", "ر", "ى", "ة", "و", "ز", "ظ", "د", "ز", "ظ", "د","\n",
                                                 LV_SYMBOL_CLOSE, LV_SYMBOL_LEFT, " ", LV_SYMBOL_RIGHT, LV_SYMBOL_OK, ""
                                                };
```
but the size of the keys was fixed in the drawing so, it looked like this 
[image]

at least the letters are connected correctly at the text input 😄 ❤️ 

====================
another question please,
I suffered alot with my current custom font.
it wasn't an open type and I changed it manually placing every symble using fontforge program.
and I have a missing character "ى" 
although I can find it in the font .c file 
[image]

but both of the letterforms cause a missing glyphs error, and I have no idea how to debug that ...

====================
and I can't find the newline symbol at Font Awesome to add to my custom font.
A couple of the symbols declared at the library are now missing actually from Font Awesome. 
I don't know why.

ashrafkamel5, 2020-11-25:
Here is a screenshot of my phone Arabic keyboard for your reference.

ashrafkamel5, 2020-11-25:
here is a screenshot of the google Arabic keyboard for your reference.
it is what I used in my try
[image]

embeddedt, 2020-11-25:
For the first issue you probably need to adjust the button widths [here](https://github.com/lvgl/lvgl/blob/9d0335f38cbda1a3023b4041dadbc60b4a72f4a0/src/lv_widgets/lv_keyboard.c#L56-L61). The numbers are the relative widths and the OR-ed values are [control flags](https://docs.lvgl.io/latest/en/html/widgets/btnmatrix.html#control-buttons). You can play around with the widths till you get the desired behavior.

kisvegabor, 2020-11-27:
> A couple of the symbols declared at the library are now missing actually from Font Awesome.

FontAwesome split the fonts into multiple files. I've joined them into a single [woff file](https://lvgl.io/assets/others/FontAwesome5-Solid+Brands+Regular.woff). It should contain New line too.

> I think we would need more than one keyboard layout for arabic.
to solve things like this

What are these arrows above the keyboard?
[image]

The "character form selector box" when you long-press a button is missing from the keyboard now. :(

ashrafkamel5, 2020-12-12:
@kisvegabor 
sorry for being so late.
many thanks for the fontAwesome file.
Now it worked fine <3.
and this is my modification for the keyboard.
[lv_keyboard.zip](https://github.com/lvgl/lvgl/files/5683922/lv_keyboard.zip)

I have added an Arabic button matrix
```c
static const char * const default_kb_map_ar[] =
{
    "1#", "ض", "ص", "ث", "ق", "ف", "غ", "ع", "ه","خ", "ح","ج", "\n",
    "ش", "س", "ي", "ب", "ل", "ا", "ت", "ن", "م", "ك", "ط", LV_SYMBOL_BACKSPACE, "\n",
    "ذ", "ء", "ؤ", "ر", "ى", "ة", "و", "ز", "ظ", "د", "ز", "ظ", "د","\n",
    LV_SYMBOL_CLOSE,"abc", LV_SYMBOL_LEFT, " ", LV_SYMBOL_RIGHT,LV_SYMBOL_NEW_LINE,LV_SYMBOL_OK, ""
};

static const lv_btnmatrix_ctrl_t default_kb_ctrl_ar_map[] = {
    LV_KEYBOARD_CTRL_BTN_FLAGS | 5, 4, 4, 4, 4, 4, 4, 4, 4, 4, 4, 4,
    4, 4, 4, 4, 4, 4, 4, 4, 4, 4, 4, 7,
    1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
    LV_KEYBOARD_CTRL_BTN_FLAGS | 2,LV_KEYBOARD_CTRL_BTN_FLAGS | 2, 2, 6, 2, 3, LV_KEYBOARD_CTRL_BTN_FLAGS | 2
};
```
[image]

I think this layout is fine.
and I have added a "أب" button to the reset of the keyboards 
and added `LV_KEYBOARD_MODE_TEXT_ARABIC` to the lv_keyboard_mode_t enum
### example code
```c
            kb = lv_keyboard_create(lv_scr_act(), NULL);
            lv_obj_set_event_cb(kb, kb_event_cb);
            lv_keyboard_set_mode(kb, LV_KEYBOARD_MODE_TEXT_ARABIC);
            lv_indev_wait_release(lv_indev_get_act());
```

**future work** would be to find a way around to represent a solid Alef hamza letter forms "أ, إ, ا, آ"
in other words, mimicking the shift key in real keyboard or long pressing in google touch keyboard

and the other forms of the other letters but the Alef is the most important and the used one
=============================
Sadly, I still don't know why the "ى" letter doesn't work with my font and gives a missing glyph error. 
"and the problem reproduced with the same letter in other Arabic fonts"
if you can refer me to a place where I could learn to debug this problem would be great.
it is fine, the project would survive without it.
but it is important to me.
============================
waiting for your feedback <3

kisvegabor, 2020-12-15:
Thank you @ashrafkamel5. I like it!

Can you send a Pull Request to the `dev` branch?

Regarding the `ى` character. I've tested with this code, and it works well for me:
```c
  lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
  lv_obj_set_style_local_text_font(label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, &lv_font_dejavu_16_persian_hebrew);
  lv_label_set_text(label, "ى");
```
How did you generate your font? Can you send a screenshot or the commands you used?

ashrafkamel5, 2020-12-15:
@kisvegabor 
Ok, I will.
as for the font
I know it works fine with the dejavu font.
but it doesn't in mine and some others.

[image]

this is a screen from the font.c file showing that the font has the 'ى' forms converted but yet gives an error when using them.
here are the font files
[ReemKufi.zip](https://github.com/lvgl/lvgl/files/5695446/ReemKufi.zip)
if you have the time to investigate the issue, It would be very much appreciated. <3
I have used the online converter with these ranges.
`0x20-0x7f,0x600-0x6FF,0xFE70-0xFEFF` + the awsome font file
and it works fine. except for this letter as you can see in my keyboard screenshot, it is the only empty key.

ashrafkamel5, 2020-12-15:
I think I knew the problem ...
omg :D 
it is in the `lv_txt_ap.c` file

should I include this fix in the pull request?
or make an issue for it.

ashrafkamel5, 2020-12-15:
and it is fixed :D 
[image]

hhhhhhh
it is weird how effective talking out loud a problem to someone else is
it is a small bug in the file as I said
I will make a bug issue now

ashrafkamel5, 2020-12-15:
ok I jinxed myself
other problem presented 
I will get back at the new issue
many thanks Mr Gabor

kisvegabor, 2020-12-16:
Hm, I wonder how it worked with the devaju font then. :laughing: 
Anyway, it's great to hear that you have font the issue! 

Please don't forget the PR with Arabic layout if you have time for it :slightly_smiling_face:

ashrafkamel5, 2020-12-19:
I did the PR ❤️

closed 2020-12-19 by ashrafkamel5