lvgl/lvgl #1933: Scrollbars may ignore parent visibility?

embeddedt, 2020-11-25:
**Describe the bug**

A list's scrollbar seems to be visible even if the list is partially transparent/covered by another object.

**To Reproduce**

Go to https://lvgl.github.io/lv_examples/lv_demo_printer/index.html (or run `lv_demo_printer`). Click on Print. Click on USB.

Note that the list's scrollbar appears instantly, long before the animation has finished.

**Expected behavior**

The scrollbar should fade/animate in like the rest of the list.

**Screenshots**

[image]

**Additional context**

I haven't looked into the code yet, but maybe the scrollbar is not taking the list's opacity into account?

kisvegabor, 2020-11-26:
Thanks, I've just pushed a fix.

closed 2020-11-26 by kisvegabor

embeddedt, 2020-11-27:
Great!

This may be a similar issue: https://forum.lvgl.io/t/bug-of-table-click-to-change-style/3842

commit 313f7f30 (2020-11-26): fix(style): invalidate cache on every PART

Inherited properties on parts other than MAIN  needs to be recached on property change beacuse they
are inherted from MAIN

fixes #1933