lvgl/lvgl #1934: Keyboard not opening when touching text area

engr-mujahidabdullah, 2020-11-25:
I am working on lvgl keyboard, On pressing "X" key keyboard gona hide and it did not appear ore on pressing txt area.
Kindly guide me in this regard.

embeddedt, 2020-11-25:
By default, the keyboard is deleted when the X button is pressed.

You will need to show the keyboard again manually - the library does not manage showing/hiding it automatically. Or, you can set a custom event handler and override the button behavior as you need.

engr-mujahidabdullah, 2020-11-25:
> manage showing/hiding it automatically. You could also set a custom event

i shall be very thankful to you if you give me some example of it.

kisvegabor, 2020-11-27:
I've updated the keyboard example:
https://github.com/lvgl/lv_examples/blob/master/src/lv_ex_widgets/lv_ex_keyboard/lv_ex_keyboard_1.c

engr-mujahidabdullah, 2020-11-27:
Thank you all,
The other thing is that how i can get data when i press Ok button. Untill
now i set event callback to keyboard when I press OK button then i want to
read data from text box. by calling function lv_textarea_get_text. But i
did not return any thing. Kindly address this issue.

kisvegabor, 2020-12-21:
Sorry, for the late reply, I didn't notice your last comment. 
If you still have questions about how to use the keyboard and the text area please open a topic in the forum.
https://forum.lvgl.io/

closed 2020-12-21 by kisvegabor