lvgl/lvgl #1936: Segfault when using LV_FIT_MAX

KammutierSpule, 2020-11-26:
**Describe the bug**

I'm creating a simple test with simulation environment.
In the next code if I use LV_FIT_MAX it will seg fault at
lv_obj.c line 3994
`else if(sign == LV_SIGNAL_GET_TYPE) return lv_obj_handle_get_type_signal(param, LV_OBJX_NAME);`

if I change LV_FIT_MAX to LV_FIT_TIGHT it works.

**To Reproduce**

```
    lv_obj_t * mc = lv_cont_create(lv_scr_act(), NULL);

    lv_obj_set_pos(mc, 0, 0);
    lv_obj_set_size(mc, LV_HOR_RES_MAX, LV_VER_RES_MAX);

    lv_cont_set_fit(mc, LV_FIT_TIGHT);
    lv_cont_set_layout(mc, LV_LAYOUT_COLUMN_MID);

    lv_obj_t * label = lv_label_create( mc, NULL );
    lv_label_set_text( label, "T1" );

    lv_obj_t *f = lv_cont_create(mc, NULL);
    lv_cont_set_fit(f, LV_FIT_MAX);
    lv_cont_set_layout(f, LV_LAYOUT_COLUMN_MID);

    lv_obj_t *f2 = lv_cont_create(f, NULL);
    lv_cont_set_fit(f2, LV_FIT_TIGHT);
    lv_cont_set_layout(f2, LV_LAYOUT_ROW_MID);

    lv_obj_t * label1 = lv_label_create(f2, NULL );
    lv_label_set_text(label1, "L1" );

```
**Expected behavior**

Expect to not crash

**Screenshots**

**Additional context**

Using v7.7.2

embeddedt, 2020-11-26:
It could be stack overflow. There are certain combinations of container layouts that are difficult to detect beforehand and [result in infinite loops](https://github.com/lvgl/lvgl/issues/1539#issuecomment-633520731).

Can you check the backtrace and see if there is a long line of repeated functions?

If this is the case, perhaps a workaround for 7.x is to simply set a limit on how deeply nested container layout calls can be. For example, if the function detects that it has been called for the fourth or fifth time, it either aborts or just gives up on the layout after printing a warning to the log.

In 8.x we are moving towards and will be encouraging the use of flexbox and grid layouts which should not have this issue.

KammutierSpule, 2020-11-26:
Is there some way to avoid this? Use less containers?

It is crashing at accessing the function parameters, so yeah, my suspicious is the stack:

[image]

embeddedt, 2020-11-26:
That stacktrace looks exactly like the problem. The only workaround I know of at the moment is to choose layouts which don't try to compete with each other. For example in @kisvegabor's post above he explained how `LV_FIT_PARENT` tries to position the text area to fill the parent's left and right edges, while `LV_LAYOUT_ROW_MID` tries to align it to the left at the same time. `LV_FIT_MAX` may have a similar problem.

kisvegabor, 2020-12-20:
As we can't fix it in v7 I close this issue. The workaround is to not use this kind of fit auto size together.

FYI, the new layouts in v8 won't suffer from such issues.

closed 2020-12-27 by embeddedt

embeddedt, 2020-12-27:
@kisvegabor I suspect quite a few more people are going to run into this issue. What do you think about my [suggested workaround](https://github.com/lvgl/lvgl/issues/1936#issuecomment-734283881) above? It definitely doesn't solve the problem, but it would probably save hours of debugging.

kisvegabor, 2020-12-30:
@embeddedt 
I'm ok with that. Do you have an idea about how to implement this?

embeddedt, 2020-12-30:
Thanks to the fully event-driven nature of LVGL I think a counter variable tracking the recursion level of `lv_cont_refr_autofit` should suffice. How many nested calls need to be allowed?

liebman, 2020-12-30:
Maybe configurable in `Kconfig`?

kisvegabor, 2020-12-31:
> Thanks to the fully event-driven nature of LVGL I think a counter variable tracking the recursion level of lv_cont_refr_autofit should suffice. How many nested calls need to be allowed?

Oh really, I was thinking about when to clear the counter, but with recursive functions, it's obvious: increment in the beginning and decrement in the end.

> Maybe configurable in Kconfig?

I think it's a too specific config to make it directly available for the user. I can imagine something like this in `lv_cont.c`:
```c
#ifndef LV_LAYOUT_MAX_DEEPNESS
#define LV_LAYOUT_MAX_DEEPNESS 10
#endif
```

So the user still overwrites it if he wants.

embeddedt, 2020-12-31:
I'm thinking `MAX_RECURSION` would be a better term, but otherwise that sounds good. I will use 10 as a default.

kisvegabor, 2020-12-31:
> I'm thinking MAX_RECURSION would be a better term

I agree,

embeddedt, 2020-12-31:
See #1986.