lvgl/lvgl #1937: esp32 IRAM usage should be optional and not set in CMakeLists.txt

liebman, 2020-11-26:
**Describe the bug**

`IRAM` space on esp32 is limited and LVGL as component in esp-idf puts many drawing routines there due to this in `CMakeLists.txt`:
```
target_compile_definitions(${COMPONENT_LIB} PUBLIC "-DLV_ATTRIBUTE_FAST_MEM=IRAM_ATTR")
```

This should be an opting in `Kconfig`

**To Reproduce**

Have code that uses some of the `IRAM` space already, compile and:
```
/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 11376 bytes
```

**Expected behavior**

LVGL use of IRAM for anything except functions expected to be called from interrupt services should be optional.

C47D, 2020-11-26:
I can add an option to use IRAM or not later this week.

C47D, 2020-11-27:
@liebman can you check if #1939 fixes the issue for you?

liebman, 2020-11-27:
@C47D yes it does indeed! 💯

PR #1939 (2020-11-27): ESP-IDF: Let the user set or disable IRAM as LV_ATTRIBUTE_FAST_MEM.
Tries to fix #1937 , against master because it fixes an compilation error.
closed 2020-11-27 by embeddedt