lvgl/lvgl #1942: Setting lv_textarea_set_pwd_mode to true does not update the textarea

fvanroie, 2020-11-29:
**Describe the bug**

I experience an issue when setting `lv_textarea_set_pwd_mode` to `true`. The displayed characters are not replaced by bullets right away. Instead, the bullets only appear when you enter or delete a character.

**Screenshots**

[image]

In this example, the characters are not hiden until a character ( e.g. `g`) is entered.

**To Reproduce**

I have implemented the 'Password Peek' button like this:

```c
static void peek_password_cb(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_VALUE_CHANGED) {
        bool stat=lv_btn_get_state(obj);
        lv_obj_set_style_local_value_str(obj, LV_BTN_PART_MAIN, LV_STATE_DEFAULT,
                                         stat ? LV_SYMBOL_EYE_OPEN : LV_SYMBOL_EYE_CLOSE);
        lv_textarea_set_pwd_mode(pwd_ta, !stat);
    }
}
```

**Expected behavior**

Immediately update the textarea with bullets when calling `lv_textarea_set_pwd_mode(pwd_ta,true);`

**Additional context**

I found that adding `pwd_char_hider(ta)` to the following code in [textarea.c](https://github.com/lvgl/lvgl/blob/b1a7dbadcdf34c5a26f90a02ae7103c3a68a5d34/src/lv_widgets/lv_textarea.c#L740) fixes the issue:

```c
    ext->pwd_mode = en == false ? 0 : 1;
    if(en)pwd_char_hider(ta);    // immediately hide the characters
```

Now I see the correct behaviour:
[image]

embeddedt, 2020-11-29:
I believe this is what [this block of code](https://github.com/lvgl/lvgl/blob/b1a7dbadcdf34c5a26f90a02ae7103c3a68a5d34/src/lv_widgets/lv_textarea.c#L719-L731) is supposed to be doing.

fvanroie, 2020-11-29:
Oh I see. Now I understand the intent of that part. But it seems to have no effect.

So, I set a breakpoint and stepped through that part of the code.

`pwd_char_hider(ta);` is indeed being called, but it returns without effect because `ext->pwd_mode` has not been updated at that point in time. If I add the second call after the update, it updates the textarea.

embeddedt, 2020-11-29:
Okay, so it looks like what we should be doing is storing the previous value of `ext->pwd_mode`, then changing it, and finally going through that if statement. Do you want to make a PR with that change or would you like me to just apply it to `master` (credit will be given of course)?

fvanroie, 2020-11-29:
I can create a PR, but for a one-liner it might be a lot of overhead... Feel free to update, no worries about credits.

embeddedt, 2020-11-29:
Alright; thank you very much for the report and the investigation!

closed 2020-11-29 by embeddedt

fvanroie, 2020-11-29:
Great, I just tested the commit and confirm it's working as expected!

commit bc2ca71b (2020-11-29): Fix #1942: lv_textarea_set_pwd_mode not changing to bullets

Thanks to @fvanroie for investigating.