lvgl/lvgl #1945: Build LVGL without IDE

embexus, 2020-12-02:
I would like to integrate lvgl in my automated build system (Yocto) to build for ARM Linux.

I can see that cmake is supported but there is no hint how to configure and build ?

can you please provide more info ?

embeddedt, 2020-12-02:
All you need to do is add LVGL to your include path and compile the C files
in src. An IDE is not necessary.

Responding on mobile right now so let me know if that's unclear and I will
do my best to elaborate.

embexus, 2020-12-02:
@embeddedt please elaborate when you have chance, I didn't get what you mean!

embeddedt, 2020-12-02:
https://docs.lvgl.io/latest/en/html/get-started/quick-overview.html#add-lvgl-into-your-project

There is no need to do any configuration; you just need to set up your build system to compile the library's C files.

embexus, 2020-12-02:
@embeddedt what do you mean with 'set up your build system to compile the library's C files' ?!
actually this was my question !

embeddedt, 2020-12-02:
> 'set up your build system to compile the library's C files' ?!

I am not familiar with Yocto; you will need to research how it handles things. Generally you have an existing application composed of C files and then you add LVGL to that list. Do you not have an application already? That would change things.

embeddedt, 2020-12-02:
You can also take a look at the [Linux framebuffer](https://github.com/lvgl/lv_port_linux_frame_buffer) project; it is built using Make. That might be easier to start with.

embexus, 2020-12-02:
@embeddedt thanks the project Linux framebuffer is building and work fine for me.

Maybe lvgl should also cmake or makefiles ? and generate static/shared library and header to be easily integrated with Projects ?

embeddedt, 2020-12-02:
We prefer not to target one build system, as LVGL can be used in lots of scenarios and the current setup works very well for most users.

ValentiWorkLearning, 2020-12-11:
@embexus  indeed, it's nice to have a common way for simplifying the integration of LVGL-based targets into existing projects. But there are several modern buidsystems for project. Which of them should be supported?
A short list can be:
CMake
meson
Add LVGL to Conan and vcpkg package managers
What else?

embexus, 2020-12-12:
@ValentiWorkLearning cmake its widely used

closed 2021-01-25 by stale[bot]

rzr, 2021-08-25:
Hi is there any ongoing ffort to support bitbake for cross compiling it ?

I didn't find any recipe at:
http://layers.openembedded.org/layerindex/branch/master/recipes/?q=lvgl

kisvegabor, 2021-08-25:
I don't know bitbake. Form [Wikipedia](https://en.wikipedia.org/wiki/BitBake) it's not clear what it really does.

How it's related to https://layers.openembedded.org/