lvgl/lvgl #1947: undefined reference to `lv_font_montserrat_12_compr_az' If i run lv_demo_benchmark

Trackhe, 2020-12-02:
**Describe the bug**

libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.txt_small_compr_cb+0x0): undefined reference to `lv_font_montserrat_12_compr_az'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.txt_medium_compr_cb+0x0): undefined reference to `lv_font_montserrat_16_compr_az'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.txt_large_compr_cb+0x0): undefined reference to `lv_font_montserrat_28_compr_az'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.img_rgb_cb+0x0): undefined reference to `img_cogwheel_rgb'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.img_argb_cb+0x0): undefined reference to `img_cogwheel_argb'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.img_ckey_cb+0x0): undefined reference to `img_cogwheel_chroma_keyed'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.img_index_cb+0x0): undefined reference to `img_cogwheel_indexed16'
libraries/lv_examples/lv_demo_benchmark/lv_demo_benchmark.c.o:(.literal.img_alpha_cb+0x0): undefined reference to `img_cogwheel_alpha16'

To Reproduce

Build with Arduino IDE. And i have installed lvgl and lvgl_examples via library manager.

Expected behavior

working demo

kisvegabor, 2020-12-04:
Hi,

I've just pushed a fix to `lv_examples`.

Trackhe, 2020-12-04:
working

closed 2020-12-04 by Trackhe

el-samiyel, 2021-12-26:
Same issue here as of today.. 

`LVGL_Arduino.ino:122: undefined reference to `lv_demo_widgets'
collect2.exe: error: ld returned 1 exit status
Error compiling for board Teensy 4.0.
`

phamvant, 2022-01-09:
I got the same issue even though i did exaclty the same thing in the docs 
May PlatformIO cause the issue?

el-samiyel, 2022-01-10:
> Recently we made some changes to make the examples and demos easier to use
> with Arduino. See the updated docs:
> https://docs.lvgl.io/master/get-started/arduino.html

Even with the updated documents, the errors I posted above still exists. I
can’t compile not a single example...

kisvegabor, 2022-01-10:
Have you copied the `demo` folder to `src`?
If so please send a screenshot about the directory layout.

phamvant, 2022-01-10:
> Have you copied the `demo` folder to `src`? If so please send a screenshot about the directory layout.

[image]

I already did

kisvegabor, 2022-01-10:
Ok, looks good. Have you enabled `LV_USE_DEMO_WIDGET`s in `lv_conf.h`?

phamvant, 2022-01-10:
> Ok, looks good. Have you enabled `LV_USE_DEMO_WIDGET`s in `lv_conf.h`?

Thank you it compiled. But it doesn't show any thing
[image]

I can only run simple label
[image]

phamvant, 2022-01-10:
I also can run benchmark

kisvegabor, 2022-01-10:
Probably `LV_MEM_SIZE` is too small.

phamvant, 2022-01-11:
> Probably `LV_MEM_SIZE` is too small.

Thank you so much you my hero

kisvegabor, 2022-01-11:
Cool! It seems everything is working now. :slightly_smiling_face:

C47D, 2022-01-11:
Documenting the fix in a troubleshooting section of the docs might be a good idea, what do you think?

kisvegabor, 2022-01-15:
@C47D 
I think we should rather add some comments in `lv_conf.h` when the demos are enabled because the user surely goes there and will see the notes.

I don't know how many people reads it but we have an [FAQ](https://docs.lvgl.io/master/intro/index.html#faq) part in the docs. There we can mention that if LVGL doesn't what you can try. 

Probably we can make the FAQ page more visible by adding links to in in issue and topic templates.

What do you think?

C47D, 2022-01-15:
Sounds like a good idea!

kisvegabor, 2022-01-15:
All done. 
I added [this section]()https://docs.lvgl.io/master/intro/index.html#lvgl-doesn-t-start-nothing-is-drawn-on-the-display-what-can-be-the-problem to the docs, linked the FAQ in the bug report issue template on GitHub, in the how-to and get started forum category's topic templates, and added comment to lv_conf.h. (86525c35b)

phamvant, 2022-01-17:
Hi, can i just ask one more question

I want to use Edgeline so i downgrade to lvgl v7. I tried to run demos in lv_examples lib but it seems like freeze at the first frame (I tried benchmark and there is no animation). What should i do?

kisvegabor, 2022-01-19:
It's independent from EdgeLine, right?

Form this info I can't tell much. Please enable logging to see what happens.