lvgl/lvgl #1951: lv_style.c:1146:11: error: The address of local variable 'prop' is accessed at non-zero index. [objectIndex]

Harvie, 2020-12-05:
Hello,
i've run lvgl through cppcheck and it found some errors:

```
$ cppcheck . >/dev/null
src/lv_core/lv_style.c:1146:11: error: The address of local variable 'prop' is accessed at non-zero index. [objectIndex]
    prop_p[1] = style->map[idx + 1];
          ^
src/lv_core/lv_style.c:1144:35: note: Address of variable taken here.
    uint8_t * prop_p = (uint8_t *)&prop;
                                  ^
src/lv_core/lv_style.c:1146:11: note: The address of local variable 'prop' is accessed at non-zero index.
    prop_p[1] = style->map[idx + 1];
          ^
src/lv_gpu/lv_gpu_nxp_pxp.c:59:0: error: Syntax error in #elif [preprocessorErrorDirective]
#elif
^
src/lv_conf_kconfig.h:14:0: error: No header in #include [preprocessorErrorDirective]
#  include LV_CONF_KCONFIG_EXTERNAL_INCLUDE
^
```

embeddedt, 2020-12-06:
Thanks for the reports. The `lv_gpu_nxp_pxp` one is a bug and has been fixed.

I think the other two are expected behavior. For Kconfig we rely on including a header based on an external define, which presumably `cppcheck` cannot detect. The style code is also safe as it is indexing into a [`uint16_t`](https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_style.h#L203) using a [`uint8_t`](https://github.com/lvgl/lvgl/blob/ba5ee315e9bf56284fddb8de46cc2fc1a8f4d106/src/lv_core/lv_style.c#L1144) array.

If you know of a way to silence cppcheck on those, feel free to send a pull request that adds the necessary comment.

embeddedt, 2020-12-28:
For now I will close this issue as the remaining issues appear to just be false positives. A PR is still welcome. :slightly_smiling_face:

closed 2020-12-28 by embeddedt