lvgl/lvgl #1954: lv_img issue, image glitching after swapping in 2nd and 3rd image

murtuzaq, 2020-12-07:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

The bug:
image glitching each time I repaint over an existing image using lv_img_set_src(...).

**To Reproduce**
This bug is reproducible on the PC simulator (VS studio), on the current master branch of the code (v7.8.0).  This code snippet was condensed into Main, for easy duplication for community members to reproduce.  

`int main(int argc, char** argv)
{
    /*Initialize LittlevGL*/
    lv_init();

    /*Initialize the HAL for LittlevGL*/
    hal_init();

   
	static uint32_t img_index = 0;
	lv_img_dsc_t img_bkg;
	lv_obj_t* obj_img_bkg;
	uint32_t width;
	uint32_t height;
	uint32_t file_size;

	lv_obj_t* lv_obj_scrn = lv_obj_create(NULL, NULL);

	lv_scr_load(lv_obj_scrn);

	LV_IMG_DECLARE(ico_ClosLid_raw);  // png image file with raw + alpha 
	LV_IMG_DECLARE(ico_RubbGask_raw); //png image file with raw + alpha 

	const int MAX_IMAGE_INDEX = 2;
	static const lv_img_dsc_t* swap_img_dsc[] =
	{
		&ico_ClosLid_raw,
		&ico_RubbGask_raw,
	};

    while (1) 
	{
		printf("======================= image = %d \n", img_index);
		const uint8_t* ptr_file_img = swap_img_dsc[img_index]->data;
		file_size = swap_img_dsc[img_index]->data_size;
		

		lodepng_decode32((uint8_t**)(&img_bkg.data), &width, &height, ptr_file_img, file_size);

		img_bkg.header.w = width;
		img_bkg.header.h = height;
		img_bkg.header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA;
		img_bkg.header.always_zero = 0;

		//============= REARRANGE THE BITS AFTER THE PNG DECODER====================
		//---------raw data is 32 bit, convert down to 24 bit for decoding;
		//                     0         1      2       3
		//4 BYTES = 32 BItS; [RED ] [ GREEN] [BLUE] [ALPHA];   //lodepng output
		//                        :5      :6      :5     :8
		//565 + ALPHA = 24 BITS [RED ] [GREEN] [ BLUE] [ALPHA]  //expected rgba result!

		uint32_t      i;
		for (i = 0; i < width * height; i++)
		{
			uint8_t *img = img_bkg.data;

			uint8_t r = img[i * 4];
			uint8_t g = img[i * 4 + 1];
			uint8_t b = img[i * 4 + 2];
			uint8_t a = img[i * 4 + 3];

			uint32_t idx = 3 * i;

			img[idx] = (r & 0xF8) + ((g & 0xE0) >> 5);				    //get Red + Grn High;  
			img[idx + 1] = ((g & 0x1C) << 3) + ((b >> 3) & 0x1F);   	// get Grn Low + Blue
			img[idx + 2] = a;											//get alpha
		}
		img_bkg.data_size = 3 * img_bkg.header.w * img_bkg.header.h;
		//====================END OF IMG DESCRIPTOR DEFINITION=============
		
		obj_img_bkg = lv_img_create(lv_obj_scrn, NULL);
		lv_img_set_src(obj_img_bkg, &img_bkg);        /*Set the created file as image*/
		lv_obj_align(obj_img_bkg, NULL, LV_ALIGN_CENTER, 0, 0);

		lv_task_handler();

		if (img_index++ >= MAX_IMAGE_INDEX)
		{
			img_index = 0;
		}

		free((uint8_t*)img_bkg.data);
		lv_obj_del(obj_img_bkg);

        Sleep(5000);       /*swap a new image every second */
    }

    return 0;  //will never get here;
}`

**Expected behavior**

What i expected to happen was to rotate through a series of image files on the screen.  What i end up getting, is the first image file paints properly, but every image file afterwards get painted incorrectly or glitches in some way.     

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Additional context**
A similar piece of code to this worked correctly in v5.3.

embeddedt, 2020-12-07:
Newer versions of LVGL have an image cache - perhaps try [cleaning the cache](https://docs.lvgl.io/latest/en/html/overview/image.html#clean-the-cache) for `&img_bkg`.

murtuzaq, 2020-12-08:
Thank you, i have retried on the both the simulator and our embedded platform, and yes, this does correct the image from glitching.

So i suppose i wouldn't have needed to call lv_img_cache_invalidate_src if i was using the lv_img_decorder? In the lv_conf file, i see we must use a CACHE size of atleast 1 or greater.  Would like to have set this number to 0, so that the cache wouldn't take place.  With this cache however, i suppose defining lv_img_dsc_t  manually outside of the lv_img_decoder will use up extra RAM space without helping the application, is that correct?  

Some context: The reason we originally stayed away from using the lv_img_decoder was because we didnt want any of our images to be cached.

kisvegabor, 2020-12-10:
I've just added an option to the `dev` branch to disable caching by setting `LV_IMG_CACHE_DEF_SIZE 0`.

Can you try it?

closed 2021-01-25 by stale[bot]