lvgl/lvgl #1956: borders on polygon don not work

uraich, 2020-12-10:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Borders on polygons do not work correctly

**Reproduce**

Draw a polygon on a canvas with borders. In my case: blue background, white border
I uploaded the code to 
https://github.com/uraich/lv_mpy_examples/blob/main/st7735_demos/test_polygon.py

**Expected behaviour**

A polygon filled with the background colour and surrounded with a border

**Screenshots**
[image]

**Additional context**

Only horizontal and vertical border lines appear.

embeddedt, 2020-12-10:
I believe it's caused by the fact that polygons are drawn by masking corners off of a rectangle. Thus, the borders are missing on those corners.

I'm not sure it's a trivial issue to solve, but let's wait for @kisvegabor's comment. In the meantime, a workaround might be to draw the border yourself using lines. It's a very ugly workaround, but the best I can think of at the moment.

kisvegabor, 2020-12-11:
> I believe it's caused by the fact that polygons are drawn by masking corners off of a rectangle. Thus, the borders are missing on those corners.

Exactly.

> a workaround might be to draw the border yourself using lines. It's a very ugly workaround, but the best I can think of at the moment.

Yes, it's a possible workaround. I was thinking about creating a `lv_polygon` widget derived from`lv_line` that draws the border from a line and fills the backgrond as a polygon.

The only issue is the border will be placed symmetrically on the sides. E.g. with  a 2px border, 1 px inside, 1 px outside.

uraich, 2020-12-12:
Why don't you do it the the other way round: First draw the polygon and then the border on top of it?

kisvegabor, 2020-12-15:
> Why don't you do it the the other way round: First draw the polygon and then the border on top of it?

I meant this way, but it stills makes the polygon larger. And if the border has 50% opacity, on the half line width the polygon will be visible, and on the other the parent

uraich, 2020-12-15:
I though that the border would only be drawn only on the inside of the polygon, not changing the polygon size.

kisvegabor, 2020-12-20:
It'd require transformation on the coordinates of the polygon. That is to calculate a smaller polygon from the original and draw to border on it. It should be "only" scaling down every vertex relative to the center. However, the rounding error could introduce color beeling and other visual issues.

kisvegabor, 2021-01-18:
I've opened a feature request topic in the Forum to make the idea more visible:
https://forum.lvgl.io/t/border-on-polygon/4310

closed 2021-01-18 by kisvegabor