lvgl/lvgl #1958: Missing micropython binding of lv_calendar.get_day_of_week().

rockindy, 2020-12-14:
Hi,

I want to get the day of week from lv_calendar in micropython binding, but can't find the binding of lv_calendar.get_day_of_week(). Is this missed or intended?

Thanks

kisvegabor, 2020-12-15:
Hi,

is not a public API function.  I C it look like this. I think you can almost copy-paste it to MicroPython:
```c
static uint8_t get_day_of_week(uint32_t year, uint32_t month, uint32_t day)
{
    uint32_t a = month < 3 ? 1 : 0;
    uint32_t b = year - a;

#if LV_CALENDAR_WEEK_STARTS_MONDAY
    uint32_t day_of_week = (day + (31 * (month - 2 + 12 * a) / 12) + b + (b / 4) - (b / 100) + (b / 400) - 1) % 7;
#else
    uint32_t day_of_week = (day + (31 * (month - 2 + 12 * a) / 12) + b + (b / 4) - (b / 100) + (b / 400)) % 7;
#endif

    return day_of_week;
}
```

`month` is in 1..12 `day` is in 1..31 range.

rockindy, 2020-12-16:
Hi,

Because the Micropython RTC class needs the day of week to set time, I think it is helpful to let `get_day_of_week()` be a public API.

kisvegabor, 2020-12-20:
Alright, actually it won't harm anything. I added it as `lv_calendar_get_day_of_week()` in `master`.

closed 2020-12-20 by kisvegabor