lvgl/lvgl #1959: letter ى Arabic processing error

ashrafkamel5, 2020-12-15:
**Describe the bug**
when Having an Arabic font that covers only the Arabic Presentation Forms-B.
the letter ى produces a missing glyph error.
**explain**
I found an error in the table of Unicode at `ap_chars_map[]` at ` lv_txt_ap.c `file
**fix**
at line 72
add this
`    {39, 0xFEF0, 0, 0, -1,  {1, 0}},   // ى`

I didn't want to delete the line existing at 72. because even if I think it is a mistake and the Unicode there is a duplicate of the one at 74
but yet. it doesn't harm being there. 
and maybe it works fine with other forms for Persian writing, I don't know we should ask Mr. Hamid who wrote the file.
so just adding that line fixes it.
**Screenshots**
look at the keyboard and the missing letter.
before
[image]

after
[image]

kisvegabor, 2020-12-16:
Thank you!
@hamidrm, could you take a look at this?

hamidrm, 2020-12-18:
@ashrafkamel5 You are right! It was my mistake. I thought the glyph of 0x6CC and 0x649 are the same (Although, In the Persian, they are the same!). According to https://en.wikipedia.org/wiki/Arabic_script_in_Unicode, you are right.
So, we should replace your change (`{39, 0xFEF0, 0, 0, -1, {1, 0}}, //ی`), with current line 72.
Thank you!
[image]

ashrafkamel5, 2020-12-18:
It's great adding full Arabic support to the library

@kisvegabor there is a small bug at dropdown widget
`lv_dropdown_add_option`
Doesn't process Arabic letters.
You need to add the `lv_txt_ap_proc` there

ashrafkamel5, 2020-12-19:
@kisvegabor 
and the window's name also needs the same fix 😆 
`lv_win_set_title`
it would be repetitive if I did an issue for both of them 
hope you just fix them ❤️ 
really hyped for the upcoming release!

kisvegabor, 2020-12-20:
@ashrafkamel5 
I've fixed both the dropdown and window, thanks for the report. 

@hamidrm or @ashrafkamel5 
Please send a PR to fix that Arabic letter.

commit cee779a5 (2020-12-23): Fix #1959: wrong glyph data
closed 2020-12-23 by embeddedt