lvgl/lvgl #1960: Heap buffer overflow - password mode

Qhesz, 2020-12-15:
**Describe the bug**

Adding unicode chars (with more than one byte in utf-8) to a password mode text area leads to buffer overrun and memory corruption.

The following line in lv_textarea_add_char assumes that the char is one byte long:
```
ext->pwd_tmp = lv_mem_realloc(ext->pwd_tmp, strlen(ext->pwd_tmp) + 2);  /*+2: the new char + \0 */
```

It should be something more like:
```
ext->pwd_tmp = lv_mem_realloc(ext->pwd_tmp, strlen(ext->pwd_tmp) + strlen(letter_buf) + 1);
```

**To Reproduce**

Add a 3-byte character (such as LV_SYMBOL_SETTINGS) into an empty password mode text area 3 times (on a 32-bit system, using the builtin heap). Observe the buffer overrun corrupting the next header entry in the heap.

**Expected behavior**

lv_textarea_add_char doesn't cause memory corruption.

**Screenshots**

**Additional context**

Add any other context about the problem here.

kisvegabor, 2020-12-20:
Hi,

Sorry for the late reply. Somehow I haven't noticed this issue. :slightly_frowning_face: 

I've just fixed it.

closed 2020-12-20 by kisvegabor

embeddedt, 2020-12-20:
It must have been a GitHub bug - I got no email notification about this one either.

kisvegabor, 2020-12-21:
It's only me who feels that GitHub is less stable since Microsoft acquired it?

embeddedt, 2020-12-21:
I think they are evolving it very rapidly, and as such there is a lot more churn and downtime than before.

commit c4d978fa (2020-12-20): fix(textarea): buffer overflow in password mode with UTF-8 characters

fixes #1960