lvgl/lvgl #1965: fatal error: ../../lv_conf.h: No such file or directory

kerniew, 2020-12-20:
**Describe the bug**

I'm using a fresh install of Visual Studio Code with PlatformIO.  I start a new project and use the LVGL library by searching for lvgl in the Libraries for PlatformIO.

This is my platformio.ini:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:seeed_wio_terminal]
platform = atmelsam
board = seeed_wio_terminal
framework = arduino
lib_deps = lvgl/lvgl@^7.8.1

This is my main.cpp:
#include <Arduino.h>
#include <lvgl.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

The result of compiling produced this error:
In file included from .pio/libdeps/seeed_wio_terminal/lvgl/src/lv_core/lv_obj.h:16:0,
                 from .pio/libdeps/seeed_wio_terminal/lvgl/src/lv_core/lv_group.h:17,
                 from .pio/libdeps/seeed_wio_terminal/lvgl/src/lv_core/lv_group.c:9:
.pio/libdeps/seeed_wio_terminal/lvgl/src/lv_core/../lv_conf_internal.h:46:14: fatal error: ../../lv_conf.h: No such file or directory
 #include "../../lv_conf.h"                 /*Else assume lv_conf.h is next to the lvgl folder */

**To Reproduce**

I tried the same install on another clean install of VS Code and PlatformIO and the same result.

**Expected behavior**

Successful compilation

**Screenshots**

**Additional context**

I copied the lv_conf.h from:
https://raw.githubusercontent.com/lvgl/lv_sim_eclipse_sdl/master/lv_conf.h

Now it compiles.

kisvegabor, 2020-12-21:
Hi,

Take a look at this PIO project to see how LVGL is configured.
https://github.com/lvgl/lv_platformio

If it's still not working please open an issue in the linked repo.

closed 2020-12-21 by kisvegabor

ralphxyz, 2024-04-20:
Where did you save lv_conf.h to? I cannot find the libraries to save it?

What does "Else assume lv_conf.h is next to the lvgl folder " mean?