lvgl/lvgl #1968: ScrollBar is not working properly.

RinkuJ, 2020-12-21:
# All enhancement requests or questions should be directed to the Forum.

ScrollBar is not working properly. whenever i pull it up with my cursor scroll bar is going down and vice versa.
 /*Create a scroll bar style*/
    static lv_style_t style_sb;
    lv_style_copy(&style_sb, &lv_style_plain);
    style_sb.body.main_color = LV_COLOR_BLACK;
    style_sb.body.grad_color = LV_COLOR_BLACK;
    style_sb.body.border.color = LV_COLOR_WHITE;
    style_sb.body.border.width = 1;
    style_sb.body.border.opa = LV_OPA_70;
    style_sb.body.radius = LV_RADIUS_CIRCLE;
    style_sb.body.opa = LV_OPA_60;
    style_sb.body.padding.right = 3;
    style_sb.body.padding.bottom = 3;
    style_sb.body.padding.inner = 8;        /*Scrollbar width*/

    /*Create a page*/
    lv_obj_t * page = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page, 150, 200);
    lv_obj_align(page, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_page_set_style(page, LV_PAGE_STYLE_SB, &style_sb);           /*Set the scrollbar style*/

    /*Create a label on the page*/
    lv_obj_t * label = lv_label_create(page, NULL);
    lv_label_set_long_mode(label, LV_LABEL_LONG_BREAK);            /*Automatically break long lines*/
    lv_obj_set_width(label, lv_page_get_fit_width(page));          /*Set the label width to max value to not show hor. scroll bars*/
    lv_label_set_text(label, "Lorem ipsum dolor sit amet, consectetur adipiscing elit,\n"
                             "sed do eiusmod tempor incididunt ut labore et dolore magna aliqua.\n"
                             "Ut enim ad minim veniam, quis nostrud exercitation ullamco\n"
                             "laboris nisi ut aliquip ex ea commodo consequat. Duis aute irure\n"
                             "dolor in reprehenderit in voluptate velit esse cillum dolore\n"
                             "eu fugiat nulla pariatur.\n"
                             "Excepteur sint occaecat cupidatat non proident, sunt in culpa\n"
                             "qui officia deserunt mollit anim id est laborum.");

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

embeddedt, 2020-12-21:
I believe it works as intended. It functions the way it would on a phone, where sliding your finger up on the screen causes the page to go down.

closed 2020-12-21 by embeddedt

RinkuJ, 2020-12-21:
I have  implemented a scrollbar from above code only. I am testing it on my laptop. whenever I'm pulling cursor in up direction scroll bar and its content going down(bar and the content are in sync). Ideally,  if i m trying to pull the scroll bar up , the content and bar should move in up direction and its not happening.

embeddedt, 2020-12-21:
The problem is that this behavior doesn't match how it would work on a touchscreen, which is what LVGL is designed for. If you try it on a touchscreen it will seem a lot more intuitive.

Dragging the bar itself is not supported, if that's what you are trying to do. You will just end up dragging the page underneath.

[image]

RinkuJ, 2020-12-21:
does it support movement using up down button (on laptop)?

RinkuJ, 2020-12-22:
Thankyou @embeddedt for sharing this information. 
does this scrollbar support movement using up and down buttons?

kisvegabor, 2020-12-22:
> does this scrollbar support movement using up and down buttons?

On `lv_page` yes. But you need to add a keyboard driver and add the page into a group:
https://docs.lvgl.io/latest/en/html/overview/indev.html#keypad-and-encoder

RinkuJ, 2020-12-22:
Thankyou @kisvegabor . this information helped me .