lvgl/lvgl #1970: Chart scroll and zoom

kisvegabor, 2020-12-21:
Hi,

I'm integration the chart widget into v8 and just found that it's quite simple to implement scrolling and zoom with the new scroll system. It's just a test but working:
[image]

But I got a couple of questions that I'd like to discuss with you:
1. How to set the zooming factor and series "virtual" size:
  - set the number of visible points? (100 point is total but only 10 is visible -> 10 times larger virtual width to scroll)
  - set the distance between points?
  - set a zooming factor? (150% zoom in X direction) 
2. What to do with the tick labels if there are a lot of points?
  -  Make the user set the labels even if it's very long (`"1\n2\n3\n...999\n1000"`)
  - Add a callback that returns formatted labels (`my_format_cb(buf, 53, dec_pos, "kg") -> 5.3 kg`) 
  - Keep the grid and labels static (not scrolling), can be improved later

Right now I prefer "zooming factor" and "static grid". 

What do you think?

cc @ali-rostami  @upbeat27

kisvegabor, 2020-12-21:
As an experiment, I made some progress with the static grid and zoom factor.

2x zoom on X axis:
[image]

ali-rostami, 2020-12-22:
Hi @kisvegabor 
It's awesome. So good. thanks.
For me the static grid is O.K. but I think labels on the x-axis should be dynamic. In my project, it's the time and it can not be static while the chart is scrolled.

kisvegabor, 2020-12-22:
> it is showing the time -It's a trend chart- and it can not be static while the chart is being scrolled.

I also start to realize the static ticks+label are too big a limitation. I'm thinking about how to make it easy to use, fast, effecient, and dynamic.

upbeat27, 2020-12-23:
Honestly, I can't think of a situation where scrolling the data and not scrolling the grid/axis would be useful. Essentially, scrolling along the x-axis should scroll everything except the y-axis ticks/labels.

A good example of scrolling data is [this here](https://js.devexpress.com/Demos/WidgetsGallery/Demo/Charts/ZoomingAndScrolling/jQuery/Light/). Other than some things that are not under discussion (autoranging y-axis) it looks pretty much exactly how I would expect a chart scrolling to work.

As for the controlling the zoom amount: one possibility is that you specify the "zoomed" width... so if the zoomed width equals the actual width (default), there is no scrolling. This gives you very fine control over the zoom amount. You could specify the zoom width as 1px more than the width if you wanted too... it wouldn't make much sense, but then the plot could scroll 1px. This way, the user can still use floats outside of lvgl to control the zoom if they want (like setting the zoom width to 1.3f * width).

A callback for formatting axis labels sounds good. I use flot (a javascript plotting library) a lot and this is generally how this would be handled. You can view the [API here](https://github.com/flot/flot/blob/master/API.md) as a reference.

kisvegabor, 2020-12-27:
Thanks, @upbeat27 
Amongst the examples of flot, I've found  [this](http://www.flotcharts.org/flot/examples/axes-tickLabels/index.html) ([source code](https://github.com/flot/flot/blob/master/examples/axes-tickLabels/index.html)).

It shows what you can do with major/minor ticks, tick labels, and how scrolling affects the tick labels.

It seems if you `push` new data it automatically increments the X index (tick label)

embeddedt, 2021-01-10:
Relevant: https://forum.lvgl.io/t/chart-widget-enhancements/4245

kisvegabor, 2021-01-12:
Thanks! I'm thining about it.

kisvegabor, 2021-04-13:
I'm closing this issue as it's implemented in v8.

See the chart examples: https://github.com/lvgl/lvgl/tree/master/examples/widgets/chart

closed 2021-04-13 by kisvegabor