lvgl/lvgl #1971: My center is not center.

Trackhe, 2020-12-21:
**Describe the bug**

If i use `lv_obj_align(btn1, NULL, LV_ALIGN_CENTER, 0, 0);` the object dosn't start at center. it has a x and y offset.

**To Reproduce**
init display with 

dpwidth=480
dphigh=380

```
void tft_intlvl(){
  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);
  
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = dpwidth;
  disp_drv.ver_res = dphigh;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  /*Initialize the (dummy) input device driver*/
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register(&indev_drv);  
}
```

```
  lv_obj_t * cont1 = lv_cont_create(lv_scr_act(), NULL);
  lv_obj_align(cont1, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);
  lv_cont_set_fit2(cont1, LV_FIT_MAX,0);
  lv_obj_set_height(cont1, 0.08 * dphigh); // set height

  lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
  lv_obj_align(btn1, NULL, LV_ALIGN_CENTER, 0, 0);
  lv_obj_set_width(btn1, 10);
  lv_obj_set_height(btn1, 0.20 * dphigh);
  lv_obj_t * label1;
  label1 = lv_label_create(btn1, NULL);
  lv_label_set_text(label1, rezeptehm);
```

**Expected behavior**

object starts center. 
and can be centered by x-50% obj size like
```
  lv_obj_align(btn3, NULL, LV_ALIGN_IN_TOP_MID, -0.45 * dpwidth, 0.72 * dphigh);
  lv_obj_set_width(btn3, 0.90 * dpwidth);
```

but its offcenter.

It was working but after restart my ide it is now offcenter?!

embeddedt, 2020-12-21:
Usually you need to align the object after changing its size or creating child objects, not before.

Trackhe, 2020-12-21:
interesting that works!
Thanks :))

closed 2020-12-21 by Trackhe