lvgl/lvgl #1972: The interface will freeze and cannot be refreshed.

ArglyMan, 2020-12-22:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

The interface will freeze and cannot be refreshed when I frequently create and then delete a pop-up box about 40 times.

**To Reproduce**

The problem can be reproduced easily through my demo project with PC simulator( https://github.com/ArglyMan/lvgl.git ).  

**Expected behavior**

The interface can be displayed correctly. And the memory release problem after deleting the control object should be paid attention to to prevent memory leakage.

**Screenshots**
[image]
[image]

**Additional context**

Add any other context about the problem here.

kisvegabor, 2020-12-22:
Hi,

It happens because if you click "Create" twice you save the same `area_SoftPswd` and `btn_Cancel_SoftwarePwd` variables. And on the second create you overwrite the pointer to the first instance. 
The solution is simple:
1. Don't let the user create the pop-up twice
2. Delete the popup before crating it again.

BTW, there was a minor memory leak in your code. You initialized   `lv_style_t MsgboxStyle;` every time a pop up opens. I always reallocated the style properties without freeing the previous ones. 
I suggest initializing the styles in the main function only once.

---
I've just fixed the strange cursor position on chang to password mode.

ArglyMan, 2020-12-23:
Hi Kisvegabor,
 Thank you so much for your help. Your tips have helped me a lot, and my code has become more robust.   Also very grateful for your contribution to littlevgl. If you don't mind, I will ask you again when there are more difficult questions.  See you.

closed 2020-12-23 by ArglyMan

kisvegabor, 2020-12-27:
Glad to help.

Here mainly deal with the issue and features of LVGL. However for specific issues and questions we a [dedicated support service](https://lvgl.io/services). Feel free to contact me if you need further help.