lvgl/lvgl #1974: can we drag a page in inverse vertical direction?

RinkuJ, 2020-12-23:
# All enhancement requests or questions should be directed to the Forum.

 for example: if I pull the page up then showing the content of page above it and if i drag the page down will show the content below.is it possible?

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

embeddedt, 2020-12-23:
It's not possible in LVGL at the moment, but it doesn't seem like a difficult feature to implement.

RinkuJ, 2020-12-23:
@embeddedt 
I am just a beginner so could you please tell me how can i achieve this feature? i

kisvegabor, 2020-12-27:
I haven't tested it but at first look, you need to change all `act_y += proc->types.pointer.` to `act_y -= proc->types.pointer.` here:
https://github.com/lvgl/lvgl/blob/20d56ee6e97905f56eb71d30c9a391d4b4d7eb20/src/lv_core/lv_indev.c#L1316-L1365

kisvegabor, 2021-01-18:
As there was no activity here for a while  I suppose the issue is solved.

---

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2021-01-18 by kisvegabor