lvgl/lvgl #1980: Image rotation not working as expected

embeddedt, 2020-12-26:
**Describe the bug**

Images are not being rotated when `lv_img_set_angle` is used.

**To Reproduce**

The bug can be reproduced on the PC simulator.

```c
    LV_IMG_DECLARE(logo_lvgl);
    lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(img, &logo_lvgl);
    lv_obj_set_drag(img, true);
    lv_img_set_angle(img, 450);
```

This is the image file I'm using for testing: [logo_lvgl.zip](https://github.com/lvgl/lvgl/files/5743388/logo_lvgl.zip)

**Expected behavior**

The image should be rotated 45 degrees.

**Screenshots**

[image]

kisvegabor, 2020-12-27:
Hi,

It's because indexed images can't be rotated.

embeddedt, 2020-12-27:
Okay. I think we need to update the documentation in that case. 

https://docs.lvgl.io/latest/en/html/widgets/img.html#transformations

> Therefore indexed images, alpha only images or images from files can be transformed.

I think the "can" needs to be changed to "can't" there?

kisvegabor, 2020-12-30:
> I think the "can" needs to be changed to "can't" there?

Yes... typical me... :sweat_smile: 
I fixed it. The site should be updated in a few minutes.

closed 2020-12-30 by kisvegabor