lvgl/lvgl #1983: Improve custom memory allocation usage

C47D, 2020-12-30:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

In #1967 @kisvegabor suggested improvements for the custom memory allocation usage. 

**To Reproduce**

N.A.

**Expected behavior**

N.A.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

When using Kconfig the variables used by the user to set custom memory allocation are string type, when using CMake, like ESP-IDF, we can remove the quotes from the variables. This might not be possible with other build systems.

Another improvements can be added in the comments below.

liebman, 2020-12-30:
A few options off the top of my head:
1. Use a weakly bound symbol (maybe not available for all platforms)
2. Use a function pointer for these and allow them to be registered before `lv_init` is called.
3. Use a constant name like `lv_user_alloc` and `lv_user_free` if enabled via `Kconfig`

embeddedt, 2020-12-31:
Weak symbols are probably not an option because they depend on compiler-specific syntax. We only use attributes in [areas](https://github.com/lvgl/lvgl/blob/f0c52b35112e7734d52a2e25101e0d58ff50cea7/lv_conf_template.h#L256-L293) where they can be left blank as a default. I'm not sure if weakness is safe to use here. Also, as you've mentioned, some platforms might not support weak symbols at all.

I think the constant name is the best idea, as function pointers take up RAM and break the paradigm of calling `lv_init` before anything else.

kisvegabor, 2021-01-04:
I still think it's enough to make `LV_MEM_CUSTOM` configurable and let LVGL define `malloc/free` by default. 

The "host system" (where LVGL is built) can set `LV_MEM_CUSTOM_ALLOC/FREE` if needed.

The constant name idea sounds good as well but I see an issue with it. The user needs to write a wrapper to the already existing `my_malloc/free`. It's obviously possible but it's hard to imagine for me where to add these wrappers.  Not into LVGL, not into drivers. Maybe into a `lvgl_interface.c` file. But now not such "interface" is required.

C47D, 2021-01-04:
Maybe Zephyr users can let us know how they use this parameter. See https://github.com/lvgl/lvgl/pull/1967#issuecomment-752646193 and https://github.com/lvgl/lvgl/pull/1967#issuecomment-752649914 for their use case.

vanwinkeljan, 2021-01-04:
@C47D 
> Maybe Zephyr users can let us know how they use this parameter.

Currently the idea is to use Kconfig in the same way as the ESP-IDF does for Zephyr and the needed changes have been merged via #1979, the needed changes on the zephyr side are not yet there so it can be changed.

Zephyr has multiple options that the user can select for LVGL memory allocation:
1. Kernel space heap (functions: `k_malloc`/`k_free`)
2. Libc heap (functions: `malloc`/`free`)
3. Dedicated memory pool for LVGL in kernel space (functions: `lvgl_malloc`/`lvgl_free`)
4. Dedicated memory pool for LVGL in user space (functions: `lvgl_malloc`/`lvgl_free`)

closed 2021-05-14 by kisvegabor