lvgl/lvgl #1984: How to disable image mosaic

sk123sos, 2020-12-31:
when I route an image widget it will get partly repeat because of the size change of the widget,how can i disable the mosaic or is there any way to avoid this problem.

kisvegabor, 2021-01-04:
How do you mean "route and image widget"?

sk123sos, 2021-01-04:
> How do you mean "route and image widget"?

sorry I mean rotate an image with lv_img_set_angle

kisvegabor, 2021-01-04:
Can you send an image and a code snippet to reproduce?

sk123sos, 2021-01-05:
> Can you send an image and a code snippet to reproduce?

image_mp3_disk_pin = lv_img_create(image_mp3_bg, NULL);
lv_img_set_src(image_mp3_disk_pin, &image_e366_img);
lv_obj_set_pos(image_mp3_disk_pin, 412, 235);
lv_img_set_angle(image_mp3_disk_pin,450);

these code seems normal
[image_e366_img.zip](https://github.com/lvgl/lvgl/files/5767538/image_e366_img.zip)

sk123sos, 2021-01-05:
> Can you send an image and a code snippet to reproduce?
and it looks like this
[image]

sk123sos, 2021-01-05:
> Can you send an image and a code snippet to reproduce?

[image]

and it looks like this

kisvegabor, 2021-01-06:
It wróorks well for me with the latest version. Do you also use the latest LVGL?

sk123sos, 2021-01-06:
> It wróorks well for me with the latest version. Do you also use the latest LVGL?

I will try to update it later ,I m using v7.8 now.

FASTSHIFT, 2021-03-11:
@kisvegabor 
When the size of the set object is larger than the original size of the image, I hope that the picture can be displayed in the center instead of filling the mosaic. What should I do? How to disabling mosaic?

kisvegabor, 2021-03-18:
Mosaic can't be disabled. Instead of setting the size of the image can you add the image to a container, center the image, and set the container's size?

closed 2021-04-13 by kisvegabor