lvgl/lvgl #1985: [Discussion] More features on fonts

kisvegabor, 2020-12-31:
Continue https://github.com/lvgl/lv_font_conv/pull/53

I wonder how could we add more advanced features.
These can to my mind for first:

### Letter stroke
Let the user configure a stroke width in the converter. In LVGL both a letter and stroke color can be adjusted. 
It's useful:
- to highlight a text
- to write text on images. 
- just because it might look better in the design

### Emoji
E.g. if someone develops a smartwatch that can display messages it'd be nice to display the emojis too.

### Colord letter
Some examples: https://www.fontspace.com/category/color
It opens a whole new world for designers. E.g. imagine a toy that uses funky letters.
If there are only static text's images also could be used, but real texts can be changed run time and can be animated.

If the letters were images we would lose kerning which is very important with these fonts.

embeddedt, 2021-01-05:
> Let the user configure a stroke width in the converter. In LVGL both a letter and stroke color can be adjusted.

Could this also be used for a rudimentary text shadow implementation, or is that type of quality something better suited to images, given the performance limitations of runtime rendering?

kisvegabor, 2021-01-05:
> Could this also be used for a rudimentary text shadow implementation, or is that type of quality something better suited to images, given the performance limitations of runtime rendering?

By text-shadow do you mean text blur? If so it's not so complicated, just blur every letter, consider an offset, draw the blur, and then the letter. But even in CSS, you can't set `spread` for text-shadow to make the "base letter" larger.

Some great stuff here: https://www.codesdope.com/blog/article/25-creative-css3-text-shadow-effects-you-cant-miss/

embeddedt, 2021-01-05:
> If so it's not so complicated, just blur every letter, consider an offset, draw the blur, and then the letter.

Wouldn't that double the rendering cost per letter (since a blur has to be drawn as well)?

kisvegabor, 2021-01-06:
Actually, it will triple the cost:
1. Compute blur
2. Draw blur
3. Draw letter

But the effects are not free. Something for something. :(

kisvegabor, 2021-04-13:
Let's get back to this if these features will be requested by more users.

closed 2021-04-13 by kisvegabor