lvgl/lvgl #1987: ESP-IDF and LVGL, KConfig, andlv_conf_internal have errors that can be fixed as follows.

wreyford, 2021-01-01:
**Describe the bug**
Problem arises using VSCode and Espressif plugin and IDF-MenuConfig, no longer using lv_conf.h

1). When using IDF-MenuConfig, the LV_CONF_SKIP option does not display, and the fix suggested below is for components\LVGL\KConfig
2). When displaying float formatted text on the LCD, it displays an **f** and not 999.99 as an example. The offending code is in lv_conf_internal.h and the fix suggested below.

**To Reproduce**

1). Open IDF-MenuConfig, in VSCode, using espressif plugin. You will notice that **Define CONFIG_LV_CONF_SKIP so we can use LVGL without lv_conf.h file** is not displaying as it should.

2). The following code fails to display correctly, even with the correct **Text Setting** of **Disable float in built-in (v)snprintf functions**
```c
lv_label_set_text_fmt(scale_total, "%.2f", 200.00);
```

**Expected behavior**

1) The IDF MenuConfig option discussed above should display to the end user.
2) I should see **200.00** on the display, but getting **f**.

**Solution**
1) Alter components/lvgl/Kconfig from line 13
```c
 # Define CONFIG_LV_CONF_SKIP so we can use LVGL
    # without lv_conf.h file, the lv_conf_internal.h and
    # lv_conf_kconfig.h files are used instead.
    config LV_CONF_SKIP
        bool "Define CONFIG_LV_CONF_SKIP so we can use LVGL without lv_conf.h file"
        default y
        help Define CONFIG_LV_CONF_SKIP so we can use LVGL
             without lv_conf.h file, the lv_conf_internal.h and
             lv_conf_kconfig.h files are used instead.
```
2) Change line 1394 from #define  LV_SPRINTF_DISABLE_FLOAT 1 to #define  LV_SPRINTF_DISABLE_FLOAT 0
```c
#else   /*!LV_SPRINTF_CUSTOM*/
#ifndef LV_SPRINTF_DISABLE_FLOAT
#  ifdef CONFIG_LV_SPRINTF_DISABLE_FLOAT
#    define LV_SPRINTF_DISABLE_FLOAT CONFIG_LV_SPRINTF_DISABLE_FLOAT
#  else
#    define  LV_SPRINTF_DISABLE_FLOAT 0
#  endif
#endif
#endif  /*LV_SPRINTF_CUSTOM*/
```
**Additional context**
[See the forum thread](https://forum.lvgl.io/t/lv-label-set-text-fmt-outputting-f-on-display-and-not-the-actual-text/4164)

**Suggested enhancements**
Please include the fact that if using KConfig we no longer require lv_conf.h ,and this will also be more apparent when the option shows up in IDF-MenuConfig.

I request that you add to the KConfig, the option for Custom Fonts, and then a place where we can enter those font  names and perhaps the path to those if needed, as when no longer using lv_conf.h we have no way of letting the compiler know about our custom fonts.

C47D, 2021-01-01:
Thanks for the detailed report, I've submitted a PR to try fix it. Will update the documentation to state Kconfig based configurations no longer need `lv_conf.h`, there are some configuration options we can't add to the Kconfig file because it's limited types, but will try to figure something out.

> I request that you add to the KConfig, the option for Custom Fonts, and then a place where we can enter those font names and perhaps the path to those if needed, as when no longer using lv_conf.h we have no way of letting the compiler know about our custom fonts.

See above, some configuration options are not possible to be added into Kconfig, we will try something out. I haven't used custom fonts, but as far as I remember you can add custom fonts outside the `lv_conf.h` file. Can you help me confirm this @kisvegabor @embeddedt ?

C47D, 2021-01-02:
@wreyford Can you confirm #1988 fixes the sprintf problem? So it can be merged in master.

wreyford, 2021-01-03:
>@wreyford Can you confirm #1988 fixes the sprintf problem? So it can be merged in master.

Fantastic. Yes, I have undone my changes to the lv_conf_internal.h (that is Generated) , and applied your suggested fix to lv_conf_kconfig.h and **confirm that it works as it should in both the enabled and disabled state** when applying the setting:
   **Disable float in built-in (v)snprintf functions** 
in IDF-MenuConfig.

Please don't forget to attend to the Kconfig fix required that makes out first part of this post.

C47D, 2021-01-03:
You mean making `LV_CONF_SKIP` visible, right? We hide it because wanted to always disable using `lv_conf.h` when using LVGL in Kconfig based projects. Any thought @kisvegabor ?

liebman, 2021-01-03:
One can always show hidden options with the `A` key in `menuconfig`.

C47D, 2021-01-25:
I think this can be closed

kisvegabor, 2021-01-25:
Okay, thank you!

closed 2021-01-25 by kisvegabor