lvgl/lvgl #1989: lv_conf_internal: Some CONFIG_* symbols are lowercase

C47D, 2021-01-01:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Symbols generated by Kconfig are always uppercase, in `lv_conf_internal.h` `CONFIG_lv_snprintf` and `CONFIG_lv_vsnprintf` are not.

**To Reproduce**

See 
https://github.com/lvgl/lvgl/blob/f0c52b35112e7734d52a2e25101e0d58ff50cea7/src/lv_conf_internal.h#L1375-L1388

**Expected behavior**

`CONFIG_lv_snprintf` and `CONFIG_lv_vsnprintf` should be uppercase. This should require updating the `lv_conf_checker.py`.

kisvegabor, 2021-01-04:
Sorry, I've seen your PR only when I pushed the fix. 
Thank you anyway! Now it should be fixed in aba10b0

I've added a new commit to use `upper()` as you did.

closed 2021-01-04 by kisvegabor

C47D, 2021-01-04:
No problem, thanks for the quick fix.

commit aba10b01 (2021-01-04): fix(lv_conf_internal): be sure Kconfig defines are always uppercase

fixes #1989