lvgl/lvgl #1992: Possibility for divide by zero error in lv_cpicker.c widget

fvanroie, 2021-01-03:
**Describe the bug**

There is a divide by zero possibility when the height of a rounded rectangular color picker widget is 1 larger then its width and radius>h/2. 

**To Reproduce**

```c
    lv_obj_t *cp = lv_cpicker_create(lv_scr_act(),NULL);
    lv_cpicker_set_type(cp, LV_CPICKER_TYPE_RECT);

    lv_obj_set_width(cp,82);
    lv_obj_set_height(cp,83); // width + 1
    lv_obj_set_style_local_radius(cp,LV_CPICKER_PART_MAIN,LV_STATE_DEFAULT,50); // r > h / 2
```

causing `lv_area_get_width(&grad_area)` and `grad_w` to be 0 and this division to fail:

```c
   lv_coord_t grad_w = lv_area_get_width(&grad_area);
    uint16_t i_step = LV_MATH_MAX(LV_CPICKER_DEF_QF, 360 / grad_w);
```

The math leading upto this result is in the [lv_cpicker.c code](https://github.com/lvgl/lvgl/blob/f0c52b35112e7734d52a2e25101e0d58ff50cea7/src/lv_widgets/lv_cpicker.c#L555-L582)

**Expected behavior**

Check `grad_w` before division so it won't cause an exception.

**Screenshots**

[image]

```
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
```

**Additional context**

My application lets the user specify the position and size of objects, so the height and width are variable. During testing of the widgets with some *randomized* values I accidentally hit this error on ESP32. It can be reproduced in Code::Blocks and any values that fit the conditions described causes an exception.

I could code around it by preventing the edge case but the logical solution would be to test the result of `grad_w` before using it in the division.

kisvegabor, 2021-01-05:
Thanks for the report.
It seems the best would be to return if `grad_w < 1`. IN this there is nothing useful to draw anyway.
```c
   lv_coord_t grad_w = lv_area_get_width(&grad_area);
    if(grad_w < 1) return;
```

What do you think?

fvanroie, 2021-01-05:
Looks good. I had added *`    if (grad_w==0) grad_w = 1; `* as a workaround, but a `return` does make more sense since there is nothing to draw. I added the `return` statement instead which indeed fixes the crashes and all my tests run OK.

The circumstance when the exception occurs is very limited, but still worth that extra check. Thanks for your quick response!

embeddedt, 2021-01-05:
@kisvegabor I wonder if we should take advantage of @fvanroie's idea and add a CI test that tests all the objects with random sizes.

kisvegabor, 2021-01-06:
@fvanroie 
Thanks, I've added the fix.

@embeddedt 
Good idea. It should be a pseudo-random number to make the test reproducible.

closed 2021-01-06 by kisvegabor

embeddedt, 2021-01-06:
Having thought about it some more I wonder if looping through all width and height combinations from 0 to 100 would catch 99% of the issues.

kisvegabor, 2021-01-09:
All options seem overkill to me, but IMO an even wilder stress test with carefully selected or random values would filter out most of the issues. 
Note that now we perform only a build test, and test some modules and don't even try to create a single widget.

fvanroie, 2021-01-10:
I'm currently only validating setter/getter pairs. My test cases are not truly random either, but deliberately picked test cases with normal values and abnormal values for multiple parameters. Then I used these values to *also* mix them up a bit and use them as input for other parameters.

This creates a limited data set with normal and abnormal values that sometimes produces unexpected results. Usually it breaks something in my application, but in this case it uncovered an issue with a library.

I'm mainly using this approach to stress-test my application, look for memory leaks and validate the MQTT interface.

commit 8cd07b6f (2021-01-06): fix(cpicker): fix division by zero

fix #1992