lvgl/lvgl #1993: needle not entirely erased

uraich, 2021-01-03:
**Describe the bug**

When using an image as needle with the gauge widget the needle from the old position is not entirely erased, when moving the needle

**To Reproduce**

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally this should work in the PC simulator unless the problem is specific to one platform.
reproduce:

https://github.com/uraich/lv_mpy_examples/blob/main/st7735_demos/voltmeter/voltmeter2.py

See the needle move without traces from previous positions

A clear and concise description of what you expected to happen.

**Screenshots**

[image]

You see the blue dots from previous needle positions

kisvegabor, 2021-01-04:
Thanks for the report, I've fixed it in `master`.

closed 2021-01-04 by kisvegabor

uraich, 2021-01-04:
Checked it. It works fine now. 
Thanks

commit de44f745 (2021-01-04): fix(gauge): fix redraw with image needle

fix #1993