lvgl/lvgl #1994: [v8] Rename x_DISABLE options to x_ENABLE

C47D, 2021-01-04:
It's the only option I have checked, because of the issue related to it, I can check other configuration options as well.

I was thinking on the name of this configuration, maybe call it ENABLE (and set it to 0 by default) instead of DISABLE, but we can discuss about in the future.

_Originally posted by @C47D in https://github.com/lvgl/lvgl/issues/1988#issuecomment-753404659_

kisvegabor, 2021-01-04:
What about `LV_SPRINTF_USE_FLOAT`? It doesn't feel perfect to me but we already use the `USE` word a lot in configs.

There are many config options with `1` default value but I think we are interested only in the ones with inverted logic. I.e. `1` means disabled.

embeddedt, 2021-01-04:
> I think we are interested only in the ones with inverted logic. I.e. 1 means disabled.

Correct.

> What about LV_SPRINTF_USE_FLOAT?

`LV_SPRINTF_SUPPORT_FLOAT`?

C47D, 2021-01-04:
> I think we are interested only in the ones with inverted logic. I.e. 1 means disabled.

Yes

> LV_SPRINTF_SUPPORT_FLOAT?

This sounds fine to me.

puzrin, 2021-01-04:
`LV_SPRINTF_FLOAT_ON`

kisvegabor, 2021-01-05:
I'm ok with any of them.

C47D, 2021-01-05:
Ok, will send a PR against `dev` later this week.

kisvegabor, 2021-04-13:
It's fixed in v8. See https://github.com/lvgl/lvgl/blob/834e2177e7fd5c493abf92aa097f54d49e57917d/lv_conf_template.h#L205

So I close this issue.

closed 2021-04-13 by kisvegabor