lvgl/lvgl #1997: screen fill problems

Dmitry-Rusanov, 2021-01-07:
stm32f746igt 
dysplay 1024x600
when filling the screen twitches.

#define MY_DISP_HOR_RES 1024
#define MY_DISP_ACT_ROWS 10
...
void my_flush_cb(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
	  uint32_t  Xaddress = 0;
	 // uint32_t Ypos = area->y1;
	  BSP_LCD_SelectLayer(0);
	  printf("x1 = %d x2 = %d y1 = %d y2 = %d \r\n",area->x1, area->x2, area->y1, area->y2);
	  Xaddress = (hLtdcHandler.LayerCfg[0].FBStartAdress) + 2*(MY_DISP_HOR_RES * area->y1);
	  DMA2D_CopyBuffer(color_p,Xaddress,MY_DISP_HOR_RES,MY_DISP_ACT_ROWS,0,0);
	  lv_disp_flush_ready(disp_drv);
}

main
...
  lv_init();
  /*A static or global variable to store the buffers*/
  static lv_disp_buf_t disp_buf;

  /*Static or global buffer(s). The second buffer is optional*/
  static lv_color_t buf_1[MY_DISP_HOR_RES * MY_DISP_ACT_ROWS];
  static lv_color_t buf_2[MY_DISP_HOR_RES * MY_DISP_ACT_ROWS];

  /*Initialize `disp_buf` with the buffer(s) */
  lv_disp_buf_init(&disp_buf, buf_1,buf_2, MY_DISP_HOR_RES * MY_DISP_ACT_ROWS);

  lv_disp_drv_t disp_drv;                 /*A variable to hold the drivers. Can be local variable*/
  lv_disp_drv_init(&disp_drv);            /*Basic initialization*/
  disp_drv.buffer = &disp_buf;            /*Set an initialized buffer*/
  disp_drv.flush_cb = my_flush_cb;        /*Set a flush callback to draw to the display*/
  disp = lv_disp_drv_register(&disp_drv); /*Register the driver and save the created display objects*/

my_flush_cb sending area:
x1 = 0 x2 = 1023 y1 = 54 y2 = 63 
x1 = 0 x2 = 1023 y1 = 64 y2 = 73 
x1 = 0 x2 = 1023 y1 = 74 y2 = 83 
x1 = 0 x2 = 1023 y1 = 84 y2 = 93 
x1 = 0 x2 = 1023 y1 = 94 y2 = 103 
x1 = 0 x2 = 1023 y1 = 104 y2 = 113 
x1 = 0 x2 = 1023 y1 = 114 y2 = 123 
.........................
x1 = 0 x2 = 1023 y1 = 594 y2 = 599 
x1 = 0 x2 = 1023 y1 = 54 y2 = 63 

what may be the reason for the absence of 0 (area->y1)

https://user-images.githubusercontent.com/55001381/103896263-c6adde80-5102-11eb-888b-69be28dea98e.mp4

, I tried it on versions 7 and 8, the effect is the same. At least understand in which direction they are moving.

embeddedt, 2021-01-07:
Call `SCB_CleanInvalidateDCache();` at the start of the flush function. The F7 and H7 processors need their cache flushed before doing transactions to the framebuffer.

Also check that you are not using write-through cache; I have had problems with it and the framebuffer.

Dmitry-Rusanov, 2021-01-07:
SCB_CleanInvalidateDCache(); did not help
obviously an error in the STM32CUBEIDE settings,
started a new project, pasted in my code and it worked.
Thanks for the quick response.
If not difficult to explain how to use LV_MEM_ADR in the configuration file. did not find a description.

Now static objects look very good, the refresh bar was noticeable in the dynamics,
when I used dma2d.
Changed the display driver code, instead of static buffers, indicated the display address, the bar disappeared, flickering appeared,
but this is in dynamics when objects move in your examples.

kisvegabor, 2021-01-09:
I suggest to take look at this driver: https://github.com/lvgl/lv_port_stm32f746_disco/blob/master/hal_stm_lvgl/tft/tft.c
It also for STM32F746

closed 2021-02-28 by Dmitry-Rusanov