lvgl/lvgl #1998: lvgl blur help

FlyLu, 2021-01-07:
In some setting scenes, the display needs to be blurred. 
I tested the canvas and found that it has a blurry effect. But need to put the image on the canvas

```c
  lv_obj_t *canvas = lv_canvas_create(lv_scr_act(), NULL);
  static lv_color_t canvas_buf[252 * 166];
  lv_canvas_set_buffer(canvas, canvas_buf, 252, 166, LV_IMG_CF_TRUE_COLOR_ALPHA);

  lv_draw_img_dsc_t img_dsc;
  lv_draw_img_dsc_init(&img_dsc);
  lv_canvas_draw_img(canvas, 0, 0, "z:img.png", &img_dsc);

  lv_area_t area = {0, 0, 252, 100};

  //! Blur needs to be set vertically and horizontally together
  lv_canvas_blur_hor(canvas, &area, 30);
  lv_canvas_blur_ver(canvas, &area, 30);
```

[image]

I want to blur a part of the screen, not limited to the canvas, how should I solve it

Possible ideas
1. How to get the display data of an area on the screen
2. Assuming that it can be obtained, then write it to the canvas to blur

I hope that there will be a new interface in the future, which can set an area to be blurred on the screen

kisvegabor, 2021-01-09:
Hi,

It's a little bit complicated now and not supported out of the box. 
The problem with blur is the following. Let's imagine there is a real-time blurred rectangle with a label. If you change the label's text, the text area will be invalidated and redrawn and the background's blur needs to be recalculated. However, the blurred color of the top-left coordinate of the redrawn area can not be calculated because it needs some pixels out of the redrawn area too.

I'm thinking about a feature that allows converting any widget to a canvas to perform transformations, blur, and other effects. 

But for now, as you figured out,  you need to screenshot the widget into a canvas buffer.

There are some ways to do this but I think the most simple is this:
1. Call `lv_refr_now()` just to redraw the invalid areas if any
2. Set a new `flush_cb` for your driver or set a flag the alter its behavior. 
3. Call `lv_obj_invalidate(obj_to_snapshot)`
4. Call `lv_refr_now()` to perform the redraw with the new `flush_cb` 
5. In the new or modified `flush_cb` write received pixels to the canvas's buffer.
6. Restore the original `flush_cb`

embeddedt, 2021-01-09:
If you have a predefined set of widgets to blur, you will probably also get more performance from making a blurred image in advance and switching to that, rather than doing the blur at runtime (as long as you have the flash space to store images).

FlyLu, 2021-01-11:
Thank you kisvegabor and embeddedt for your reply!

Regarding blur, I used an alternative method, which is to set a black background with 50% transparency

kisvegabor, 2021-01-11:
You're welcome. 

> Regarding blur, I used an alternative method, which is to set a black background with 50% transparency

It's really much simpler. :slightly_smiling_face: 

--- 

Note that,  we use GitHub Issues for development-related discussions.

In the future, please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

FlyLu, 2021-01-11:
Thank you, I got it.

closed 2021-01-11 by FlyLu

jianjunxia, 2022-04-06:
> Hi,
>
> It's a little bit complicated now and not supported out of the box. The problem with blur is the following. Let's imagine there is a real-time blurred rectangle with a label. If you change the [...]

Hi，Currently I find a suitable fast Gaussian filter，I want to know, how to get the background value of the area where obj is located? Like some data behind a picture.
This way I can send the value of obj to this fast filter function for blurring

https://forum.lvgl.io/t/how-to-implement-this-feature-in-lvgl-with-background-filter-blur-60px-in-css/8591?u=jianjunxia