lvgl/lvgl #2000: when disable a parent, should make all children disalbed?

incity, 2021-01-08:
1. when the parent is disabled, why don't make all children disabled?
2. when the object is disabled, it should not receive any event?

kisvegabor, 2021-01-09:
Would you expect to disable literally every child? E.g. if you have a container with 5 buttons and a drop-down, and make the container disabled, shall all the buttons and the dropdown also disabled?

embeddedt, 2021-01-09:
That would probably make sense, to be honest.

incity, 2021-01-10:
i think an object is enabled only when its ancestors is also enabled. when i disable a parent, i(maybe most users) want let its child objects also disabled, but we don't want and should not to disable all child objects one by one.

kisvegabor, 2021-01-11:
I also agree it'd be better. I'll add it n v8.

In v7 you should disable the scrollable area of the text area too. (`lv_page_get_scrollable(ta)`) This way text area won't accept clicks and keys from any input device. But a keyboard widget can still target it and push characters.

kisvegabor, 2021-03-03:
In v8 there are `lv_obj_add/clear_state(obj, <STATE>)` functions. I think adding `lv_obj_add/clear_state_recursive(obj, <STATE>)` would be the cleanest because
- you need to add the disabled manually anyway so it's also only 1 function call
- it can be used for other states too

What do you think?

embeddedt, 2021-03-03:
IMO the recursive state should be default and nonrecursive should be a separate API.

puzrin, 2021-03-03:
1. As far as i remembed, on web `disabled` is attribute of input fields only. Those are not containers.
2. Instead of introduce lot of `*_recursive`, create DOM traversal helpers (functions or macros). With new compilers, [iterators + lambdas] will be easy to use.

incity, 2021-03-03:
maybe can refactor the lv_textarea, don't let it be a container.
in Android, the EditText (same as lv_textarea) inherited from TextView (similar to lv_label), and TextView inherited from View(same as lv_obj)

kisvegabor, 2021-03-03:
@embeddedt 
> IMO the recursive state should be default and nonrecursive should be a separate API, at least for LV_STATE_DISABLED.

I think as other API functions are not recursive (e.g. `lv_obj_set_width`) it'd be counterintuitive if one of the functions worked diffrentely. 

@puzrin 
> As far as i remember, on web disabled is attribute of input fields only. Those are not containers.

Buttons, checkboxes, radio buttons and selects can be disabled too. See https://www.w3schools.com/tags/att_disabled.asp

> Instead of introduce lot of *_recursive, create DOM traversal helpers (functions or macros). With new compilers, [iterators + lambdas] will be easy to use.

It's also possible (I haven't added the dom iterator yet). However, for common things, it might inconvenient to add new functions only for the iterator.

@incity 
> maybe can refactor the lv_textarea, don't let it be a container.
in Android, the EditText (same as lv_textarea) inherited from TextView (similar to lv_label), and TextView inherited from View(same as lv_obj)

In v8 Textarea is inherited directly from the base object and there is a label on it. I've just tried obj->label->textare but there are some missing features in the label to handle this.

If you disable the text area the label will be disabled too because its state is inherited from the text area.

puzrin, 2021-03-03:
> Buttons, checkboxes, radio buttons and selects can be disabled too.

Checkboxes, radio, selects are `<input>` with certain `type=`. But you are rught, `<button>` is not `<input>`, and it's container. Sorry for buzz.

> It's also possible (I haven't added the dom iterator yet). However, for common things, it might inconvenient to add new functions only for the iterator.

IMO order is important. Do [design] mandatory things first, optional second (universal iterators are mandatory). In reverse order - more chances to bloat api. It's common problem of all APIs - bloating with "cool methods".

kisvegabor, 2021-03-04:
> Checkboxes, radio, selects are <input> with certain type=. 

Oh, true.

> IMO order is important. Do [design] mandatory things first, optional second (universal iterators are mandatory). In reverse order - more chances to bloat api. It's common problem of all APIs - bloating with "cool methods".

Actually, I agree. I started to "believe" in providing useful, versatile and simple tools/APIs and let the user craft things with it. 

So thinking more about it, I like the idea of using a general dom iterator for this. E.g.
```c
/**
 * start_obj: start from this obj
 * deepness: how much level to walk
 * callback: it will be called on every obj
 * user data: will be passed to the callback 
 * strategy: e.g. depth-first search, breadth-first search 
 */
lv_obj_tree_walk(start_obj, deepness, callback, user_data, strategy)
```

embeddedt, 2021-03-04:
> So thinking more about it, I like the idea of using a general dom iterator for this.

Would there still be wrappers on top of this iterator for disabling all child objects in a more intuitive manner? While a general iterator is useful, I'm not sure that all users will understand how to use it right away.

kisvegabor, 2021-03-04:
Actually, this is the first time when automatically disabling all children of a container came up. So I don't think it's needed by the vast majority of users. 

IMO in those rare cases, we can expect the users to write a function for it. We can add an example of how to do it, of course.

puzrin, 2021-03-04:
```c
lv_obj_tree_walk(start_obj, deepness, callback, user_data, strategy)
```

Not good. Iterator should have minimal number of params. Something like `(root, handler_fn)` everything else should be done via handler params & returned value.

embeddedt, 2021-03-04:
> Actually, this is the first time when automatically disabling all children of a container came up. So I don't think it's needed by the vast majority of users.

True. I was thinking more along the lines of objects which have multiple real parts, like the text area. You mentioned [in your previous comment](https://github.com/lvgl/lvgl/issues/2000#issuecomment-757799525) that v7 required the scrollable to be disabled manually. In my opinion, any real child parts of an object should have their disabled state forcefully synchronized with the parent.

kisvegabor, 2021-03-08:
For the sake of simplicity, in v8 there won't be "real" parts for the core widgets.  However, the extra widgets (e.g. tabview) do have real parts (background + tab buttons + content area + tab pages)

I'm afraid if we introduce "disabled state propagation" someone will ask how can I propagate the focused state? Or if there is an option for custom states (I still want to add it) it's reasonable to create a custom `LV_STATE_WARNING` and set it for the whole screen or so.

So in summary these possible solutions came up (I like 1 and 2 the most)
1. Add only `lv_obj_tree_walk()` and show examples of how to set the state recursively.
2. Add  `lv_obj_set/clear_state_recusrsive` 
3. Make `lv_obj_set/clear_state` reclusive and add `lv_obj_set/clear_state_single` (I think it's counterintuitive becase all other functions are non-recursive)
4. Add `lv_obj_set/clear_disable` and/or `lv_obj_set/clear_disable_recursive` (I find it too specific)

What do you think?

embeddedt, 2021-03-08:
I vote for 1 and having 2 be a wrapper around 1.

puzrin, 2021-03-08:
(1) - must have. I'd like to see final signature first, prior to think about next.

If case is specific for enable/disable (not for all states), probably we could introduce `lv_obj_disable/enable(node, recursive = false)`, but not sure. I suggest to start testing and postpone sugar to v8.1. We need to stabilize "must have fn" signatures first.

incity, 2021-03-09:
@kisvegabor i don't think that's right, when i change an object state, what i want is just change the object itself and keep all it's children's state unchanged. but all it's children should show disabled(that's mean just change user interface, but don't really modify the children's state), and when i enable the object again, the previous user interface should keep the same as before, but if you change the state recusrsive, how can you restore all the children's old state?

puzrin, 2021-03-09:
BTW, we did not considered custom user event listeners. User could subscribe to `ON_STATE_CHANGE` and care about children as needed (or do this via class inheritance).

In long term, i'd suggest to revisit node interaction process in favour of events, instead of add more and more methods. With new [classes, allocator, deferred updates] this should be not expensive.

kisvegabor, 2021-03-09:
> i don't think that's right, when i change an object state, what i want is just change the object itself and keep all it's children's state unchanged. but all it's children should show disabled(that's mean just change user interface, but don't really modify the children's state), and when i enable the object again, the user interface should keep the same as before, but if you change the state recursive, how can you restore all the children's old state?

Right now there is no concept for "state inheritance". The issue you have mentioned really exists. That is if there a container and there is a disabled text input on it, you recursively disable all children of the container, then recursively enable all children, the text input will be enabled too. I'd consider it as a limitation.

> I vote for 1 and having 2 be a wrapper around 1.

I'm ok with that. 

> BTW, we did not considered custom user event listeners. User could subscribe to ON_STATE_CHANGE and care about children as needed (or do this via class inheritance).

It's also an option but in this case, you still need a recursive state changer. 

Let's figure out the API of `lv_obj_tree_walk()`. Based on @puzrin's [comment](https://github.com/lvgl/lvgl/issues/2000#issuecomment-790698290):

```c
lv_obj_tree_walk(lv_obj_t  * base, lv_obj_tree_walk_cb * walk_cb, void * user_data, lv_obj_tree_walk_strategy_t strategy);

void my_walk_cb(lv_obj_t * obj, uint32_t deepness, void * user_data);
```

What do ou think?

puzrin, 2021-03-09:
> ```c
> lv_obj_tree_walk(lv_obj_t  * base, lv_obj_tree_walk_cb * walk_cb, void * user_data, lv_obj_tree_walk_strategy_t strategy);
> 
> void my_walk_cb(lv_obj_t * obj, uint32_t deepness, void * user_data);
> ```
> 
> What do you think?

- Namings are under question. `obj_tree` - i don't like this. IMO "naive". Consider use well known terms - dom, node.
- not specified cb return codes to control process (because features are not formalized)
- `strategy` - what is this??

puzrin, 2021-03-09:
> Theoretically, DOM is not related only to HTML, but have you ever heard it in another context? I havn't.

Since "everything is scraped from HTML", it's logical to use familiar terms.

> Why do we need return codes?

To control nesting, for example. Return codes are "commands" to terminate part of traverse.

> If deepness were moved to `lv_obj_tree_walk` LVGL could take care of the whole iteration.

Iterator signature MUST be simple. This is frequently used feature. If you add all rare params there, you will get bs instead of convenient thing for everyday use.

> > strategy - what is this??
> 
> E.g. depth-first search, breadth-first search

See my comment above. Users don't care about order too much. Typing extra useless param everytime will be very annoying.

If you really need specific strategy sometime - just add one more iterator with different name.

kisvegabor, 2021-03-09:
> Since "everything is scraped from HTML", it's logical to use familiar terms.

Yes, but all the terms now are self-deceptive without any HTML/CSS knowledge. DOM is a mosaic work that might not be known for novice users. 

> To control nesting, for example. Return codes are "commands" to terminate part of traverse.

Then a simple true/false could work too. 

See this:
(user data is important IMO)
```c
lv_obj_tree_walk(lv_obj_t  * base, lv_obj_tree_walk_cb * walk_cb, void * user_data);

bool my_walk_cb(lv_obj_t * obj, uint32_t deepness, void * user_data);
```

puzrin, 2021-03-09:
> Yes, but all the terms now are self-deceptive without any HTML/CSS knowledge. DOM is a mosaic work that might not be known for novice users.

When you use common terms, you can refer to existing docs for learing/ideas.

If novice has zero knowledge - no difference which word to remember.

> Then a simple true/false could work too.

If i wish to search node, then i may wish to completely stop on success. Already 2 cases:

- stop more deep nesting
- immediate exit

Probably, just need several iterators:

- traverse everything - handler without `deepness`, returns nothing.
- traverse children only (1 level) - handler without `deepness`, returns nothing.
- tune-able traverse (with deepness & return statuses from walk_cb)

kisvegabor, 2021-03-09:
> When you use common terms, you can refer to existing docs for learing/ideas.

I think `tree` and `dom` refer to the ~same thing (at least from our perspective) and I find "tree" less cryptic. Let's see what others say.

> If i wish to search node, then i may wish to completely stop on success. Already 2 cases:

That's true.

What about this?
```c
LV_OBJ_TREE_WALK_OK,       
LV_OBJ_TREE_WALK_STOP,         // Do not go any deeper on this obj
LV_OBJ_TREE_WALK_EXIT,         // Fully stop iterating 
```
> Probably, just need several iterators:

I believe we can find a simple yet flexible signature.

puzrin, 2021-03-09:
`LV_OBJ_TREE_WALK_SKIP_CHILDREN`

kisvegabor, 2021-03-10:
> 90% of use are "walk all" or "walk own children only" (deepness and cb value unused)

I thought we agreed on `start_obj` is a "good" parameter. With this, we are always in "walk own children" mode.  `start_obj = NULL` can be used to iterate all the objects.

puzrin, 2021-03-10:
- "walk all" means recursive scan down from start_obj
- "walk own children only" - non-reqursive scan from start_obj

kisvegabor, 2021-03-10:
I see. I'd call it "walk direct children only". Anyway,  it can be achived by returning `LV_OBJ_TREE_WALK_SKIP_CHILDREN`.

puzrin, 2021-03-10:
> Anyway, it can be achived by returning LV_OBJ_TREE_WALK_SKIP_CHILDREN.

If in 90% cases users will have to type unused "deepness" and "return" - that's not clever.

See formal logic: https://github.com/lvgl/lvgl/issues/2000#issuecomment-794412993. Universal iterator must exist, but 2 more simple additional iterators will simplify life.

kisvegabor, 2021-04-08:
I've added `lv_obj_tree_walk` in 853dc050. It works like this:
```c
lv_obj_tree_walk_res_t count(lv_obj_t * obj, void * user_data)
{
    uint32_t * x = user_data;
    (*x)++;
    return LV_OBJ_TREE_WALK_NEXT;
}

...
  uint32_t x = 0;
  lv_obj_tree_walk(lv_scr_act(), count, &x);
  printf("%d\n", x);
```

kisvegabor, 2021-05-14:
I think for now `lv_obj_tree_walk` should be enough. 

I suggest adding more specific functions if there is a larger demand for them.

So I close this issue for now. If you see it differently, please comment here and I'll reopen the issue.

closed 2021-05-14 by kisvegabor