lvgl/lvgl #2006: Implemented basic support for Rlottie in LVGL [ENHANCEMENT]

ValentiWorkLearning, 2021-01-09:
- Implemented basic support for Rlottie format in LVGL
- Tested with simulator x86
- Need test with different hardware
- Uses CMake and Conan package manager for the simulator

Repository link:
https://github.com/ValentiWorkLearning/lv_rlottie

ValentiWorkLearning, 2021-01-09:
Possible improvements:
- reducing memory consumption with implementing converter from JSON to C-array of pre-rendered data frames + implementing file-api backend for chunked reading of the corresponding file
-??? whatever :)

barbiani, 2021-01-09:
Does it render to a canvas or uses the lvgl primitives to draw the
animation?

ValentiWorkLearning, 2021-01-09:
@barbiani renders to the allocated buffer in LVGL heap. It can be a kind of optimizations too- to use LVGL primitives for a slightly optimized rendering. This solution is just Proof-Of-Concept.

kisvegabor, 2021-01-11:
Woah, that's amazing. I was dreaming a lot about an engine like this. :heart_eyes: 

I don't know the performance of this lib but I assume that ~100MHz MCUs are too slow for this and 'Raspberry level boards should be targeted instead.

It's great that it works with Cmake.

@ValentiWorkLearning 
Would you interested in adding it to a `lv_lib_rlottie` repo? (similarly to `lv_lib_gif`)
If so, I'll create a repo and give you admin access there. 

I just saw that Samsun has an SVG/vector graphics lib as well, called [thorvg](https://github.com/Samsung/thorvg).

amirgon, 2021-01-11:
Both rlottie and thorvg have a very nice [C API](https://github.com/Samsung/rlottie/blob/master/inc/rlottie_capi.h).  
It should be pretty easy to automatically create Micropython API for these libraries, if that's interesting.

ValentiWorkLearning, 2021-01-11:
@kisvegabor I'd like to add repository lv_lib_rlottie. But before that it needs some testing on different ARM boards(Raspberry as an example). Just for clarifying the expectations of potential users.

Thorvg => probably we can do that also. But, again, it requires buffer for rendering(and probably will be suitable only for powerful ARM CPUs.

amirgon, 2021-01-11:
@ValentiWorkLearning Do you think rlottie and thorvg could work on an ESP32?

ValentiWorkLearning, 2021-01-11:
@amirgon It need testing with build for ESP32. I think, that it's possible with pre-rendering for an animation with rendering it fixed-size. Another idea is to try directly using RLottie with runtime rendering. This needs to test library build with gcc-arm-none-eabi toolchain for observing the binary size.

kisvegabor, 2021-01-12:
@ValentiWorkLearning 
I think it would be easier to test if we had the `lv_lib_rlottie` repo like a module. We could discuss issues there, send PR's, and so on. No problem if it turns out that we can't use it in the end.

ValentiWorkLearning, 2021-01-12:
@kisvegabor hmm. How to properly add this module and deal with it's dependency on rlottie submodule?

ValentiWorkLearning, 2021-01-12:
I agree that it would be nice to have this as a separated module. For git-flow functionality too. 👍

embeddedt, 2021-01-12:
Rlottie could be a submodule of `lv_lib_rlottie`. Nested submodules are supported by GitHub.

kisvegabor, 2021-01-12:
@ValentiWorkLearning I've created the repo and sent the invitation.
https://github.com/lvgl/lv_lib_rlottie

ValentiWorkLearning, 2021-01-12:
@kisvegabor Thanks! Which files have to be added to this repo?

kisvegabor, 2021-01-12:
At first look the content of [lv_rolettie](https://github.com/ValentiWorkLearning/lv_rlottie/tree/master/graphics/lv_rlottie) and [rlottie_samsung](https://github.com/ValentiWorkLearning/lv_rlottie/tree/master/graphics/rlottie_samsung) as submodule.

ValentiWorkLearning, 2021-04-14:
Suppose, we can close this.

kisvegabor, 2021-04-14:
Yes, further discussion should happen in [lv_lib_rlottie](https://github.com/lvgl/lv_lib_rlottie).

closed 2021-04-14 by kisvegabor