lvgl/lvgl #2007: lv_fs_open forces all file paths to be relative

cmumford, 2021-01-09:
### Describe the bug

**lv_fs_open** strips leading path delimiters from file paths, converting absolute paths to relative. This is a problem, with no obvious workaround when using the [SPIFFS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spiffs.html) filesystem on ESP-IDF. Removing the leading `'/'` resulted in the file being unable to be found.

### To Reproduce
The [LVGL documentation sample](https://docs.lvgl.io/latest/en/html/overview/file-system.html#usage-example) was used - except with this one change

```c
res = lv_fs_open(&f, "S:/spiffs/file.txt", LV_FS_MODE_RD);
```

The code above resulted in the open callback (open_cb) being called with the following path: "spiffs/file.txt". I tried modifying the mounting of the SPIFFS filesystem by removing the "spiffs" path here:

https://github.com/BigMumf/display-keyboard/blob/014177e9918f91561e715e39c5b2d5e8ec1526c7/main/filesystem.cc#L15-L20

Setting the mount path (`base_path `) to "" or "/" still failed. So it seems that maybe SPIFFS requires absolute paths which are stripped out by LVGL.

At present I have a hack to fix this by adding a leading '/':

https://github.com/BigMumf/display-keyboard/blob/014177e9918f91561e715e39c5b2d5e8ec1526c7/main/lvgl_drive.cc#L41-L49

### Expected behavior
Absolute paths should be valid.

### Screenshots or video
N/A - not a graphical problem.

embeddedt, 2021-01-09:
I see the issue, but I'm not sure whether this needs to be fixed or not. As far as I know, the filesystem logic is written with the assumption that there is a somewhat POSIX-compliant filesystem on the other end, meaning that your driver is expected to handle the path without the leading slash.

cmumford, 2021-01-09:
Yeah, maybe not POSIX-compliant, but certainly rooted somewhere and the the assumption is that a file paths, whether they have a leading '/' or not, are relative to that path. If that's the way it works then I'm OK with that - it was just unexpected and confused me.

embeddedt, 2021-01-09:
This way lets you choose how you want to configure the driver - for example, if I recall correctly, the PC simulator uses its working directory as the "root" path, meaning that a path `S:/test.txt` would be passed in as `test.txt`, and Linux would interpret it relative to the working directory. If we always put a slash at the start of the path, it would have to manually be taken off for this type of use case.

We should definitely mention this in the documentation, though.

cmumford, 2021-01-09:
I think my preference would be to leave the file path unchanged - aside from stripping the "S:". Either way it is what it is now, so this bug is probably working as intended. Maybe if it confuses other developers some clarification could be added to the docs. Thx for your comments.

liebman, 2021-01-09:
It may only really affect SPIFFS as that filesystem does not actually have directories.  (`/x.txt` and `x.txt` are seperate files with the leading `/` being part of the filename.)

xiaoxiang781216, 2021-01-10:
@cmumford it's easy to get the leading '/' back without copy like:
https://github.com/BigMumf/display-keyboard/blob/014177e9918f91561e715e39c5b2d5e8ec1526c7/main/lvgl_drive.cc#L41-L49
```
lv_fs_res_t file_open_cb(struct _lv_fs_drv_t* drv,
                         void* file_p,
                         const char* path,
                         lv_fs_mode_t mode) {
 --path; /* recover the leading '\' */
}
```

liebman, 2021-01-10:
One problem with fixing this is that existing `SPIFFS` files with no leading `/` will no longer be accessible, breaking some existing applications. I've switched to `littlefs` where I can as I've found it much faster for large files. 
A config option to make a "fix" optional maybe?

embeddedt, 2021-01-10:
It would be best to change it in v8 if we change it at all. I think it's better to have one consistent stance (we strip the `/` or not) than to have a config which could potentially break examples.

Personally, I still think that this fix should be at the driver level. SPIFFS is the only filesystem I've heard of where `/file.txt` and `file.txt` do not reference the same file. But I'm happy to go with whatever @kisvegabor and others decide.

kisvegabor, 2021-01-12:
@cmumford 
I haven't tested, but it seems setting [LV_FS_PC_PATH](https://github.com/lvgl/lv_fs_if/blob/7566cccad590a2a9af65f2e65f8bff03a9d689c0/lv_fs_pc.c#L25-L31) to `/spiffs/` would solve the issue. This way you could write `S:/file.txt` to reach `/spiffs/file.txt`

I agree to find a consistent solution. My motivation in trimming everything from the start of the path was to handle Windows and Linux-like paths the same way: 
`C:/path/to/file.txt` or `/path/to/file.txt` -> `path/to/file.txt` -> `LV_FS_PC_PATH/path/to/file.txt`.

kisvegabor, 2021-03-17:
I've fixed it by not trimming the `/` from the beginning (Only a single `:` is trimmed). See https://github.com/lvgl/lvgl/commit/a7084509b51e8c39d4018144b38b5b6a4b5242f8

closed 2021-03-17 by kisvegabor