lvgl/lvgl #2017: Error: converting the enum constant to a boolean

amirgon, 2021-01-11:
```
../../lib/lv_bindings/lvgl/src/lv_widgets/lv_win.c:610:34: error: converting the enum constant to a boolean [-Werror,-Wint-in-bool-context]
            case LV_TXT_FLAG_FIT || LV_TXT_FLAG_EXPAND
```

Related: [Build Javascript port job failure](https://github.com/lvgl/lv_micropython/runs/1683947280?check_suite_focus=true#check-step-8:~:text=..%2F..%2Flib%2Flv_bindings%2Flvgl%2Fsrc%2Flv_widgets%2Flv_win.c%3A610%3A34%3A%20error%3A%20converting%20the%20enum%20constant,1%20error%20generated.)

embeddedt, 2021-01-11:
I believe I fixed this in master already... can you check? I had noticed
this error the last time you updated lv_bindings.

amirgon, 2021-01-11:
@embeddedt You are right. Probably lvgl submodule needs to be updated to reflect this fix.

closed 2021-01-11 by amirgon