lvgl/lvgl #2020: Touchscreen Callback not called. ESP32/Arduino/ILI9341

grisuthedragon, 2021-01-13:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x]  Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) **Not possible ESP32/Hardware related**

### Describe the bug
I installed lvgl 7.9.0 from the Arduino library manager and TFT_eSPI 2.5.36. The later one is configured and working properly, verified with their examples. I setup the lvgl library as described in https://docs.lvgl.io/v7/en/html/get-started/quick-overview.html 
but the callback for the touchscreen is never called. 

### To Reproduce
I created the following MWE according to the QuickStart guide:
```c
#include <lvgl.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI(); /* TFT instance */
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];

void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    tft.startWrite();
    tft.setAddrWindow(area->x1, area->y1, w, h);
    tft.pushColors(&color_p->full, w * h, true);
    tft.endWrite();

    lv_disp_flush_ready(disp);
}

bool my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data)
{
    uint16_t touchX, touchY;

    Serial.println("Touch CB");
    bool touched = tft.getTouch(&touchX, &touchY, 600);
    
    if(!touched) {
      data->state = LV_INDEV_STATE_REL;
    } else {
      data->state = LV_INDEV_STATE_PR;
      
      /*Set the coordinates*/
      data->point.x = touchX;
      data->point.y = touchY;
  
      Serial.print("Data x");
      Serial.println(touchX);
      
      Serial.print("Data y");
      Serial.println(touchY);
    }

    return false; /*Return `false` because we are not buffering and no more data to read*/
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Start...");

  lv_init();
  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

  // TFT 
  tft.begin();
  tft.setRotation(1);
  uint16_t calData[5]={183,3466,239,3573,1};
  tft.setTouch(calData);

  /* Display Driver */
  lv_disp_drv_t disp_drv;               /*Descriptor of a display driver*/
  lv_disp_drv_init(&disp_drv);          /*Basic initialization*/
  disp_drv.flush_cb = my_disp_flush;    /*Set your driver function*/
  disp_drv.buffer = &disp_buf;          /*Assign the buffer to the display*/
  lv_disp_drv_register(&disp_drv);    
  
  lv_indev_drv_t indev_drv;                  /*Descriptor of a input device driver*/
  lv_indev_drv_init(&indev_drv);             /*Basic initialization*/
  indev_drv.type = LV_INDEV_TYPE_POINTER;    /*Touch pad is a pointer-like device*/
  indev_drv.read_cb = my_touchpad_read;      /*Set your driver function*/
  lv_indev_drv_register(&indev_drv);         /*Finally register the driver*/

}

void loop() {
  // put your main code here, to run repeatedly:
  lv_task_handler();
}
```
After compiling and uploading the code, I get as expected a blank screen, but no information about the Touch events on ther Serial console. It seems that the touch screen callback is not called at any time. If I add 
```c
uint16_t touchX, touchY;
   bool touched = tft.getTouch(&touchX, &touchY, 600);
    
    if(!touched) {
      Serial.println("Released");
    } else {
      Serial.println("Touched");
      Serial.print("Data x");
      Serial.println(touchX);
      
      Serial.print("Data y");
      Serial.println(touchY);
    }
```
to the `loop` function I see that the touch screen reacts properly. 

### Expected behavior
The touchscreen callback is executed on user interaction and returns the information.

embeddedt, 2021-01-13:
Is lv_tick_inc called anywhere?

grisuthedragon, 2021-01-13:
Ok, this I forgot. But the Arduino example (https://github.com/lvgl/lvgl/blob/master/examples/LVGL_Arduino.ino) does not include this routine as well and thus was also not working for me.

embeddedt, 2021-01-13:
@grisuthedragon Arduino uses `LV_TICK_CUSTOM` in order to skip needing to call this function; do you have that feature enabled in your project?

grisuthedragon, 2021-01-13:
No, I that I forgot. I hint for this in the quick start guide would be nice and in the Arduino example file.

embeddedt, 2021-01-13:
The Arduino guide mentions it [here](https://docs.lvgl.io/latest/en/html/get-started/arduino.html#configure-lvgl) (last step); is there another page you were looking at?

grisuthedragon, 2021-01-13:
I started with this one https://docs.lvgl.io/v7/en/html/get-started/quick-overview.html

embeddedt, 2021-01-13:
Okay; I've [added a note](https://github.com/lvgl/docs/commit/19403b99cecae371f1c6fa6d1c1df35b23e55c2e) about `LV_TICK_CUSTOM` to it.

closed 2021-05-14 by kisvegabor