lvgl/lvgl #2021: lv_textarea_set_text is a bit slow

zhwxChaos, 2021-01-14:
Hi，   @kisvegabor 

1,   I found that the page loading was a little slow, and then I found that
execute lv_textarea_set_text once
 If the maximum length is set, the execution time is 235 milliseconds.
 If not set, the execution time is 13 milliseconds .

Can its execution speed be optimized?

My LVGL Version  is V7.7.0 and Platform is STM32F429.

2,   By the way，If lv_textarea_set_accepted_chars(textObj, "0123456789");  And lv_textarea_set_text(textOb,"12.3456");  , then text  Textobj will be displayed as 123456  
 
Is there a better plan?

alec-2018, 2021-01-14:
I also encountered the above problem. If the text is very long, the display will be very slow. The reason is that the efficiency here is relatively low, so please optimize it.

> 

- 
[image]

kisvegabor, 2021-01-14:
Hi,

In case of the textarea it should happen because with max. length set, instead of adding a text at once its characters are added one by one. 

However, both 13 and 235 ms is very long. How long is the text you are adding?

alec-2018, 2021-01-14:
Hi, @kisvegabor 
I use lv_textarea_add_char() loop to add characters on the vs simulator, and the speed will gradually slow down over time, What should I do?

embeddedt, 2021-01-14:
@alec-2018 It is better to use `lv_textarea_set_text` if possible, as there are a lot of checks which get repeated on `lv_textarea_add_char`. What is your use case for an `lv_textarea_add_char` loop?

alec-2018, 2021-01-14:
@kisvegabor  This is my test code，I want to make a text edit box, can enter text freely, and the cursor can move freely, but I found that after using textarea, the more text in the text box, the slower the speed of adding text, so I wrote this test Check the reason for the code. After viewing the source code analysis, it is found that every time you add text, all the text in the textarea control needs to be loaded and run, resulting in very low efficiency，Can its execution the algorithm optimize?

static lv_obj_t* ta;
void textarea_create(void)
{
    ta = lv_textarea_create(lv_scr_act(), NULL);

    lv_obj_set_size(ta, LV_HOR_RES, LV_VER_RES);
    lv_obj_set_pos(ta, 0, 0);
    static lv_style_t style_title;

    lv_style_init(&style_title);

    static lv_obj_t* label1 = NULL;
    label1 = lv_textarea_get_label(ta);
	//ADD font
    lv_obj_add_style(label1, LV_LABEL_PART_MAIN, &style_title);
    lv_obj_add_style(ta, LV_OBJ_PART_MAIN, &style_title);
    lv_textarea_set_text(ta, "");
    lv_textarea_add_text(ta, "3 + ? = 12");

    lv_textarea_set_cursor_pos(ta, 5);
	const char* txt = "3 + ? = 12";
    uint32_t  i = 0, pos = 0;
    while (txt[i] != '?') {
        _lv_txt_encoded_next(txt, &i);
        pos++;
    }
    printf("pos %u\n", pos);
    lv_textarea_del_char(ta);
}

  while (1) {
        i++;
        if (i >= 1) {
            i = 0;
            lv_textarea_add_char(ta, '2');
            printf("i = %u.\n", i);
        }

        lv_task_handler();
        MSleep(10);
    }

zhwxChaos, 2021-01-15:
> Hi,
> 
> In case of the textarea it should happen because with max. length set, instead of adding a text at once its characters are added one by one.
> 
> However, both 13 and 235 ms is very long. How long is the text you are adding?

 I use  lv_textarea_set_max_length(textObj, 20);   And  I think the main purpose of this function is to control user input. 

I just tested it again, the method execution time is 121 and 10ms 

[image]

kisvegabor, 2021-01-18:
@zhwxChaos Could you also add a test code that produced that slow refresh?

@alec-2018 
Adding a character is really not optimal, but it still should be faster than something you can notice by eye. I think there is another underlying issue here. 

This code adds charaters to infinity. At how long text did you see it's slow?
```c
while (1) {
   i++;
   if (i >= 1) {
     i = 0;
     lv_textarea_add_char(ta, '2');
     printf("i = %u.\n", i);
   }
    lv_task_handler();
    MSleep(10);
}
```

alec-2018, 2021-01-19:
hi，@kisvegabor result：start （time1 = 1611023852581, time2 = 1611023852588, time = 7.）， end（time1 = 1611024135730, time2 = 1611024135787, time = 57.）Conclusion: With more and more characters, the input will be slow；I used the simulator to calculate the time，The program was executed for a total of 5 minute，It takes 57 milliseconds to add a character '2' at the endminutes，If I put it on my board and use external fonts, it will be very slow. MCU main frequency is 380MHZ. The font is placed in the sd card(Currently I load the font into ram to speed it up).If there are more and more characters, the conclusions obtained on the simulator will be enlarged on muc。

long long get_time_func()
{
     time_t clock;
     struct tm tm;
     SYSTEMTIME wtm;
     GetLocalTime(&wtm);
     tm.tm_year = wtm.wYear - 1900;
     tm.tm_mon = wtm.wMonth - 1;
     tm.tm_mday = wtm.wDay;
     tm.tm_hour = wtm.wHour;
     tm.tm_min = wtm.wMinute;
     tm.tm_sec = wtm.wSecond;
     tm.tm_isdst = -1;
     clock = mktime(&tm);
     long long timestamp = clock * 1000 + wtm.wMilliseconds;
     printf("----> clock = %llu, %u.\n", clock, wtm.wMilliseconds);
     return timestamp;
}

    long long time1 = 0;
    long long time2 = 0;

    while (1) {

        time1 = my_get_time_func();
        lv_textarea_add_char(ta, '2');
        time2 = my_get_time_func();
        printf("time1 = %llu, time2 = %llu, time = %llu.\n", time1, time2, time2 - time1);
        
        lv_task_handler();
        Sleep(10);
    }

zhwxChaos, 2021-01-19:
hi, @kisvegabor 
I called lvgl in the C++ class, and used ucos (OS_TICKS_PER_SEC=1000, 1000 ticks in 1 second) to count. System clock frequency is 168M.

The test situation is roughly as follows：

`char g_fullName[20+1] ; //Read it from FRAM in advance

void  ClassA::OnInitDialog(lv_obj_t* page)
{
	 uint32_t t_a[10];
	 
	 t_a[0] = OSTimeGet();
	 
	ClassBase::OnInitDialog();
	 
	m_editFullName = addEdit(page, NULL, "");
	lv_obj_set_width(m_editFullName, 300);
	lv_obj_set_pos(m_editFullName, 160, 10);
	
	//lv_textarea_set_max_length(m_editFullName, 20);    //  Comment or uncomment to test 2 cases
	lv_obj_add_state(m_editFullName, LV_STATE_DISABLED);
	
	//init other obj
	
	
	t_a[2] = OSTimeGet();
    
	lv_textarea_set_text(m_editFullName, g_fullName);
    
    t_a[3] = OSTimeGet();
    
    printf("=CSetupAnalysisPanel::OnInitDialog,cost Time[%d,%d,%d]==\r\n",t_a[1] - t_a[0],t_a[2] - t_a[1],t_a[3] - t_a[2]);
}

//
static void lvgl_Proc(void* p_arg)
{
    //do something
        while(1)
	{
		lv_task_handler();
		
		OSTimeDly(3);
	}
}

`

kisvegabor, 2021-01-19:
@alec-2018 
> The program was executed for a total of 5 minute，It takes 57 milliseconds to add a character '2' at the endminutes，

So there are about 30 characters in the text area, right? (5 x 60 sec / 10 sec sleep = 30 chars)

> If I put it on my board and use external fonts, it will be very slow. MCU main frequency is 380MHZ. The font is placed in the sd card(Currently I load the font into ram to speed it up).If there are more and more characters, the conclusions obtained on the simulator will be enlarged on muc。

The SD card could be easily a bottleneck. 

@alec-2018 

`t_a[1]` is not set but `t_a[3] - t_a[2]` is still valid and says 121 ms according to your previous log. 

---

I've tested on STM32F746 Discovery (200 MHz) with this code:
```c

    lv_obj_t * ta = lv_textarea_create(lv_scr_act(), NULL);

    uint32_t t = lv_tick_get();
    lv_textarea_set_text(ta, "Hello");
    volatile int elaps1 = lv_tick_elaps(t);

    t = lv_tick_get();
    lv_textarea_add_char(ta, 'a');
    volatile int elaps2 = lv_tick_elaps(t);

    t = lv_tick_get();
    uint32_t i;
    for(i = 0; i < 100; i++) {
        lv_textarea_add_char(ta, 'a');
    }
    volatile int elaps3 = lv_tick_elaps(t);

    t = lv_tick_get();
    lv_textarea_add_char(ta, 'a');
    volatile int elaps4 = lv_tick_elaps(t);
```

Result (`LV_USE_ARABIC_PERSIAN_CHARS` and `LV_USE_BIDI` enabled that makes it slower):
- `elaps_1`: 0 ms
- `elaps_2`: 0 ms
- `elaps_3`: 183 ms
- `elaps_4`: 4 ms

Result (`LV_USE_ARABIC_PERSIAN_CHARS` and `LV_USE_BIDI` disabled):
- `elaps_1`: 0 ms
- `elaps_2`: 0 ms
- `elaps_3`: 60 ms
- `elaps_4`: 2 ms

It's not perfect but not that bad either. What do measure with the code above?

zhwxChaos, 2021-01-20:
@kisvegabor 
I user your code to test on my board:

Not **use lv_textarea_set_max_length(ta,20);**  :
==elaps1 [12],elaps2 [12],elaps3[1618],elaps4[27]==[2021-01-20 09:16:04.236]

Use **lv_textarea_set_max_length(ta,20);**  :
==elaps1 [82],elaps2 [12],elaps3[192],elaps4[0]==[2021-01-20 09:17:17.639]

You can add lv_textarea_set_max_length to compare the time .

alec-2018, 2021-01-20:
@kisvegabor I'm sorry，Make you understand the error. Sleep(10) means 10 milliseconds, So there are about 30000 characters in the text area (5 x 60 x 1000 / 10 milliseconds sleep = 30000 chars), 

This example will illustrate the problem（Made some changes to your test code），Have ”hello“ in the textarea at the beginning，Adding a char 'a' takes 0 milliseconds(elaps2), After executing the for loop，You can get the time after adding a char ‘a’（elaps4），This time will be much longer than the initial addition（elaps4 >> elaps2）；What will be the result if there are a lot of texts？

guess: When there are enough characters in the textarea (for example: >100KB), it will be very slow for the user to continue editing and adding a character in this textarea

    lv_obj_t * ta = lv_textarea_create(lv_scr_act(), NULL); 

    uint32_t t = lv_tick_get();
    lv_textarea_set_text(ta, "Hello");
    volatile int elaps1 = lv_tick_elaps(t);

    t = lv_tick_get();
    lv_textarea_add_char(ta, 'a');
    volatile int elaps2 = lv_tick_elaps(t);

    t = lv_tick_get();
    uint32_t i;
    for(i = 0; i < 30000; i++) {
        lv_textarea_add_char(ta, 'a');
    }
    volatile int elaps3 = lv_tick_elaps(t);

    t = lv_tick_get();
    lv_textarea_add_char(ta, 'a');
    volatile int elaps4 = lv_tick_elaps(t);

embeddedt, 2021-01-20:
I would like to point out the impracticality for an embedded GUI of managing 100KB of raw character data in a text area. At that point you would probably need some type of virtualization logic to get reasonable performance.

alec-2018, 2021-01-20:
@kisvegabor Okay, I understand，At present, I need to load the font into the RAM and limit the length of the text area to have a good experience

kisvegabor, 2021-01-20:
A simple optimization would be to not try to add new characters if the length is already equal to the max length. This way adding the 21th character would "cost" ~0 time. 

> But it may be a pity that it will be slow when inputting more characters. If the number of characters in the text area does not affect the input speed, it will be perfect!!!

I agree but needs some deeper reconsideration. I'm sure it can be done better but I think it's a lower priority issue as the text area is usually used to store 10-20 characters.

zhwxChaos, 2021-01-26:
Hi @kisvegabor ，After setting lv_textarea_set_max_length, the execution time is 6~12 times longer than before. Can this be solved? 

> A simple optimization would be to not try to add new characters if the length is already equal to the max length. This way adding the 21th character would "cost" ~0 time.

kisvegabor, 2021-01-27:
For how many characters?

zhwxChaos, 2021-01-27:
> For how many characters?

20 .

kisvegabor, 2021-02-01:
I see. At first look, it'd require some serious changes. Now I'm deeply immersed in finalizing v8 and can't work on this issuefor a couple of weeks. 

It'd be great if someone could investigate the problem and suggest solutions. 

Anyway, let's leave this open as a reminder.

kisvegabor, 2021-05-14:
I didn't get any good ideas to speed it up, so I close this issue for now.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2021-05-14 by kisvegabor