lvgl/lvgl #2022: textarea Cannot set cursor style

alec-2018, 2021-01-15:
Is the cursor style of 7.9.0 version cut off? I cannot find a function to set the cursor style of the textarea control

embeddedt, 2021-01-15:
https://docs.lvgl.io/latest/en/html/widgets/textarea.html#parts-and-styles

Change the properties of `LV_TEXTAREA_PART_CURSOR` to adjust the cursor style.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2021-01-15 by embeddedt

alec-2018, 2021-01-15:
@embeddedt yes,I have successfully set the cursor style using the following code, thank you very much
    lv_style_set_bg_opa(&style_title, LV_STATE_DEFAULT, LV_OPA_0);
    //lv_style_set_bg_color(&style_title, LV_STATE_DEFAULT, LV_COLOR_BLUE);
    lv_style_set_border_side(&style_title, LV_STATE_DEFAULT, LV_BORDER_SIDE_BOTTOM);
    lv_obj_add_style(ta, LV_TEXTAREA_PART_CURSOR, &style_title);