lvgl/lvgl #2025: recent changes break compilation with micropython

uraich, 2021-01-15:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When updating lvgl in my micropython I get a compilation issue having to do with the garbage collector of micropython
in file included from ../../py/obj.h:33,
                 from ../../py/mpstate.h:35,
                 from ../../lib/lv_bindings/lvgl/src/lv_misc/lv_gc.h:56,
                 from ./mpconfigport.h:263,
                 from ../../py/mpconfig.h:62,
                 from <stdin>:29:
../../py/qstr.h:42:14: fatal error: genhdr/qstrdefs.generated.h: No such file or directory
   42 |     #include "genhdr/qstrdefs.generated.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

### To Reproduce

### Expected behavior

### Screenshots or video

embeddedt, 2021-01-15:
Maybe try a fresh clone or clean build? CI seems to be [passing](https://github.com/lvgl/lvgl/runs/1690142938) at the moment.

uraich, 2021-01-16:
Thanks for trying. I will do the same on my side.
Start with up-to-date MicroPython and add lv_bindings and lvgl?
What do you mean with: "CI seems to be passing at the moment" ?

uraich, 2021-01-16:
I think the reason is that at one stage I upgraded the underlying micropython to 1.13 and the new version of lvgl is incompatible with this. I have similar problems when I try to add lv_bindings to the new version of micropython.
The problems come with the include of LV_GC_INCLUDE which is defined to be py/mpstate.h in several files in lv_misc.
Maybe I can figure out how to solve this and how to make lv_bindings compatible with the new version of micropython.
When I use the current version of lv_micropython, which claims to be v1.9.4-2927, then everything compiles fine.

embeddedt, 2021-01-16:
lv_micropython always reports the wrong version; it has already been
updated to 1.13.

uraich, 2021-01-16:
yes, I read this somewhere. This is why I said: "claims to be" 
However, when I try to compile lv_bindings with the newest micropython I get these errors.
Something must have changed in micropython recently to cause it. Since I now have a micropython version which compiles and the newest one, which does not, I should somehow be able to figure out where the problem comes from.
I see that the include of LV_GC_INCLUDE causes it.
... and this is the result:

In file included from ../../lib/lv_bindings/lvgl/src/lv_misc/lv_fs.c:19:
../../lib/lv_bindings/lvgl/src/lv_misc/lv_fs.c: In function ‘_lv_fs_init’:
../../py/mpstate.h:284:40: error: ‘mp_state_vm_t’ {aka ‘struct _mp_state_vm_t’} has no member named ‘_lv_drv_ll’
  284 | #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
      |                                        ^
./mpconfigport.h:311:23: note: in expansion of macro ‘MP_STATE_VM’
  311 | #define MP_STATE_PORT MP_STATE_VM
      |                       ^~~~~~~~~~~
../../lib/lv_bindings/lv_conf.h:114:25: note: in expansion of macro ‘MP_STATE_PORT’
  114 | #  define LV_GC_ROOT(x) MP_STATE_PORT(x)
      |                         ^~~~~~~~~~~~~
../../lib/lv_bindings/lvgl/src/lv_misc/lv_fs.c:59:18: note: in expansion of macro ‘LV_GC_ROOT’
   59 |     _lv_ll_init(&LV_GC_ROOT(_lv_drv_ll), sizeof(lv_fs_drv_t));
If I manage to find the cause I will lez you know

uraich, 2021-01-16:
found it!
Problems came from an incorrect mpconfigport.h
Now at least the unix port compiles fine again. I am quite confident to be able to get the esp32 port compiled as well.
Like this I should have an up-to-date lv_micropython with the latest micropython edition as base.

uraich, 2021-01-17:
Everything ok now. ESP32 port also working fine. Thanks for your help

closed 2021-01-17 by uraich