lvgl/lvgl #2027: lvgl image cache access a pointer after free the memory

FlyLu, 2021-01-16:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

https://github.com/lvgl/lvgl/blob/873551bb5e84c259404f3897be6d5ca2584d9e1c/src/lv_draw/lv_img_cache.c#L84-L86

When the cache matches, it will look up the source type

https://github.com/lvgl/lvgl/blob/81b13bf3b6219c3b1b05b05209a8449abf7ab84b/src/lv_draw/lv_draw_img.c#L204-L214

In the lv_img_src_get_type function, if the source is released externally(call lv_obj_del), it should not be accessed, such as u8_p[0]

I think the source is **LV_IMG_SRC_VARIABLE** type will cause this problem

### To Reproduce

Use LV_MEM_CUSTOM = 1

```C
#  define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#  define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#  define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
```
test code (canvas use LV_IMG_SRC_VARIABLE type as source)

```C
  void lv_task_test_cb(struct _lv_task_t * task)
  {
      lv_obj_del(task->user_data);
  
      lv_obj_t * img1 = lv_img_create(lv_scr_act(), NULL);
      lv_img_set_src(img1, "z:test.png");
      lv_task_del(task);
  }

  lv_obj_t *canvas = lv_canvas_create(lv_scr_act(), NULL);
  static lv_color_t canvas_buf[252 * 166];
  lv_canvas_set_buffer(canvas, canvas_buf, 252, 166, LV_IMG_CF_TRUE_COLOR_ALPHA);

  lv_task_t *test_task = lv_task_create(lv_task_test_cb, 100, LV_TASK_PRIO_HIGH, canvas);
```
lv_canvas_set_buffer will set the image source, when refreshed it will be cached
lv_task_test_cb  simulates deleting the object and refreshing the next image

A simple fixs on lv_img_signal api 
```C
    if(sign == LV_SIGNAL_CLEANUP) {
        if (LV_IMG_SRC_VARIABLE == ext->src_type && ext->src) {
            lv_img_cache_invalidate_src(ext->src);
        }
    }
```

This problem also exists in other libraries, such as [lv_lib_gif](https://github.com/lvgl/lv_lib_gif)  and [lv_lib_qrcode](https://github.com/lvgl/lv_lib_qrcode) 

Because lv_lib_gif (image) and lv_lib_qrcode (canvas) also use LV_IMG_SRC_VARIABLE type as source

The problem occurs when the object is deleted and the image is refreshed next time for matching cache

kisvegabor, 2021-01-18:
If the image source is manipulated externally [lv_img_cache_invalidate_src](https://docs.lvgl.io/latest/en/html/overview/image.html#clean-the-cache) to let LVGL know about the change.

> This problem also exists in other libraries, such as lv_lib_gif and lv_lib_qrcode

Ah, it seems I also missed to take care of it. Thanks for the report!

kisvegabor, 2021-04-14:
Sorry for the late reply.
I've fixed it for QR code: https://github.com/lvgl/lv_lib_qrcode/commit/c7f5379e19714e4bf6fac1e027b1f20d679fede3
and GIF: https://github.com/lvgl/lv_lib_gif/commit/a9d1c03fb9db5eb357cf18e5222bbc0ba5164c00

closed 2021-04-14 by kisvegabor