lvgl/lvgl #2032: arc fill angle mismatch with arc current value

jupeos, 2021-01-21:
## V7.9.1 using simulator

### Describe the bug
When creating a simple arc using `lv_arc_set_angles` the arc is shown as filled (full) between the 2 angles. However the current value is 0 (obtained with `lv_arc_get_value`).
Aside: because of this, in order to set the value to 0 using `lv_arc_set_value` it is necessary to use:
```
lv_arc_set_value(arc, 1); // or some non-zero value
lv_arc_set_value(arc, 0);
```

### To Reproduce
```
    lv_obj_t* arc = lv_arc_create( lv_scr_act(), NULL );
    lv_obj_set_size( arc, 200, 200 );
    lv_obj_align( arc, NULL, LV_ALIGN_CENTER, 0, 0 );

    lv_arc_set_angles( arc, 135, 45 );
    int16_t currentValue = lv_arc_get_value( arc ); // this is 0
    lv_arc_set_value( arc, 0 );
```
Results in
[image]

### Expected behavior
The fill angle of the arc should be consistent with the current value (in this case 0), i.e. the arc should look like this
[image]

embeddedt, 2021-01-21:
Not sure that this will fix your issue, but I think you have the angles backwards in `lv_arc_set_angles`. Unless that is your intent?

jupeos, 2021-01-22:
> Not sure that this will fix your issue, but I think you have the angles backwards in `lv_arc_set_angles`. Unless that is your intent?

This is what I intended and it makes no difference if you reverse the values other than the visual
[image]

kisvegabor, 2021-01-23:
HI @jupeos,

At this moment there are some issues with using `value` and `angles` together. In short, you can control the arc either by the angle directly or let the arc control it by setting the `value`.

Note that,  `value` is mapped to angles relative to the `bg_angles`. `value == min` means `bg_start_angle` and `value == max` means `bg_end_angle`. 

So you should either remove `lv_arc_set_angles( arc, 135, 45 );` to leave the default value or use `lv_arc_set_bg_angles( arc, 135, 45 );` to set the "angular range" of the arc when `value` is used.

jupeos, 2021-01-27:
Understood. In the interim it would be helpful to make this clear in the documentation to either use 1 approach or the other but not both.

kisvegabor, 2021-01-28:
Added here: https://github.com/lvgl/docs/commit/518b99139dab84fcc772ab82e1fb4ee3a3ecd5f7

It will be visoble when the docs is rebuilt.

embeddedt, 2021-02-06:
Documentation rebuilding is now fixed (and automated) so this is now mentioned [here](https://docs.lvgl.io/latest/en/html/widgets/arc.html#range-and-values).

closed 2021-02-06 by embeddedt